espressif/esp-idf #10951: ESP32-S3 wont support BT but need to handle hand free protocol

saitez, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

currently iam using esp32s3 .
i came to know that esp32s3 supports only BLE but i have to work on HFP protocol to read incoming calls and outgoing calls from the mobile..
Can you please suggest me how can i do Hand free protocol with ble -5.0.
or else suggest me the esp32s3 pin config module which supports BT/BLE because we have our custom hardware.
Thanks

wmy-espressif, 2023-03-10:
Hi @saitez , HFP is a legacy Bluetooth Profile that builds on Classic Bluetooth transport. ESP32-S3 does not support Classic Bluetooth so it cannot be used to build HFP application. ESP32 supports Dual Mode Bluetooth, and you can evaluate it for your project.

saitez, 2023-03-10:
I dont have  a chance to change the hw.
No issue iam looking into bleANCS code

Thanks for your response

wmy-espressif, 2023-03-10:
All right, thanks for your question and information.

Alvin1Zhang, 2023-03-10:
Thanks for sharing the updates, feel feel to reopen.

closed 2023-03-10 by Alvin1Zhang