espressif/esp-idf #10953: What is the esp-idf stable version for esp32s3? "i2c slave devices are not working in esp32s3 but working esp32"

aswathreddy, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a eeprom cat24m1 & rtc ds1388 which is fairly working with esp32. But when i try to run same code on esp32s3 it does not work properly. I feel like it is esp-idf version problem. Can someone help me out to find the correct version esp-idf for esp32s3
My current esp-idf version is ESP-IDF v4.4.2
ESP32 code which is working 

unsigned char readRegister(unsigned char reg)
{
	// printf("Entered the readvalue loop\n");
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	ESP_ERROR_CHECK(i2c_master_start(cmd));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (RTC_ADDRESS << 1) | I2C_MASTER_WRITE, true));//RTC_ADDRESS
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, reg, 1)); // start address
	ESP_ERROR_CHECK(i2c_master_start(cmd));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (RTC_ADDRESS << 1) | I2C_MASTER_READ, true));//RTC_ADDRESS
	uint8_t data = 0;
	ESP_ERROR_CHECK(i2c_master_read_byte(cmd, &data, false));
	ESP_ERROR_CHECK(i2c_master_stop(cmd));
	//i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_PERIOD_MS);
 
    esp_err_t ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if (ret == ESP_OK) {
        //printf("1   Read OK\n");
    } else if (ret == ESP_ERR_TIMEOUT) {
        printf("1    Bus is busy\n");
    } else {
        printf("     Read Failed\n");
    }

	//i2c_cmd_link_delete(cmd);
	return data;
}
 For ESP32S3 i have commented out ESP_ERROR_CHECK(i2c_master_stop(cmd)); which works for rtc 
 and i have no luck with eeprom

CC-MASTER05, 2023-03-11:
have you check timer configuration? or you can try below example
https://github.com/espressif/esp-idf/tree/v5.0.1/examples/peripherals/i2c/i2c_simple

CC-MASTER05, 2023-03-11:
In ESP32 support HIGH SPEED MODE and ESP32S3 didn't support those mode it may generate issue you need to check once

mythbuster5, 2024-11-06:
4.4 is EOL, I will close this issue.

closed 2024-11-06 by mythbuster5