espressif/esp-idf #10954: Has KRACK resurfaced?

fpgax, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Considering the release of https://github.com/espressif/esp-idf/releases/tag/v5.0.1 and mention of CVE-2017-13077, CVE-2017-13078, CVE-2017-13079, CVE-2017-13080, CVE-2017-13081 and CVE-2017-13082.

Is this an indication that there is a regression for the KRACK vulnerability? Which are the affected versions? Is this just on the 5.0 branch or also older branches? Please provide all available information so that I can judge the impact on my end.

kapilkedawat, 2023-03-10:
HI @fpgax , issues were mentioned since we synced the patches from upstream hostapd, no issue was reported during our testing.

fpgax, 2023-03-10:
@kapilkedawat Did you test specifically for KRACK? Are the WiFi allicance tests automatically run (see for reference: https://github.com/espressif/esp-idf/issues/1173)?

Reason I am bringing this up is that the commit where you merged and mentioned the CVEs is being picked up by automatic tooling and probably getting flagged by many organisations around the world, so it would be good to just get a clear statement. Even for the 5.0.0 version whether it existed there or not.

kapilkedawat, 2023-03-24:
@fpgax we have tested KRACK on both versions and there is no issue before and after the change.

fpgax, 2023-03-24:
Thank you @kapilkedawat , closing.

closed 2023-03-24 by fpgax