espressif/esp-idf #10955: BLE-MESH: esp_bt_controller_init failed when using arduino-esp32 as a component in project

hipal31, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3 , ESP32

### Power Supply used.

External 3.3V

### What is the expected behavior?

When I add arduino-esp32 as a component of a ble-mesh project then there must no error on runtime.

### What is the actual behavior?

In my project, I'm developing product using BLE-Mesh. And I need arduino-esp32 as a component in my project. When I add arduino-esp32 as a component then `esp_bt_controller_init ` failed to initialize.

### Steps to reproduce.

1. Take a onoff_server example of ble-mesh
2. Add arduino-esp32(v2.0.7) as a component
3. Compile it 
4. Flash it on esp32c3 board (also try on esp32)

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16cc
load:0x403cc710,len:0x944
load:0x403ce710,len:0x2dfc
entry 0x403cc710
I (30) boot: ESP-IDF v4.4.4-1-g8a624b9b5f-dirty 2nd stage bootloader
I (30) boot: compile time 14:20:18
I (30) boot: chip revision: v0.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (72) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (79) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (87) boot:  3 phy_init         RF data          01 01 00018000 00001000
[... 170 lines cut ...]
3fca9ae0: 0x00000001 0x00000001 0x00000000 0x00000001 0x00000000 0x0000000c 0x00000000 0x00000000
3fca9b00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x42007df2
3fca9b20: 0x00000000 0x00000000 0x00000000 0x4216e9f2 0x00000000 0x00000000 0x00000000 0x4038e614
3fca9b40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca9b60: 0xa5a5a5a5 0x00000154 0x3fca97b0 0x3fca6b64 0x3fc9a36c 0x3fc9a36c 0x3fca9b68 0x3fc9a364
3fca9b80: 0x00000018 0xbc2be427 0x91886245 0x3fca9b68 0x00000000 0x00000001 0x3fca8b64 0x6e69616d
3fca9ba0: 0x8c401b00 0xc9c2ac75 0x004187e7 0x00000000 0x3fca9b60 0x00000001 0x00000000 0x00000000
3fca9bc0: 0x00000000 0x00000000 0x3fca71d8 0x3fca7240 0x3fca72a8 0x00000000 0x00000000 0x00000001
3fca9be0: 0x00000000 0x00000000 0x00000000 0x4215af82 0x00000000 0x00000000 0x00000000 0x00000000
3fca9c00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 3895d7f8f56ffd09
ELF file SHA256: 3895d7f8f56ffd09

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806ba
0x403806ba: esp_restart_noos_dig at D:/Frameworks/esp-idf-v4.4.2/components/esp_system/esp_system.c:53 (discriminator 1)
```

### More Information.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10930771/sdkconfig.txt)

hipal31, 2023-03-09:
@JiaLWang Can you help with this?

hipal31, 2023-03-11:
@xiewenxiang Can you help me with this?

xiewenxiang, 2023-03-16:
@hipal31 Please provide the elf file corresponding to the previous log.

hipal31, 2023-03-16:
@xiewenxiang please find [elf file](https://github.com/espressif/esp-idf/files/10986767/BLE.elf.txt)

hipal31, 2023-03-20:
@xiewenxiang have you gone through the elf file?

hipal31, 2023-03-22:
@xiewenxiang any update on this?
Can anyone help?

hipal31, 2023-03-29:
@xiewenxiang  any update?

MingyaoLiu, 2023-03-31:
i think i have the same issue, try using a lower version of arduino-esp32. 
I have problem using platform io v6.1.0, revert to 5.3.0 no more panic.

hipal31, 2023-04-11:
@igrr Can you help?

esp-cjh, 2023-06-06:
Hi, @hipal31 
I can't convert the CPU register or backtrace value to code locations with your elf file.
You should find someone responsible for arduino-esp32 to help.

hipal31, 2023-09-08:
@zhp0406 Can you please help me with this?

zhp0406, 2023-09-11:
@hipal31 
I am currently reproducing the issue.I could not checkout to v4.4.4-1-g8a624b9b5f,
could you show the version of IDF you are using?

hipal31, 2023-09-11:
@zhp0406 
You can checkout released version v4.4.4 for reproducing the issue.

zhp0406, 2023-09-11:
@hipal31 
I could not reproduce the issue using esp-idf,
are you using arduino-ide? https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

The following is my log, and I didn't find any issues.

I (30) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (30) boot: compile time 11:11:25
I (30) boot: chip revision: v0.3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2aec0h (175808) map
I (126) esp_image: segment 1: paddr=0003aee8 vaddr=3fc90400 size=02e98h ( 11928) load
I (128) esp_image: segment 2: paddr=0003dd88 vaddr=40380000 size=02290h (  8848) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b56c4h (743108) map
I (258) esp_image: segment 4: paddr=000f56ec vaddr=40382290 size=0dfa8h ( 57256) load
I (274) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (286) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (296) cpu_start: cpu freq: 160000000
I (296) cpu_start: Application information:
I (298) cpu_start: Project name:     onoff_server
I (304) cpu_start: App version:      v4.4.4
I (309) cpu_start: Compile time:     Sep 11 2023 11:11:22
I (315) cpu_start: ELF file SHA256:  ef7e316c7061326a...
I (321) cpu_start: ESP-IDF:          v4.4.4
I (326) heap_init: Initializing. RAM available for dynamic allocation:
I (333) heap_init: At 3FC9A370 len 000423A0 (264 KiB): DRAM
I (339) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (346) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (352) spi_flash: detected chip: generic
I (357) spi_flash: flash io: dio
W (361) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (374) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (388) coexist: coexist rom version 9387209
I (393) cpu_start: Starting scheduler.
I (397) EXAMPLE: Initializing...
I (397) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (407) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (417) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (427) BTDM_INIT: BT controller compile version [02bfb7f]
I (427) phy_init: phy_version 950,11a46e9,Oct 21 2022,08:56:12
I (467) system_api: Base MAC address is not set
I (467) system_api: read default base MAC address from EFUSE
I (467) BTDM_INIT: Bluetooth MAC: 58:cf:79:ea:77:32

I (547) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (557) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (557) EXAMPLE: BLE Mesh Node initialized

hipal31, 2023-09-11:
@zhp0406 Did you add arduino-esp32 as a component in your project.
https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html like this?

zhp0406, 2023-09-18:
@hipal31 
Did you make any modifications to the 'onoff_server example'?

hipal31, 2023-09-18:
@zhp0406 NO

hipal31, 2023-10-04:
Issue is Solved in newest version of Arduino-esp32(v2.0.13) and esp-idf(v4.4.4.)

closed 2023-10-04 by hipal31