espressif/esp-idf #10956: Wifi: `assert failed multi_heap_malloc` when connecting to wifi AP

DCSBL, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1 / MON16R8

### Power Supply used.

USB

### What is the expected behavior?

The device should connect to the ESP all the times

### What is the actual behavior?

Instead the ESP crashes sometimes.

### Steps to reproduce.

Hard to tell, it happens not very often. But when it happens is seems to be related with trying to connect to fast after boot.

E.g. The device (in this case an iPhone) is connected to the AP and has an IP. The ESP reboots/resets and the phone retries the connection. The ESP receives the connection request quite early.

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1684
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2ef0
entry 0x403c9904
I (25) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 14:10:33
I (25) boot: chip revision: v0.1
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
<redacted>
I (379) boot: Loaded app from partition at offset 0x30000
I (380) boot: Disabling RNG early entropy source...
I (393) octal_psram: vendor id    : 0x0d (AP)
I (393) octal_psram: dev id       : 0x02 (generation 3)
I (393) octal_psram: density      : 0x03 (64 Mbit)
[... 126 lines cut ...]
0x4205c5b1: hostap_input at ??:?

0x4205acb9: ap_rx_cb at ??:?
```

### More Information.

SPIRAM is used
```
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
CONFIG_SPIRAM_SPEED_120M=y
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=8192
```

Heap poisoning is enabled during early development of the product
```
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
```

nachiketkukade, 2023-03-13:
This issue is same as #9973 , fix is merged on master and will be available on v5.0 in couple of days

DCSBL, 2023-03-13:
Ah yes. Searching is hard. Thanks!

closed 2023-03-13 by DCSBL