espressif/esp-idf #10957: esp32c2 heap corruption during bluetooth de-init

rpiveom, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 v5.1-dev-3839-gafbdb0f3ef

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP8684-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

Expected: Successfully de-init  bluetooth stack while "Light" heap poisoning is enabled

### What is the actual behavior?

Actual: Memory poisoning detects corruption when doing ble de-init

### Steps to reproduce.

1. use blufi example application.
2. If using ESP8684-DevKitM-1, don't forget to set crystal freq to 26MHz
3. Enable CONFIG_HEAP_POISONING_LIGHT
4. Add the following lines at the end of app_main() function:

`    vTaskDelay(5000 / portTICK_PERIOD_MS); // wait 5s`
`    BLUFI_INFO("--------------- Stop BLE\n");`
`    ESP_ERROR_CHECK(esp_blufi_host_deinit());`

5. Flash and run
6. wait 5s

### Debug Logs.

```plain
I (1099) BLUFI_EXAMPLE: BLUFI VERSION 0103

W (1099) wifi:<ba-add>idx:0 (ifx:0, 84:16:f9:6e:98:e4), tid:0, ssn:0, winSize:64
W (1109) wifi: [pm.c,491] AP's beacon interval = 102400 us, DTIM period = 1
I (2099) esp_netif_handlers: sta ip: 192.168.0.101, mask: 255.255.255.0, gw: 192.168.0.254
I (2099) BLUFI_EXAMPLE: BLUFI BLE is not connected yet

I (6099) BLUFI_EXAMPLE: --------------- Stop BLE

E (6099) BT_BTC: BLUFI already de-initialized
I (6099) NimBLE: GAP procedure initiated: stop advertising.

I (6099) BLUFI_EXAMPLE: advertise complete; reason=29
E (6109) BLUFI_EXAMPLE: error setting advertisement data; rc=30

CORRUPT HEAP: Bad tail at 0x3fccbc0c. Expected 0xbaad5678 got 0x01500018

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)
Stack dump detected
Core  0 register dump:
MEPC    : 0x403805d4  RA      : 0x40388c6c  SP      : 0x3fcba4c0  GP      : 0x3fcb03f0  
0x403805d4: panic_abort at /home/test/esp-idf/components/esp_system/panic.c:451

0x40388c6c: __ubsan_include at /home/test/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x3fc9754c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000007f  S1      : 0x00000001  A0      : 0x3fcba4fc  A1      : 0x3fcb1d0d  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fcb6000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcba618  
S4      : 0x3fcb1d0c  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x3fcba4fc "assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)") at /home/test/esp-idf/components/esp_system/panic.c:451
451	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fcba4fc "assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)") at /home/test/esp-idf/components/esp_system/panic.c:451
#1  0x40388c6c in esp_system_abort (details=details@entry=0x3fcba4fc "assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)") at /home/test/esp-idf/components/esp_system/port/esp_system_chip.c:78
#2  0x4038dc4a in __assert_func (file=file@entry=0x3c0992ea "", line=line@entry=276, func=<optimized out>, func@entry=0x3c099338 <__func__.2> "", expr=expr@entry=0x3c0992c4 "") at /home/test/esp-idf/components/newlib/assert.c:81
#3  0x4038d194 in multi_heap_free (heap=0x3fcb7950, p=p@entry=0x3fccbc08) at /home/test/esp-idf/components/heap/multi_heap_poisoning.c:276
#4  0x40380b2a in heap_caps_free (ptr=0x3fccbc08) at /home/test/esp-idf/components/heap/heap_caps.c:376
#5  0x40381f72 in bt_osi_mem_free (ptr=<optimized out>) at /home/test/esp-idf/components/bt/porting/mem/bt_osi_mem.c:49
#6  0x42022032 in r_ble_ll_generic_data_deinit ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: 61c6eaa332a113e9

Rebooting...
```

### More Information.

I've tried with both Nimble and Bluedroid with same result.
(If you want to test blufi example with bluedroid on esp32c2, ensure to enable CONFIG_BT_BLE_BLUFI_ENABLE and bluedroid Bluetooth 4.2 functionalities).

For comparison, I've tried on:
- **esp32**: no corruption detected.
- **esp32c3**: no corruption detected. This is the closest to esp32c2 architecture

I've added several calls to "heap_caps_check_integrity_all()' and traces during the de-init process
Everything works fine until  **esp_bt_controller_deinit()** is called

the corruption is apparently detected in a function called **r_ble_ll_generic_data_deinit**  but may actually have been done before.

I've also tried on release v5.0 and v5.0.1 and the problem was already present.

rpiveom, 2023-04-03:
Hi,

did you reproduce the issue?

luomanruo, 2023-11-17:
This problem has been fixed. The fixed commit id is b39180d7a1. Please test on the latest branch.

rpiveom, 2023-11-17:
I confirm the issue is now solved on master branch

closed 2023-11-17 by rpiveom