espressif/esp-idf #10958: esp_http_client MEMORY LEAK,

AndrzejZawislan, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

No memory leaks

### What is the actual behavior?

Memory leak

### Steps to reproduce.

**Code to reproduce**

```
ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );

esp_http_client_config_t config = {};

config.url = "https://www.google.com/";
config.cert_pem = nullptr;
config.use_global_ca_store = true;
config.transport_type = HTTP_TRANSPORT_OVER_SSL;
config.crt_bundle_attach = esp_crt_bundle_attach;

esp_http_client_handle_t client = esp_http_client_init(&config);
esp_http_client_set_header(client, "Content-Type", "application/json");
esp_http_client_perform(client);
esp_http_client_cleanup(client);

delay(100); //100 ms

ESP_ERROR_CHECK( heap_trace_stop() );
heap_trace_dump();```
                   

### Debug Logs.

```plain
15 allocations trace (100 entry buffer)
4 bytes (@ 0x3ffeb438) allocated CPU 0 ccount 0x66fa1b0c caller 0x400b8f23:0x400c8d0c:0x400c8d8a:0x400ca399:0x400b4f58:0x4011457f:0x4011494c:0x40115006:0x40114096:0x40133748
0x400b8f23: mem_malloc at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/mem.c:237
[... 274 lines cut ...]
0x400bf7bd: sys_check_timeouts at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/timeouts.c:411

0x400b7c27: tcpip_timeouts_mbox_fetch at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/api/tcpip.c:106
 (inlined by) tcpip_thread at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/api/tcpip.c:148

16 bytes (@ 0x3ffe8d60) allocated CPU 0 ccount 0x95b625d8 caller 0x400b8f23:0x400b8f9e:0x400b9008:0x400bf610:0x400bf6b8:0x400bf7bd:0x400b7c27:
0x400b8f23: mem_malloc at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/mem.c:237

0x400b8f9e: do_memp_malloc_pool at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/memp.c:254

0x400b9008: memp_malloc at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400bf610: sys_timeout_abs at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/timeouts.c:201

0x400bf6b8: sys_timeout at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/timeouts.c:325
 (inlined by) tcpip_tcp_timer at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/timeouts.c:165

0x400bf7bd: sys_check_timeouts at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/core/timeouts.c:411

0x400b7c27: tcpip_timeouts_mbox_fetch at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/api/tcpip.c:106
 (inlined by) tcpip_thread at C:/Espressif/frameworks/esp-idf-v4.4.4/components/lwip/lwip/src/api/tcpip.c:148

1272 bytes 'leaked' in trace (15 allocations)
total allocations 354 total frees 12074
```

KGajewskiAMC, 2023-03-23:
Hi,

I am experiencing similar issue using ESP-TLS on IDF v.5.0.1 (release version). Running code below resulted in esp heap memory rapidly decreasing and then stabilizing for some reason after losing approximately 35kB.

Code to reproduce:
```
void app_main(void)
{
    wifi_init();

    while (true)
    {
        printf("free heap: %ld\n", esp_get_free_heap_size());
        if (_wifiOnline)
        {
            esp_tls_t *tls;
            esp_tls_cfg_t cfg = {.timeout_ms = 5000};

            char *hostname = "www.google.com";
            char *payload = "memory_leak";
            tls = esp_tls_init();
            esp_tls_conn_new_sync(hostname, strlen(hostname), 443, &cfg, tls);
            esp_tls_conn_write(tls, payload, strlen(payload));
            esp_tls_conn_destroy(tls);
        }
        vTaskDelay(50 / portTICK_PERIOD_MS);
    }
}
```

AdityaHPatwardhan, 2023-03-24:
@kayzen9 Thanks for linking the relevant issues here.

@KGajewskiAMC 
This is not a memory leak as such. It is expected behaviour. Please find more details about the behaviour in the comments linked above.
Thanks,
Aditya

AdityaHPatwardhan, 2023-03-27:
Closing this issue, Please feel free to re-open if required.

closed 2023-03-27 by AdityaHPatwardhan