espressif/esp-idf #10959: Secure Download Mode - Failed to enter Flash download mode 01060000

ddomnik, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The Security Features are set as the following:
[image]

Disable download stub is also selected.
The eFuse is burnt to Secure Download mode. 
I am able to flash the app partitions, bootloader and so on. Also via esptool.py.

However, I am not able to flash generic .bins that contain data on e.g. data-nvs partitions. 
Is this an intended behavior?

ddomnik, 2023-03-10:
@mahavirj  Thanks for the reply, sorry missed that. Here are some logs:

Full flashing via IDE:
```
Executing action: flash
Running ninja in directory c:\users\USER\documents\esp\ksc-2n\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\USER\Documents\esp\ksc-2n\build\esp-idf\esptool_py && python C:/Espressif/frameworks/esp-idf-v5.0/components/partition_table/check_sizes.py --offset 0xc000 partition --type app C:/Users/USER/Documents/esp/ksc-2n/build/partition_table/partition-table.bin C:/Users/USER/Documents/esp/ksc-2n/build/Appl.bin"
Appl.bin binary size 0x1e1000 bytes. Smallest app partition is 0x280000 bytes. 0x9f000 bytes (25%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\USER\Documents\esp\ksc-2n\build\bootloader\esp-idf\esptool_py && python C:/Espressif/frameworks/esp-idf-v5.0/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x0 C:/Users/USER/Documents/esp/ksc-2n/build/bootloader/bootloader.bin"
Bootloader binary size 0xa000 bytes. 0x2000 bytes (17%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.0\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.0 -D SERIAL_TOOL=python;;C:/Espressif/frameworks/esp-idf-v5.0/components/esptool_py/esptool/esptool.py;--chip;esp32s3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=no_reset;--no-stub;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/USER/Documents/esp/xxx/build -P C:/Espressif/frameworks/esp-idf-v5.0/components/esptool_py/run_serial_tool.cmake"
esptool esp32s3 -p COM4 -b 1500000 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x10000 Appl.bin 0xc000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin

esptool.py v4.5.1
Serial port COM4
Connecting....
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 1500000
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x001f0fff...
Flash will be erased from 0x0000c000 to 0x0000cfff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Erasing flash...
Took 1.08s to erase flash block
Writing at 0x00010000... (0 %)
Writing at 0x00010400... (0 %)
Writing at 0x00010800... (0 %)
-> works
```

Generic .bin flashing:
```
esptool.py -p COM4 write_flash 0x613000 .\generic_data\nvs.bin
esptool.py v4.5.1
Serial port COM4
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00613000 to 0x0061bfff...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```

Writing built partition table:
```
esptool.py -p COM4 write_flash 0xc000 .\build\partition_table\partition-table.bin
esptool.py v4.5.1
Serial port COM4
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000c000 to 0x0000cfff...
Erasing flash...
Took 0.06s to erase flash block
Wrote 3072 bytes at 0x0000c000 in 0.4 seconds (69.8 kbit/s)...

Leaving...
Hard resetting via RTS pin...
```

Erasing flash:
```
esptool.py -p COM4 erase_flash
esptool.py v4.5.1
Serial port COM4
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-S3 ROM does not support function erase_flash.
```

Could it be that I have to somehow sign manually created binaries?

gm-jiang, 2023-03-11:
@ddomnik Please check the start address + length and flash chip size. (0x00613000 + 0x0061cfff > flash chip size?)

ddomnik, 2023-03-13:
@gm-jiang thanks for the reply, but the size and offset should be fine. (And it also worked at that region before enabling secure download mode)

Chip has 16MB flash. 

Reading info does not work either now anymore ...
```
esptool.py -p COM4 chip_id
esptool.py v4.5.1
Serial port COM4
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
Traceback (most recent call last):
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\cmds.py", line 999, in chip_id
    chipid = esp.chip_id()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\targets\esp32.py", line 284, in chip_id
    raise NotSupportedError(self, "chip_id")
esptool.util.NotSupportedError: Function chip_id is not supported for ESP32-S3.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "runpy.py", line 194, in _run_module_as_main
  File "runpy.py", line 87, in _run_code
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\__main__.py", line 9, in <module>
    esptool._main()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\__init__.py", line 1032, in _main
    main()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\__init__.py", line 832, in main
    operation_func(esp, args)
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\cmds.py", line 1003, in chip_id
    read_mac(esp, args)
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\cmds.py", line 989, in read_mac
    mac = esp.read_mac()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\targets\esp32s3.py", line 212, in read_mac
    mac0 = self.read_reg(self.MAC_EFUSE_REG)
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 705, in read_reg
    val, data = self.command(
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 418, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode
```

gm-jiang, 2023-03-13:
@ddomnik Could you please add option `--trace` for esptool.py, it will show more detailed logs to help debugging.

mahavirj, 2023-03-13:
@ddomnik 

Please note that once secure DL mode is enabled, esptool can only support limited commands. Please refer to this [section](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/security.html#uart-dl-mode). You may get basic efuse summary using `esptool.py get_security_info` command.

As I see it from the logs you shared:

```
esptool esp32s3 -p COM4 -b 1500000 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x10000 Appl.bin 0xc000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
```

This command works fine.

```
esptool.py -p COM4 write_flash 0x613000 .\generic_data\nvs.bin
```

This one fails. Is it possible for you to try this command with similar arguments (flash configuration, baud etc.) as the 1st one and see the result? Also does NVS offset makes any difference here, maybe keep it just after partition table?

Note: Secure signing does not apply to NVS partition. This problem is more of flash programming failure. And it would help if you try these experiments with `--trace` option to `esptool.py`.

ddomnik, 2023-03-13:
@mahavirj  @gm-jiang 
Thanks, I am aware that secure download mode limits some functions, although the doc is not very precise here.

```
esptool.py --trace --chip esp32s3 -p COM4 -b 1500000 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x613000 .\generic_data\nvs.bin
esptool.py v4.5.1
Serial port COM4
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.003 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.005 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.010 Read 1 bytes: c0
TRACE +0.003 Read 58 bytes:
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 0108             | . ........
TRACE +0.001 Received full packet: 010804000707122000000000
TRACE +0.003 Received full packet: 010804000707122000000000
TRACE +0.003 Received full packet: 010804000707122000000000
TRACE +0.003 Received full packet: 010804000707122000000000
TRACE +0.006 Read 53 bytes:
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 0108040007071220 | . .............
    00000000c0                        | .....
TRACE +0.001 Received full packet: 010804000707122000000000
TRACE +0.001 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000

TRACE +0.005 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.016 Read 1 bytes: 08
TRACE +0.001 Read 109 bytes:
    0400070712200105 0000c0c001080400 | ..... ..........
    0707122001050000 c0c0010804000707 | ... ............
    122001050000c0c0 0108040007071220 | . .............
    01050000c0c00108 0400070712200105 | ............. ..
    0000c0c001080400 0707122001050000 | ........... ....
    c0c0010804000707 122001050000c0c0 | ......... ......
    0108040007071220 01050000c0       | ....... .....
TRACE +0.002 Received full packet: 010804000707122001050000
TRACE +0.003 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.001 Write 10 bytes: c00014000000000000c0
TRACE +0.009 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000050000 | ................
    c0                                | .
TRACE +0.001 Received full packet:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000050000 | ................
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 1500000
[... 120 lines cut ...]
TRACE +0.010 Read 1 bytes: c0
TRACE +0.001 Read 33 bytes:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000000000 | ................
    c0                                | .
TRACE +0.000 Received full packet:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000000000 | ................
Flash will be erased from 0x00613000 to 0x0061bfff...
Erasing flash...
TRACE +0.004 command op=0x02 data len=20 wait_response=1 timeout=3.000 data=
    0090000024000000 0004000000306100 | ....$........0a.
    00000000                          | ....
TRACE +0.000 Write 30 bytes:
    c000021400000000 0000900000240000 | .............$..
    0000040000003061 0000000000c0     | ......0a......
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010204000707122001060000c0
TRACE +0.000 Received full packet: 010204000707122001060000

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```

I tested both commands on a board that has secure download mode disabled as well.
The second one with less arguments worked, the one with more arguments failed there as well.

gm-jiang, 2023-03-13:
@mahavirj @ddomnik I think it is a esptool.py bugs, please see these logs
> TRACE +0.003 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.001 Write 10 bytes: c00014000000000000c0
TRACE +0.009 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000050000 | ................
    c0                                | .
TRACE +0.001 Received full packet:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000050000 | ................
Chip is ESP32-S3 in Secure Download Mode

> Configuring flash size...
TRACE +0.002 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.001 Write 10 bytes: c00014000000000000c0
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000000000 | ................
    c0                                | .
TRACE +0.001 Received full packet:
    0114180007071220 0400000000000000 | ....... ........
    0000000c09000000 0000000000000000 | ................

In fact, op=0x14 is used for `GetSecurityInfo`, so the `Configuring flash size` did not succeed at all, so it will cause erase flash failed. (the default flash size is set to 2MB)

gm-jiang, 2023-03-14:
@ddomnik  can you have  a try `esptool.py --trace --chip esp32s3 -p COM4 -b 1500000 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 0x1000000 0x613000 .\generic_data\nvs.bin`

ddomnik, 2023-03-14:
@gm-jiang Thanks! The `--flash_size 16MB` parameter does the trick! Ofc `erase_flash` is still not working, but I could flash an empty bin as a workaround for that.

If this ticket is not needed for internal bug tracking, feel free to close it.

mahavirj, 2023-04-24:
@ddomnik We have added a fix in the `esptool` (https://github.com/espressif/esptool/commit/0e9c85ed2a5c80a51a550efea1b99156bbaacc63) to report an error in case flash size is not specified with `--keep` option for the secure ROM DL mode case. This should help to detect the scenario reported here. Closing the issue.

closed 2023-04-24 by mahavirj