espressif/esp-idf #10962: docker exec without entrypoint.sh

ligantx, 2023-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I'm using esp-df docker container and i would like to do frequent rebuilds without recreating the container everytime.
The problem is, if i create the container:
 `docker run -dit --name test -w /project espressif/idf`
And then i do:
`docker exec test idf.py build` 
The idf.py is not recognized, as the entrypoint.sh is running only during container creation for the initial session.
One solution is:
`docker exec test /opt/esp/entrypoint idf.py build` 
but i will have to wait for some time everytime i rebuild, even for small changes until the entrypoint script is over.
Another option is to:
`docker attach test` and then type `idf.py build`
But this is not a good option for my case, as i would like to stay in the host's terminal and run all the commands from the same session.
I would like to know if there are only some mandatory env variables that i can set during container creation, instead of running the entrypoint in every docker exec.

### What is the actual behavior?

`docker exec test idf.py build` 
idf.py is not recognized

### Steps to reproduce.

1.  `docker run -dit --name test -w /project espressif/idf`
2. `docker exec test idf.py build`

igrr, 2023-03-09:
Hi @ligantx,

Something like this might work for you:

1. Create the export shell script which sets the environment variables:
   ```bash
   docker exec test /bin/bash -c '$IDF_PATH/tools/idf_tools.py export --format=shell > env.sh'
   ```
2. Source the script before each run:
   ```bash
   docker exec test /bin/bash -c '. ./env.sh; $IDF_PATH/tools/idf.py build'
   ```

`idf_tools.py export` also supports `--format=key-value`, the output of which can be saved as an .env file.

Unfortunately, `docker exec --env-file .env idf.py build` won't directly work with such an .env file, since it contains an unexpanded reference to `$PATH`. You would need to run the .env file through envsubst or equivalent, first, before passing the .env file to `docker exec --env-file`.

Note that I wouldn't classify this as a bug

igrr, 2023-03-10:
> but can you explain this sentence: Unfortunately, docker exec --env-file .env idf.py build won't directly work with such an .env file, since it contains an unexpanded reference to $PATH.?
You mean that the first time it'll run $PATH is not known, so in a way you have to run "twice"?

When you run `idf_tools.py export --format=key-value`, one of the lines of output will look like this:
```
PATH=<stuff added by IDF>:$PATH
```

But the issue is, when you pass this file to `docker exec --env-file`, it doesn't expand `$PATH` variable:
```
# create the env file
$ docker exec test /bin/bash -c '/opt/esp/idf/tools/idf_tools.py export --format=key-value > .env'
# check the value of PATH
$ docker exec --env-file .env test /bin/bash -c 'echo $PATH'
<stuff added by IDF>:$PATH
```
and because of that, none of the standard programs (in /bin, /usr/bin, /usr/local/bin, etc) will be found.

I think it's a docker limitation that you can't extend PATH via the --env-file, or at least I haven't found a way around this.

Something like this could probably work, but we don't have `envsubst` in our docker image:
```
$ docker exec test /bin/bash -c '/opt/esp/idf/tools/idf_tools.py export --format=key-value | envsubst > .env'
```

---

> Is it possible to create the env file at the time of container creation with the entrypoint and keep the container running? Because,  docker run -dit espressif/idf:latest /bin/bash -c '$IDF_PATH/tools/idf_tools.py export --format=shell > env.sh' and then docker exec returns You cannot attach to a stopped container, start it first

This looks like a generic Docker question, not something ESP-IDF specific? You can add some command in the end which won't terminate, like `tail`:
```
docker run -dit --name test espressif/idf:latest /bin/bash -c '$IDF_PATH/tools/idf_tools.py export --format=shell > env.sh; tail'
```

ligantx, 2023-03-10:
>and because of that, none of the standard programs (in /bin, /usr/bin, /usr/local/bin, etc) will be found

Ok, I see, this works as i imagined..
>This looks like a generic Docker question, not something ESP-IDF specific? You can add some command in the end which won't terminate, like tail

I also tried [this ](https://github.com/docker/getting-started/issues/201) or just `tail`:
`docker run -dit --name test espressif/idf:latest /bin/bash -c '$IDF_PATH/tools/idf_tools.py export --format=shell > env.sh; tail -f /dev/null'`

but 
`docker exec test /bin/bash -c 'source env.sh; $IDF_PATH/tools/idf.py build`
returns:
`Please use idf.py only in an ESP-IDF shell environment.`
But, running `docker exec idf_tools.py export` after `docker run ... espressif/idf` that runs `entrypoint.sh` is just a few seconds more, and only one time, so i'll stick with the initial 2 `docker exec` solution.
Issue closed. Thank you