espressif/esp-idf #10963: Trying to use w5500 as the PHY for Ethernet communication on ESP32

imammy-hacomono, 2023-03-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to use w5500 as the PHY for Ethernet communication on ESP32.
I tried running the esp_ping command to ping "google.com", but the processing time is unusually long.

Using ESP-IDF Version: v5.0.1

~~~
I (29363) [Eth]: 64 bytes from 142.250.207.46 icmp_seq=1 ttl=60 time=994 ms
I (30363) [Eth]: 64 bytes from 142.250.207.46 icmp_seq=2 ttl=60 time=999 ms
I (31363) [Eth]: 64 bytes from 142.250.207.46 icmp_seq=3 ttl=60 time=998 ms
I (32363) [Eth]: 64 bytes from 142.250.207.46 icmp_seq=4 ttl=60 time=999 ms
I (32363) [Eth]: 
--- 142.250.207.46 ping statistics ---

I (32363) [Eth]: 4 packets transmitted, 4 received, 0% packet loss, time 3990ms
~~~
It seems to always take more than 900ms.
I have also tried HTTP communication, but I still feel that the communication time is long.

The Ethernet PHY speed is set to 10 Mbps. Will this affect the processing time?
I tried to set it to 100 Mbps using ioctl, but when I read the configuration value after setting it, it remained at 10 Mbps.
~~~
eth_speed_t speed = ETH_SPEED_100M;
    ret = esp_eth_ioctl(eth_handle_spi, ETH_CMD_S_SPEED, &speed);
    if (ret != ESP_OK) {
        printf("Failed to set Ethernet speed\n");
    }

 eth_speed_t eth_speed;
 ret = esp_eth_ioctl(eth_handle_spi, ETH_CMD_G_SPEED, &eth_speed);
 if (ret == ESP_OK) {
    printf("Ethernet PHY speed: %s\n", eth_speed == ETH_SPEED_100M ? "100 Mbps" : "10 Mbps");
} else {
    printf("Failed to get Ethernet PHY speed, error = %d\n", ret);
}
~~~

~~~
ip_addr_t target_addr;
    struct addrinfo hint;
    struct addrinfo* res = NULL;
    memset(&hint, 0, sizeof(hint));
    memset(&target_addr, 0, sizeof(target_addr));
    getaddrinfo("google.com", NULL, &hint, &res); 
    struct in_addr addr4 = ((struct sockaddr_in*) (res->ai_addr))->sin_addr;
    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
    freeaddrinfo(res);

    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    ping_config.target_addr = target_addr;
    ping_config.count = 4;

    esp_ping_callbacks_t cbs = {
        .on_ping_success = OnPingSuccessCallback,
        .on_ping_timeout = OnPingTimeoutCallback,
        .on_ping_end = OnPingEndCallback,
    };

    esp_ping_stop(s_ping_handle);
    esp_ping_delete_session(s_ping_handle);
    esp_ping_new_session(&ping_config, &cbs, &s_ping_handle);
    esp_ping_start(s_ping_handle);
~~~

imammy-hacomono, 2023-03-13:
@ESP-Marius 
Could you (or someone) help me with this issue, please? Thank you.

ESP-Marius, 2023-03-13:
Sorry, I dont know much about `esp_eth` related stuff, maybe @kostaond would be able to assist with this?

imammy-hacomono, 2023-03-13:
@ESP-Marius , thank you for your response.
To @kostaond , I will send the code for initializing the PHY of the w5500.
Thank you.

~~~
static void EthernetHardwareEventHandler(void* arg, esp_event_base_t event_base,
                                         int32_t event_id, void* event_data) {
    uint8_t mac_addr[6] = { 0 };
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t*) event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(kW5500LogTag, "Ethernet Link Up");
        ESP_LOGI(kW5500LogTag, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(kW5500LogTag, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(kW5500LogTag, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(kW5500LogTag, "Ethernet Stopped");
        break;
    default:
        break;
    }
    s_event_callback(event_id);
}

static void GetIpEventHandler(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data) {
    ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
    const esp_netif_ip_info_t* ip_info = &event->ip_info;

    ESP_LOGI(kW5500LogTag, "Ethernet Got IP Address");
    ESP_LOGI(kW5500LogTag, "~~~~~~~~~~~");
    ESP_LOGI(kW5500LogTag, "ETHIP:" IPSTR, IP2STR(&ip_info->ip));
    ESP_LOGI(kW5500LogTag, "ETHMASK:" IPSTR, IP2STR(&ip_info->netmask));
    ESP_LOGI(kW5500LogTag, "ETHGW:" IPSTR, IP2STR(&ip_info->gw));
    ESP_LOGI(kW5500LogTag, "~~~~~~~~~~~");
    memcpy(&s_ip_info, &event->ip_info, sizeof(s_ip_info));
}

void W5500Init(void) {
    // Initialize TCP/IP network interface (should be called only once in application)
    ESP_ERROR_CHECK(esp_netif_init());
    // Create default event loop that running in background
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    //  Create instance(s) of esp-netif for SPI Ethernet(s)
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_ETH();
    esp_netif_config_t cfg_spi = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH
    };
    esp_netif_t* eth_netif_spi = NULL;
    char if_key_str[10];
    char if_desc_str[10];
    char num_str[3];
    itoa(0, num_str, 10);
    strcat(strcpy(if_key_str, "ETH_SPI_"), num_str);
    strcat(strcpy(if_desc_str, "eth"), num_str);
    esp_netif_config.if_key = if_key_str;
    esp_netif_config.if_desc = if_desc_str;
    esp_netif_config.route_prio = 30;
    eth_netif_spi = esp_netif_new(&cfg_spi);

    // Init MAC and PHY configs to default
    eth_mac_config_t mac_config_spi = ETH_MAC_DEFAULT_CONFIG();

    eth_phy_config_t phy_config_spi = ETH_PHY_DEFAULT_CONFIG();
    phy_config_spi.autonego_timeout_ms = 0;
    phy_config_spi.reset_gpio_num = -1;

     gpio_install_isr_service(0);

    // Init SPI bus
    spi_bus_config_t buscfg = {
        .miso_io_num = GPIO_NUM_19,
        .mosi_io_num = GPIO_NUM_23,
        .sclk_io_num = GPIO_NUM_18,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

    // Configure SPI interface and Ethernet driver for specific SPI module
    esp_eth_mac_t* mac_spi;
    esp_eth_phy_t* phy_spi;
    esp_eth_handle_t eth_handle_spi = NULL;
    spi_device_interface_config_t spi_devcfg = {
        .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
        .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
        .spics_io_num = GPIO_NUM_5,
        .mode = 0,
        .clock_speed_hz = 33 * 1000 * 1000,
        .queue_size = 20
    };

    // Set remaining GPIO numbers and configuration used by the SPI module
    phy_config_spi.phy_addr = 1;

    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(SPI2_HOST, &spi_devcfg);
    w5500_config.int_gpio_num = 4;
    mac_spi = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
    phy_spi = esp_eth_phy_new_w5500(&phy_config_spi);

    esp_eth_config_t eth_config_spi = ETH_DEFAULT_CONFIG(mac_spi, phy_spi);
    ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config_spi, &eth_handle_spi));

    uint8_t mac_addr[6];
    esp_read_mac(mac_addr, ESP_MAC_ETH);

    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle_spi, ETH_CMD_S_MAC_ADDR, mac_addr));

    // attach Ethernet driver to TCP/IP stack
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif_spi, esp_eth_new_netif_glue(eth_handle_spi)));

    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &EthernetHardwareEventHandler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &GetIpEventHandler, NULL));

    /* start Ethernet driver state machine */
    ESP_ERROR_CHECK(esp_eth_start(eth_handle_spi));
    ESP_ERROR_CHECK(esp_netif_init());
~~~

kostaond, 2023-03-13:
@imammy-hacomono let's have a look at it.

The PHY is set to auto-negotiation by default. Therefore, it depends on result of negotiation between the W5500 and endnode device. To get the correct speed, you need to wait for "Link UP" event.

You should be able to configure the W5500 to different speed manually, you only need to do it when Ethernet is stopped and you also need to configure the other device to the same speed. However, again,  you need to wait for "Link UP" event to display the speed properly.

What is your setup? What is your endnode device? Is it router, is it PC? However, in any case, ~1 ms to ping response is too much even for 10 mbps. Could you try to ping W5500 in your local network, e.g. from your PC?

imammy-hacomono, 2023-03-13:
@kostaond 

Thank you!
At the Link UP event, it seems that the speed is 100Mbps.

~~~
static void EthernetHardwareEventHandler(void* arg, esp_event_base_t event_base,
                                         int32_t event_id, void* event_data) {
    uint8_t mac_addr[6] = { 0 };
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t*) event_data;
    eth_speed_t speed;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(kW5500LogTag, "Ethernet Link Up");
        ESP_LOGI(kW5500LogTag, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        esp_eth_ioctl(eth_handle, ETH_CMD_G_SPEED, &speed);
        ESP_LOGI(kW5500LogTag, "Ethernet Speed:%d", speed);
        break;
~~~

~~~
I (17955) [W5500]: Ethernet Link Up
I (17955) [W5500]: Ethernet HW Addr 70:b8:f6:9b:c7:47
I (17955) [W5500]: Ethernet Speed:1

I (20455) [W5500]: Ethernet Got IP Address
I (20455) [W5500]: ~~~~~~~~~~~
I (20455) [W5500]: ETHIP:192.168.0.209
I (20465) [W5500]: ETHMASK:255.255.255.0
I (20465) [W5500]: ETHGW:192.168.0.1
I (20475) [W5500]: ~~~~~~~~~~~
~~~

The end node device is a router. 
When I sent a ping to the W5500 from a PC on the same network, I got the following results.

Is there anything that can be understood from these?
~~~
PING 192.168.0.209 (192.168.0.209): 56 data bytes
Request timeout for icmp_seq 0
64 bytes from 192.168.0.209: icmp_seq=0 ttl=255 time=1170.306 ms
64 bytes from 192.168.0.209: icmp_seq=1 ttl=255 time=168.225 ms
64 bytes from 192.168.0.209: icmp_seq=2 ttl=255 time=161.588 ms
64 bytes from 192.168.0.209: icmp_seq=3 ttl=255 time=157.454 ms
^C
--- 192.168.0.209 ping statistics ---
4 packets transmitted, 4 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 157.454/414.393/1170.306/436.443 ms
~~~

kostaond, 2023-03-13:
150 ms is really poor performance. It should be something around ~2 ms. Do you use multiple SPI devices on the same bus as the W5500? Do you run some extensive application on ESP32? Could you please perform a measurement with just `basic` Ethernet example?

imammy-hacomono, 2023-03-13:
@kostaond 
Only W5500 is connected to the SPI bus.
I tried running only Ethernet in the application, but the results were not significantly different.

~~~
I (490) [InitializeTask]: ESP-IDF Version:v5.0.1-dirty
I (520) w5500.mac: version=4
I (530) system_api: Base MAC address is not set
I (530) system_api: read default base MAC address from EFUSE
I (540) esp_eth.netif.netif_glue: 70:b8:f6:9b:c7:47
I (540) esp_eth.netif.netif_glue: ethernet attached to netif
I (560) [W5500]: Ethernet Started
I (2560) [W5500]: Ethernet Link Up
I (2560) [W5500]: Ethernet HW Addr 70:b8:f6:9b:c7:47
I (2560) [W5500]: Ethernet Speed:1
I (5060) esp_netif_handlers: eth0 ip: 192.168.0.209, mask: 255.255.255.0, gw: 192.168.0.1
I (5060) [W5500]: Ethernet Got IP Address
I (5060) [W5500]: ~~~~~~~~~~~
I (5060) [W5500]: ETHIP:192.168.0.209
I (5070) [W5500]: ETHMASK:255.255.255.0
I (5070) [W5500]: ETHGW:192.168.0.1
I (5080) [W5500]: ~~~~~~~~~~~

I (8510) [Eth]: 64 bytes from 216.58.220.110 icmp_seq=1 ttl=60 time=989 ms
I (9510) [Eth]: 64 bytes from 216.58.220.110 icmp_seq=2 ttl=60 time=991 ms
I (10510) [Eth]: 64 bytes from 216.58.220.110 icmp_seq=3 ttl=60 time=991 ms
I (11510) [Eth]: 64 bytes from 216.58.220.110 icmp_seq=4 ttl=60 time=991 ms
~~~

~~~
64 bytes from 192.168.0.209: icmp_seq=0 ttl=255 time=1095.790 ms
64 bytes from 192.168.0.209: icmp_seq=1 ttl=255 time=94.801 ms
64 bytes from 192.168.0.209: icmp_seq=2 ttl=255 time=90.130 ms
64 bytes from 192.168.0.209: icmp_seq=3 ttl=255 time=84.761 ms
64 bytes from 192.168.0.209: icmp_seq=4 ttl=255 time=80.282 ms
64 bytes from 192.168.0.209: icmp_seq=5 ttl=255 time=82.467 ms
C64 bytes from 192.168.0.209: icmp_seq=6 ttl=255 time=75.077 ms
~~~

kostaond, 2023-03-13:
Could you please run the `basic` example?

kostaond, 2023-03-13:
you can find it [here](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic).

imammy-hacomono, 2023-03-13:
@kostaond 
I am getting an error. What should I do to fix it?
~~~
I (348) spi_flash: detected chip: generic
I (351) spi_flash: flash io: dio
W (355) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (369) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (479) w5500.mac: w5500_reset(222): reset timeout
E (479) w5500.mac: emac_w5500_init(701): reset w5500 failed
I (479) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (489) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x400d76ee
0x400d76ee: app_main at /Users/keigo.imaizumi/workspace/ESP-32_Examples/basic/main/ethernet_example_main.c:203 (discriminator 1)

file: "./main/ethernet_example_main.c" line 203
func: app_main
expression: esp_eth_driver_install(&eth_config_spi, &eth_handle_spi[i])

abort() was called at PC 0x40086ab3 on core 0
0x40086ab3: _esp_error_check_failed at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x40081cbe:0x3ffb5420 0x40086abd:0x3ffb5440 0x4008c50a:0x3ffb5460 0x40086ab3:0x3ffb54d0 0x400d76ee:0x3ffb5500 0x40102b1f:0x3ffb5630 0x400897e1:0x3ffb5660
0x40081cbe: panic_abort at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/panic.c:423

0x40086abd: esp_system_abort at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/esp_system.c:153

0x4008c50a: abort at /Users/keigo.imaizumi/esp/esp-idf/components/newlib/abort.c:38

0x40086ab3: _esp_error_check_failed at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/esp_err.c:47

0x400d76ee: app_main at /Users/keigo.imaizumi/workspace/ESP-32_Examples/basic/main/ethernet_example_main.c:203 (discriminator 1)

0x40102b1f: main_task at /Users/keigo.imaizumi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x400897e1: vPortTaskWrapper at /Users/keigo.imaizumi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

~~~

kostaond, 2023-03-13:
Did you properly configure SPI GPIO in Menuconfig?

imammy-hacomono, 2023-03-13:
[image]
@kostaond 
Here are the settings.

kostaond, 2023-03-13:
SCKL and MISO are both 18 if I see correctly. Please fix and give it another try.

imammy-hacomono, 2023-03-13:
@kostaond 
I successfully built it and sent a ping from my PC to the ESP32, but the result remained unchanged.
~~~
PING 192.168.0.56 (192.168.0.56): 56 data bytes
Request timeout for icmp_seq 0
64 bytes from 192.168.0.56: icmp_seq=0 ttl=255 time=1943.101 ms
64 bytes from 192.168.0.56: icmp_seq=1 ttl=255 time=942.621 ms
64 bytes from 192.168.0.56: icmp_seq=2 ttl=255 time=1012.169 ms
64 bytes from 192.168.0.56: icmp_seq=3 ttl=255 time=1020.645 ms
64 bytes from 192.168.0.56: icmp_seq=4 ttl=255 time=948.403 ms
64 bytes from 192.168.0.56: icmp_seq=5 ttl=255 time=970.579 ms
~~~

kostaond, 2023-03-13:
Try to connect the W5500 directly to your PC. You are on Windows, right? Configuring the DHCP Server on Windows is not so easy as on Linux. Therefore, it will be easier to statically configure the IP address on your PC and statically configure IP address on the ESP32. You can find some inspiration of how to do it [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/static_ip/main/static_ip_example_main.c#L65). Once you are done, try the ping again.

imammy-hacomono, 2023-03-13:
@kostaond 

My PC is a Mac. Does this mean that I can set up a DHCP server on my Mac and connect an ESP32 directly? It seems like it might take a while, as my MacBook only has a USB-C port and I don't have a converter to connect the Ethernet cable.
By the way, what is the purpose of confirming whether this is feasible?

kostaond, 2023-03-13:
I am not able to reproduce the issue locally at my side. Therefore, we need to try to isolate the problem in your setup step by step. This is usual approach... Excluding potential trouble makers from the chain. However, I've just noticed and realized that the ping delay is suspiciously close to one sec. It could indicate you may have problems with interrupt since there is [1 sec timeout when no interrupt is received](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_w5500.c#L644). Please double check that your wiring and connection of INT signal is correct.

> Does this mean that I can set up a DHCP server on my Mac and connect an ESP32 directly

Yes, or configure the IP addresses statically.

imammy-hacomono, 2023-03-14:
@kostaond 
I'm sad to say that I've discovered that the INT signal isn't connected to the ESP32... 
In order to receive packets quickly, I think the only option is to reduce the value of "1000" in the following code, which will shorten the polling interval for checking whether packets have been received. 
Is my understanding correct?

~~~
if (ulTaskNotifyTake(pdTRUE, pdMS_TO_TICKS(1000)) == 0 &&    // if no notification ...
~~~

In this case, the CPU load will increase because the polling interval will be shorter...

kostaond, 2023-03-14:
Do you have a custom PCB or do you use W5500 module connected to the ESP32 via wires? If PCB, you can solder extra wire to INT and fix it properly in the new revision. It is quite normal that the very first PCB revision has some issues...

Shortening the timeout in W5500 driver will partially solve your issue but it is not ideal. The CPU will be utilized more, as you also pointed, and the responsiveness of the Ethernet would be worse than if you used the interrupt.

It seems we were able to identify the root cause of your problem, could you please close this GH issue?

imammy-hacomono, 2023-03-14:
@kostaond 
Since it's a PCB, I'm going to solder INTs to unused GPIOs...
I would like to use interrupts as much as possible.
Thank you! You just solved my problem! Thank you so much!

closed 2023-03-14 by imammy-hacomono