espressif/esp-idf #10964: ESP OTA failing when secure boot and flash encryption enabled on N16R8 ESP32S3

Ramakrishna247, 2023-03-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 N16R8 

### Power Supply used.

External 3.3V

### What is the expected behavior?

When over OTA image gets downloaded , 2nd stage bootloader should verify(secure boot and flash encryption enabled) downloaded image , switch partition and load image from second partition.

### What is the actual behavior?

OTA image bin got downloaded , signature of the image got verified , but a stack over flow is happening as soon as image signature got verified. Also if i disable secure boot and flash encryption and test the OTA , OTA is happening successfully at that time no stackoverflow is happening. Also i have added some logs in ESP OTA API files and i have observed this issue is happening in esp_ota_ops.c file.  this is the same case with IDF version 4.4, 4.4.3, 5.0

### Steps to reproduce.

1. Enable secure boot and flash encryption with  on ESP32S3 N16R8 board using vscode
2. Use https_ota library and do simple OTA
3. with below partition table
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs_key,data,nvs_keys,0xc000,4K,encrypted
nvs,data,nvs,0xd000,64K,
otadata,data,ota,0x1d000,8K,
phy_init,data,phy,0x1f000,4K,
ota_0,app,ota_0,0x20000,2M,
ota_1,app,ota_1,0x220000,2M,
file,data,spiffs,0x420000,12160K,

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fcd0270,len:0x3590
load:0x403b6000,len:0xe28
load:0x403ba000,len:0x5258
entry 0x403b6274
I (72) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
[... 332 lines cut ...]

Spi_Pad_Config fuses:
SPI_PAD_CONFIG_CLK (BLOCK1)                        SPI CLK pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI Q (D1) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI D (D0) pad                                     = 0 R/W (0b000000)
BTLC_GPIO_ENABLE (BLOCK0)                          Enables BTLC GPIO                                  = 0 R/W (0b00)
DIS_USB_JTAG (BLOCK0)                              Disable usb_serial_jtag-to-jtag function           = True R/W (0b1)
DIS_USB_SERIAL_JTAG (BLOCK0)                       Disable usb_serial_jtag module                     = False R/W (0b0)
USB_PHY_SEL (BLOCK0)                               Select internal/external PHY for USB OTGand usb_se = False R/W (0b0)
                                                   rial_jtag

Vdd_Spi Config fuses:
VDD_SPI_XPD (BLOCK0)                               The VDD_SPI regulator is powered on                = True R/W (0b1)
VDD_SPI_TIEH (BLOCK0)                              The VDD_SPI power supply voltage at reset
   = Connect to VDD_RTC_IO R/W (0b1)
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD_SPI R/W (0b1)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.

C:\esp\esp_projects\test_project>

mahavirj, 2023-03-10:
@Ramakrishna247 

> ***ERROR*** A stack overflow in task main has been detected.

Try increasing main task stack size (`CONFIG_ESP_MAIN_TASK_STACK_SIZE`), please see this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/startup.html#running-the-main-task). Alternatively you can use IDF provided OTA examples which typically spawns an independent task to handle OTA updates. Secure boot verification process involves crypto operations and hence higher stack size requirement.

Ramakrishna247, 2023-03-10:
Thanks for the reply, I will check and update.

Ramakrishna247, 2023-03-10:
@mahavirj 

Thank you very much for the support , I have increased the stack size from 3584 to 4608 and OTA worked, but is there is any document which can tell how much additional stack size required for OTA  crypto operations on average, also are there any API's to know real time stack usage values?

mahavirj, 2023-03-10:
> document which can tell how much additional stack size required

Please refer to this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/performance/ram-usage.html#reducing-stack-sizes) to fine tune the task stack size.

I would go ahead and close the issue as its fixed on your side.

closed 2023-03-10 by mahavirj