espressif/esp-idf #10966: ESP32 S2/S3 USB HID Host for gamepads and joysticks

relaxibus, 2023-03-10:
### Is your feature request related to a problem?

Looking for a way to connect USB gamepads and joysticks to the ESP32 S2/S3 and to convert the USB streams into UART and CAN.

### Describe the solution you'd like.

@Dazza0 wrote : Joy sticks (and other non-boot protocol HID devices) will take longer to support due to the need to parse HID report descriptors. Currently, there is already some logic in the esp_hid component for HID report descriptors. But our long term plan is to tidy up esp_hid and integrate USB Host HID with esp_hid somehow so that HID applications it can share the same HID descriptor logic with BT/BTLE HID devices as well. This will allow users to...

use a common  esp_hid (or similar) API for all HID devices, regardless of what protocol the device is running on (i.e., USB/BT/BTLE).
used the provided HID report descriptor APIs to easily add their own drivers for less common/custom HID devices
As for USB Hubs in Host mode, that is also in our roadmap. But, unfortunately I can't offer a release date for that yet.

What is the status of this topic?

Dazza0, 2023-03-14:
@relaxibus Unfortunately, we still don't have any updates on the `esp_hid` components (i.e., out of the box report descriptor parsing). However, were working on some updates to the Host HID driver API such that the user can obtain raw reports.

This should allow supporting of joysticks, though you'll need to implement your own parsing of reports and report descriptors.

relaxibus, 2024-07-02:
Hi, any update on the topic of reading and parsing Gamepads/Joysticks?