espressif/esp-idf #10968: AC Dimming not happening in ESP32S3

CC-MASTER05, 2023-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have tested AC dimming in ESP32 and they are supported LEDC_HIGH_SPEED_MODE and they working fine.
now those same code i implement on ESP32-S3 and change some required function still i didn't get AC Dimming.

Used Hardware :
Micro Controller : ESP32-S3-DevkitV1
Circuit  : Z -Cross AC detector 

Please found in below code :
```
Change 1:
ledc_timer_config_t timer_config = {
        .speed_mode = LEDC_LOW_SPEED_MODE,  //change mode
        .timer_num = 0,
        .duty_resolution = DUTY_BIT_DEPTH,
        .freq_hz = HZ * 2,
    };
    ESP_ERROR_CHECK(ledc_timer_config(&timer_config));

    ESP_LOGI(LOG_STARTUP, "Configured timer.\n");
    ledc_cbs_t callbacks = {
        .fade_cb = cb_ledc_fade_end_event};
    SemaphoreHandle_t counting_sem = xSemaphoreCreateCounting(NUM_CHANNELS, 0);
    for (int i = 0; i < NUM_CHANNELS; ++i)
    {
        ledc_channel_config_t led_config = {
            .gpio_num = channel_pins[i],
            .speed_mode = LEDC_LOW_SPEED_MODE,  //change mode
            .channel = i,
            .timer_sel = LEDC_TIMER_0,
            .duty = (1 << DUTY_BIT_DEPTH) - 1,
            .intr_type = LEDC_INTR_DISABLE,
        };
        ESP_ERROR_CHECK(ledc_channel_config(&led_config));
        LEDC.channel_group[0].channel[i].duty.duty = TRIAC_GATE_IMPULSE_CYCLES << 4;
        // Initial brightness of 0, meaning turn TRIAC on at very end:
        LEDC.channel_group[0].channel[i].hpoint.hpoint = (1 << DUTY_BIT_DEPTH) - 1;
        LEDC.channel_group[0].channel[i].conf0.sig_out_en = 1;
        LEDC.channel_group[0].channel[i].conf1.duty_start = 1;
        ledc_cb_register(LEDC_LOW_SPEED_MODE, i, &callbacks, (void *)counting_sem);
    }

Change 2:

//ESP32 Working part to set Fade
        ESP_LOGE(LOG_FADE, "#");
        // Not currently fading, use current value, but remember to invert it (low hpoint is high brightness).
        start_brightness = ((1 << DUTY_BIT_DEPTH) - 1) - LEDC.channel_group[0].channel[channel].hpoint.hpoint;

        /////////////////////////////////////////
        /* Change log for ESP32S3 Module */

        ledc_fade_start(LEDC_LOW_SPEED_MODE,channel,LEDC_FADE_NO_WAIT); 
        // ledc_set_duty_with_hpoint(LEDC_LOW_SPEED_MODE, channel, start_brightness, LEDC.channel_group[0].channel[channel].hpoint.hpoint);
        ledc_set_fade_with_time(LEDC_LOW_SPEED_MODE,channel,(1 << DUTY_BIT_DEPTH) - 1,1000);  //after add DC dimming okay
        // ledc_update_duty(LEDC_LOW_SPEED_MODE, channel);

        ///////////////////////////////////////////

I have taken reference code: (https://github.com/masoncj/esp32-dimmer/tree/master)

Can you please tell me why i am not able to do dimming in AC supply.

CC-MASTER05, 2023-03-11:
Sorry for i didn't mention some details:

Compiling: idf.py build
Framework : ESP-IDF

songruo, 2023-03-23:
Hi @CC-MASTER05, I don't quite understand what is changed in "change 2". I took a look at the reference code you mentioned, it looks like it didn't use the LEDC driver APIs to do fading, but implemented the software fading in the application. Now, you would like to use the hardware fading with the driver APIs?

Not sure if it is a copy & pasta problem, but in your code, the calling order of `ledc_fade_start` and `ledc_set_fade_with_time` is wrong. You should first set up a fade with `ledc_set_fade_with_time`, and then call `ledc_fade_start` to start the fade :)

CC-MASTER05, 2023-03-23:
@songruo thanks for the answer but acutely when i am changing and load into my esp32s3 it not working but when this same code i implement in ESP32 at a time it working fine.

have you any idea?

songruo, 2023-03-24:
@CC-MASTER05 The code you posted is not very clear for me. Can you show how you configure the fading on esp32 and esp32s3 explicitly?

CC-MASTER05, 2023-04-08:
Hey @songruo  i am using LEDC library and Zero-cross detect circuit for AC dimming.

Once , i try to change duty cycle at a time not sync zero-cross detect signal and output PWM signal so we are not able to manage same time AC signal calipering.

Please find below my code.

Summary of Code:

- I am taking once UART based serial input and accordingly those input set and update duty cycle in between taking zero-cross signal interrupt using ISR.
- In ISR used semaphore binary
- Implement one timer reset in Zero-cross signal detect at time and made one small delay

**NOTE:** No flickering happening at any point just issue with signal sync.

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/ledc.h"
#include "soc/ledc_struct.h"
#include "soc/gpio_struct.h"
#include "driver/gpio.h"
#include "driver/uart.h"
#include "freertos/semphr.h"
#include "esp_log.h"
#include "stdio.h"
#include "strings.h"
#include <sys/time.h>

#define LEDC_CH_NUM 0
#define LEDC_CH_FREQ 50
#define LEDC_CH_RESOLUTION LEDC_TIMER_13_BIT
#define AC_ZERO_CROSS_GPIO GPIO_NUM_4
#define AC_DIMMER_GPIO GPIO_NUM_12
#define UART_PORT_NUM UART_NUM_0
#define TIMER LEDC_TIMER_0
#define BUF_SIZE (1024)

int ledc_duty = 0;
int reg_sel = 0;
static QueueHandle_t uart0_queue;
static volatile SemaphoreHandle_t zero_cross_semaphore = NULL;
uart_event_t event;

static ledc_channel_config_t ledc_channel = {
    .gpio_num = AC_DIMMER_GPIO,
    .speed_mode = LEDC_LOW_SPEED_MODE,
    .channel = LEDC_CH_NUM,
    .intr_type = LEDC_INTR_FADE_END,
    .timer_sel = TIMER,
    // .gpio_num = AC_ZERO_CROSS_GPIO,
    .duty = 0,
    .hpoint = 0};

static void IRAM_ATTR ac_zero_cross_isr_handler(void *arg)
{

    static BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    xSemaphoreGiveFromISR(zero_cross_semaphore, &xHigherPriorityTaskWoken);
}

static void ac_dimmer_task(void *arg)
{
    while (true)
    {
        if (xSemaphoreTake(zero_cross_semaphore, portMAX_DELAY) == pdTRUE)
        {
            if (GPIO.in & (1 << AC_ZERO_CROSS_GPIO))
            {
                 LEDC.timer_group[0].timer[0].conf.rst = 1;
[... 130 lines cut ...]

    // Configure UART
    static const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_DEFAULT};

    ESP_ERROR_CHECK(uart_param_config(UART_PORT_NUM, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(UART_PORT_NUM, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
    uart_driver_install(UART_PORT_NUM, BUF_SIZE * 2, BUF_SIZE * 2, 10, &uart0_queue, 0);

    // Create task to handle UART input
    xTaskCreate(uart_task, "UART Task", 2048, NULL, 10, NULL);

    xTaskCreate(ac_dimmer_task, "AC Dimmer Task", 2048, NULL, 10, NULL);

    while (1)
    {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
 ```

songruo, 2023-05-17:
@CC-MASTER05 I have tried your code. It looks fine to me, the LEDC signal is working, UART is working, and zero crossing IO can trigger the interrupt properly. I am not sure what is your issue now, are you trying to say that the duty change is not synced with the zero-crossing moment? If so, two suggestions:
1. You don't need to call `ledc_set_duty_and_update` in the `uart_task`, right? `ledc_set_duty_and_update` should only be called in the `ac_dimmer_task` to make it synchronize with zero-crossing, instead of UART event.
2. Is your `AC_ZERO_CROSS_GPIO` pin floating when you are testing? A floating pin can trigger the interrupt randomly. You should either connect the pin to external circuit or enable internal pull-up / pull-down resistors in `gpio_config_t`.

CC-MASTER05, 2023-05-17:
Thanks for the input i have tested and working fine with your suggestion.

closed 2023-05-17 by CC-MASTER05

noushad2, 2025-09-02:
Hello,

Can I use this source code in our application that involves an ESP32 and the RobotDyn AC dimmer module ([https://www.amazon.com/GENUINE-RobotDYN-Programmable-Light-Dimmer/dp/B07KDNMTSF?&tag=robotdyn11-20](https://www.amazon.com/GENUINE-RobotDYN-Programmable-Light-Dimmer/dp/B07KDNMTSF?tag=robotdyn11-20&utm_source=chatgpt.com)
)?