espressif/esp-idf #10971: Required build files for debugging

ligantx, 2023-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using a esp-idf docker image to build my project, but i want to do the debugging locally and transfer only the necessary files from the docker container (i want to avoid transferring all the build folder).

According to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#command-line) i need the `build/blink.elf` file. 
Do i need any more files from the build folder (because according to the instructions, i must be **inside** the project folder to run the debugging) and if yes:
1. will the `build/required_files` work if the `idf.py build` ran in a different machine (i.e.  docker container)?
2. which specific files do i need?

Alvin1Zhang, 2023-03-13:
Thanks for reporting.

dobairoland, 2023-03-13:
Yes, you need only the ELF file for debugging.

> i must be inside the project folder

This is just so you could debug while seeing your own (uncompiled) code while debugging. Without this you still can debug but you will see assembler instruction.

> will the build/required_files work if the idf.py build ran in a different machine (i.e. docker container)?

I'm sorry, I don't understand this question.

igrr, 2023-03-13:
> will the build/required_files work if the idf.py build ran in a different machine (i.e. docker container)?

I think this might be related to the issue of source file paths being embedded into the ELF file.

When the program is built, debug information added into the ELF file will normally contain the absolute paths of source files. If you move the ELF file from the container to the host, the absolute paths of source files will likely be different. You can address this using `gdb set substitute-path` command, mapping the "container paths" to "host paths" so that GDB can find the source files. Please refer to GDB manual regarding this part.

ligantx, 2023-03-16:
Update:

I setup the whole thing with [Microsoft C/C++ extension](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-microsoft-cc-extension-to-debug) and all i had to do was to add in `launch.json` this:
```
"sourceFileMap": {
          "/project": "${workspaceFolder}"
}
```
and use the `.elf` file and the workspace folder.

closed 2023-03-16 by dobairoland