espressif/esp-idf #10972: FreeRTOS software timers strange behawior

diablo1281, 2023-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Similar issue #8234

Custom board with ESP32-WROOM-32E.

I am running two software timers in FreeRTOS, both have a period of 1 second (100 ticks). They read data from sensors.
At first, both timers go as they should. After 32 seconds, one of the timers (always the same) starts to get double callbacks (with the same expiry time) and misses his next one (like changing the period to 2 seconds). After 48 seconds from the boot, that timer dies. The second one works without a problem and continues to work after the first dies.

Happens every time almost at the same time. Changing the period of the first timer results in the same results but quicker. When only one timer works the problem doesn't occur.

Code: [https://gist.github.com/diablo1281/12a43050ed0614791835efeb25d8d202](url)

Below screen from the console (colors are only for ease of reading).
Each message contains: Timer name [time from boot] [period in ticks] [expiry time in ticks] [current tick count]
[image]

nopnop2002, 2023-04-03:
About xQueueSend from here:

https://github.com/espressif/esp-idf/issues/10972

This function must not be called from an interrupt service routine. See xQueueSendFromISR() for an alternative which may be used in an ISR.

diablo1281, 2023-04-03:
Hi, @nopnop2002 thank you for the replay. I didn't consider timer callbacks as ISR routines. Anyway, after applying changes to callback functions, the problem still persists. 

I got one more observation, after a while second timer started also to have doubled calls and after a few seconds both timers stopped getting calls (but timers was never stopped by software).

nopnop2002, 2023-04-04:
I used two FreeRTOS tasks with very simple code and it works fine.
```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/queue.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "esp_system.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

// Timer callback
static void timerCallback1(TimerHandle_t xTimer)
{
        QueueHandle_t queue = ( QueueHandle_t ) pvTimerGetTimerID( xTimer );
        float value = 0;
        xQueueSendFromISR(queue, &value, NULL);
}

static void timerCallback2(TimerHandle_t xTimer)
{
        QueueHandle_t queue = ( QueueHandle_t ) pvTimerGetTimerID( xTimer );
        float value = 0;
        xQueueSendFromISR(queue, &value, NULL);
}

void task(void *pvParameters)
{
        QueueHandle_t queue = ( QueueHandle_t ) pvParameters;
        ESP_LOGI(pcTaskGetName(0), "Start");
        float value;
        while(1) {
                if(xQueueReceive(queue, &value, portMAX_DELAY)) {
                        ESP_LOGI(pcTaskGetName(0), "xQueueReceive");
                }
        }
        ESP_LOGI(pcTaskGetName(0), "Finish");
        vTaskDelete( NULL );
}

void app_main(void)
{
        // Create Queue
        QueueHandle_t xQueue1 = xQueueCreate(10, sizeof(float));
        configASSERT( xQueue1 );
        QueueHandle_t xQueue2 = xQueueCreate(10, sizeof(float));
        configASSERT( xQueue2 );

        // Start Task
        // Pass QueueHandle using pvParameters
        TaskHandle_t pxCreatedTask1;
        xTaskCreate(task, "TASK1", 1024*2, xQueue1, 2, &pxCreatedTask1);
        TaskHandle_t pxCreatedTask2;
        xTaskCreate(task, "TASK2", 1024*2, xQueue2, 2, &pxCreatedTask2);

        // Create Timer
        // Pass QueueHandle using pvTimerID
        int TimerPeriod = 1 * 1000;
        TimerHandle_t timerHandle1 = xTimerCreate("Trigger", TimerPeriod/portTICK_PERIOD_MS, pdTRUE, xQueue1, timerCallback1);
        configASSERT( timerHandle1 );
        TimerHandle_t timerHandle2 = xTimerCreate("Trigger", TimerPeriod/portTICK_PERIOD_MS, pdTRUE, xQueue2, timerCallback2);
        configASSERT( timerHandle2 );

        // Start Timer
        if (xTimerStart(timerHandle1, 0) != pdPASS) {
                ESP_LOGE(TAG, "Unable to start Timer1");
        }
        if (xTimerStart(timerHandle2, 0) != pdPASS) {
                ESP_LOGE(TAG, "Unable to start Timer2");
        }

        while(1) {
                vTaskDelay(100);
        }
}
```

diablo1281, 2023-04-04:
Your code runs flawlessly, but I can't find a significant difference between yours and my code.
Unfortunately, I can't share it publicly. I can share the whole project directly with you. I have no idea why it's behavior like this, I can't see any error for now (except previously pointed out by you).

nopnop2002, 2023-04-04:
I looked at your code and can't see the difference.

Code: https://gist.github.com/diablo1281/12a43050ed0614791835efeb25d8d202

diablo1281, 2023-04-04:
For now, I use a simple workaround (task making calls for all sensors one by one). Maybe during development, I found the bug, I let you know. Thanks for your support. If you have any ideas, don't hesitate to post them here, and I will check any potential solutions.

someburner, 2023-04-12:
Try adjusting the software timer priority in SDK config. A few things to note:

- FreeRTOS software timer callbacks must not block at all- Putting code to read temperature probably shouldn't be in a callback.
- If xQueueSend is called from a lower priority task it's likely to cause a context switch to a higher priority receiving task, so your FreeRTOS timer wont exit quickly. If that task does not block again quickly enough, likely to cause some problems.
- Even putting debugs/logs in the timer handler not advised, especially ESP_LOGx which requires a mutex internally.

Dazza0, 2023-11-21:
@diablo1281 It's unlikely to be a bug with FreeRTOS timers themselves. Any active timer is internally stored in one of two linked lists (`pxCurrentTimerList` and `pxOverflowTimerList`) where the active timers are sorted in timeout order. An active timer will remain on those lists until they are stopped or have timed out. Since only the FreeRTOS daemon task have access to those lists, it's unlikely that those lists are corrupted.

Since all the timer's you have created are auto-reloading, they will have never left those lists unless they were explicitly stopped somewhere. I suggest you double check your demo code's state machine logic and ensure that `xTimerStart()` and `xTimerStop()` calls are correct, especially the cases where...

- a timer is stopped via a `xTimerStop[FromISR]()` call and not restarted later with a `xTimerStart()` call
- a timer is repeatedly being restarted via `xTimerStart()` thus never reaching its timeout.

Other things that could help debug the issue:

- Using [GDB](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-guides/jtag-debugging/index.html), halt the application when it seems like the other timer's have stopped running and...
  - scan the `pxCurrentTimerList` to see what timer's are actually active (and their timeout values)
  - Use `info threads` and `bt` commands to check the current stack trace of sensor task and timer tasks
- Override the various `traceTIMER_...` macros (see `FreeRTOS.h` and `timers.c`) with debugging logs. This should give you a clearer picture as to what is going on with the FreeRTOS timers.

I'm closing this issue as I think it's unlikely to be a bug inside the FreeRTOS timers. But if you do find a bug, please feel free to reopen this issue.

closed 2023-11-21 by Dazza0