espressif/esp-idf #10973: target_compile_options seems to be ignored

albertcaba, 2023-03-12:
Answers checklist.
 I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
 I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
 I have searched the issue tracker for a similar issue and not found a similar issue.
General issue report
Similar issue https://github.com/espressif/esp-idf/issues/8234

Custom board with ESP32-WROOM-32E.

Version: v5.0.1

target_compile_options seems to not be making any effect. I have the following CMakeLists.txt inside the component:

idf_component_register(SRCS "nvs_manager.c"
                    INCLUDE_DIRS "include"
                    REQUIRES nvs_flash)

target_compile_options(${COMPONENT_LIB} PUBLIC -Wall -Werror=all)

Inside the file nvs_manager.c, I have an unused variable inside a function, so I would expect a compilation error, due to Werror compilation flag. 

Any idea what could be going on? Thanks!

igrr, 2023-03-12:
Could you please check compile_commands.json file in the build directory and find the compiler command line for this file?

You can then compare the command line with and without the call to target_compile_options in CMakeLists and see if anything changes.

albertcaba, 2023-03-12:
Thank you for your answer, I tried compiling with and without target_compile_options  and I can indeed see the differences that it seems to add the -Wall and -Werror , attached the files in case it is needed:
[buildcompile_commands_with_compilations_flags.txt](https://github.com/espressif/esp-idf/files/10950491/buildcompile_commands_with_compilations_flags.txt)
[buildcompile_commands_without_compilations_flags.txt](https://github.com/espressif/esp-idf/files/10950492/buildcompile_commands_without_compilations_flags.txt)

Why is the compilation not failing then? Shouldnt it fail with a non used local variable? Thanks!

albertcaba, 2023-03-12:
Here is the warning that is not translated into an error as I would expect:

C:/Users/Startup/Documents/workspace/somiatruites_espidf/components/nvs_manager/nvs_manager.c:39:10: warning: unused variable 'waska' [-Wunused-variable]
   39 |     bool waska;
      |          ^~~~~

igrr, 2023-03-13:
Thanks for providing the files. I think I understand the issue you are facing:

1. ESP-IDF build system enables most warnings, but then disables errors for certain warnings: https://github.com/espressif/esp-idf/blob/47852846d3e89580ef4da28210b6ffc1cb5eaa9d/tools/cmake/build.cmake#L101-L109
2. Additional flags you pass using `target_compile_options` are appended after the flags set by IDF build system.
3. However, CMake [de-duplicates compiler arguments](https://cmake.org/cmake/help/latest/command/target_compile_options.html#option-de-duplication) ([bug report](https://gitlab.kitware.com/cmake/cmake/-/issues/15826)). Because of that additional `-Werror=all` argument you have passed is not added to the compiler command line.

I think the simplest solution to this is to explicitly re-enable the warning which was disabled by the IDF build system, for example by calling 
```cmake
target_compile_options(${COMPONENT_LIB} PRIVATE -Werror=unused-variable)
```

albertcaba, 2023-03-13:
Yes you are completely right thanks, adding the suggested solution makes the compilation error pop up thanks!

closed 2023-03-13 by albertcaba