espressif/esp-idf #10974: When debugging with jtag, all variable values are displayed as when entering the rtos task

xiaohuihere, 2023-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Development Kit

ESP-WROVER-KIT-VE

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

FT2232

### OpenOCD version

v0.11.0-esp32-20221026

### Operating System

window10

### Using an IDE ?

espressif IDF

### OpenOCD command line

C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -f C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts\board\esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

20000

### ESP-IDF version

v4.4.3

### Problem Description

# When debugging with jtag, all variable values are displayed as <unavailable> when entering the rtos task
While working on a project, I wanted to try to go into one of the rtos tasks to see the value of a variable. However, I found that all the variables in the task showed <unavailable>.
The following is a screenshot of the scenario.
[image]

### Debug Logs

```shell
openOCD:
PS C:\Users\22380> C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -f C:\Users\22380\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts\board\esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 26000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32.cpu0] Target halted, PC=0x40168578, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.drom' size 100 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu0.drom' size 100 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu1.irom' size 640 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 99 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 638 KB
Info : Using flash bank 'esp32.cpu1.drom' size 100 KB
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Info : [esp32.cpu0] Target halted, PC=0x400D9231, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40168578, debug_reason=00000000
Info : [esp32.cpu0] Target halted, PC=0x400DA0CF, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400D4F46, debug_reason=00000000

vscode output:
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-02 20:43:14,465 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-02-02 20:43:14,469 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-02-02 20:43:25,132 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-02 20:45:01,071 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-02-02 20:45:01,075 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

vscode debugconsole:
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Debug Adapter initialized
Debug Adapter is running
8
WARNING! This feature can't update UI after execution.
```

### Expected behavior

It should have shown the value of the variable in the left display bar, like the int variable a, which should have had a value of 1 or 0

### Screenshots

task:
[image]

Create a task：
[image]

Austin-Lindquist, 2023-12-16:
I had this same issue, weirdly enough if you switch to Debug (-Og) in sdkconfig it works. Debug without optimization (-O0) option doesn't work. The latter seems to unset CONFIG_OPTIMIZATION_LEVEL_DEBUG and
CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG defines

I didn't dig too far into it but I'm guessing there's an issue somewhere down the line switching on debug option

KaeLL, 2023-12-19:
https://gcc.gnu.org/onlinedocs/gcc-13.2.0/gcc/Optimize-Options.html#index-Og