espressif/esp-idf #10977: When I start wifi provisioning, I get an assert. (assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

EfanovIvan, 2023-03-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-235-gfee50c208a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Waiting for http server to start processing provisioning requests.

### What is the actual behavior?

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

### Steps to reproduce.

1. initialization

  ```
wifi_prov_mgr_config_t prvConfig = {
          .scheme               = wifi_prov_scheme_softap,
          .scheme_event_handler = WIFI_PROV_EVENT_HANDLER_NONE,
          .app_event_handler    = WIFI_PROV_EVENT_HANDLER_NONE,
      };

    if (wifi_prov_mgr_init(prvConfig) != ESP_OK)
    {
        throw std::runtime_error{ "WiFi provisioning configeration failed!" };
    }
```

2. launch
```
bool Wifi::StartProvisioning() noexcept
{
    std::array<uint8_t, 6> mac;
    if (esp_wifi_get_mac(WIFI_IF_STA, mac.data()) != ESP_OK)
    {
        ESP_LOGE(TAG, "Error getting MAC address!");
        return false;
    }
    std::string service_name{ "PROV_" };
    service_name += hexstrconv::ToHexStr(mac);

    if (wifi_prov_mgr_start_provisioning(WIFI_PROV_SECURITY_1, CONFIG_POSSESSION_PIN,
                                         service_name.c_str(), nullptr) != ESP_OK)
    {
        ESP_LOGE(TAG, "Start provisioning ERROR!");
        return false;
    }
    return true;
}
```

### Debug Logs.

```plain
I (56862) wifi:mode : sta (f4:12:fa:f8:f5:b8) + softAP (f4:12:fa:f8:f5:b9)
I (56862) wifi:Total power save buffer number: 4
I (56862) wifi:Init max length of beacon: 752/752
I (56862) wifi:Init max length of beacon: 752/752
I (57652) wifi:Total power save buffer number: 4
W (57652) wifi_prov_scheme_softap: Error adding mDNS service! Check if mDNS is running

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace: 0x40376282:0x3fced8b0 0x4037e7d9:0x3fced8d0 0x40388d81:0x3fced8f0 0x4037f9ee:0x3fceda10 0x4037fb98:0x3fceda50 0x42128905:0x3fceda70 0x421289da:0x3fcedab0 0x42127f95:0x3fcedae0 0x420370af:0x3fcedb00 0x42038021:0x3fcedb20 0x42033005:0x3fcedc80 0x4203028a:0x3fcedce0 0x4200919e:0x3fcedd00 0x4200a999:0x3fcedd20 0x4200aa19:0x3fcedd40 0x4200abf2:0x3fcedd60 0x4201b283:0x3fcedd90 0x42128c7d:0x3fceddc0 0x421286ca:0x3fceddf0 0x421287a5:0x3fcede30 0x40381d59:0x3fcede50
0x40376282: panic_abort at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_system/panic.c:408

0x4037e7d9: esp_system_abort at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_system/esp_system.c:137

0x40388d81: __assert_func at /home/dev/esp/esp-idf-4.4/esp-idf/components/newlib/assert.c:85

0x4037f9ee: xQueueSemaphoreTake at /home/dev/esp/esp-idf-4.4/esp-idf/components/freertos/queue.c:1549 (discriminator 1)

0x4037fb98: xQueueTakeMutexRecursive at /home/dev/esp/esp-idf-4.4/esp-idf/components/freertos/queue.c:731

0x42128905: esp_event_handler_register_with_internal at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/esp_event.c:713

0x421289da: esp_event_handler_register_with at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/esp_event.c:761

0x42127f95: esp_event_handler_register at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/default_event_loop.c:32

0x420370af: wifi_prov_mgr_start_service at /home/dev/esp/esp-idf-4.4/esp-idf/components/wifi_provisioning/src/manager.c:370

0x42038021: wifi_prov_mgr_start_provisioning at /home/dev/esp/esp-idf-4.4/esp-idf/components/wifi_provisioning/src/manager.c:1522

0x42033005: net::Wifi::StartProvisioning() at /home/dev/work/a138-ble-gateway/firmware/esp32/components/net/wifi.cpp:236

0x4203028a: net::Manager::StartWifiProvisioning() at /home/dev/work/a138-ble-gateway/firmware/esp32/components/net/net-manager.cpp:46

0x4200919e: std::_Function_handler<void (), app_main::{lambda()#16}>::_M_invoke(std::_Any_data const&) at /home/dev/work/a138-ble-gateway/firmware/esp32/main/main.cpp:296
 (inlined by) _M_invoke at /home/dev/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297

0x4200a999: std::function<void ()>::operator()() const at /home/dev/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687

0x4200aa19: Button::ChangeState(bool) at /home/dev/work/a138-ble-gateway/firmware/esp32/components/button/include/button.hpp:38
 (inlined by) operator() at /home/dev/work/a138-ble-gateway/firmware/esp32/main/main.cpp:303
 (inlined by) _M_invoke at /home/dev/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297

0x4200abf2: std::function<void (bool)>::operator()(bool) const at /home/dev/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687

0x4201b283: GpioIsr::EventHandler(void*, char const*, int, void*) at /home/dev/work/a138-ble-gateway/firmware/esp32/components/gpio-isr/include/gpio-isr/gpio-isr.hpp:103

0x42128c7d: handler_execute at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/esp_event.c:145

0x421286ca: esp_event_loop_run at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/esp_event.c:590 (discriminator 3)

0x421287a5: esp_event_loop_run_task at /home/dev/esp/esp-idf-4.4/esp-idf/components/esp_event/esp_event.c:115 (discriminator 15)

0x40381d59: vPortTaskWrapper at /home/dev/esp/esp-idf-4.4/esp-idf/components/freertos/port/xtensa/port.c:142
```

### More Information.

On this commit everything works **8153bfe4125e6a608abccf1561fd10285016c90a**

AxelLin, 2023-05-04:
You can try 'git bisect' to find the bad commit. Suppose 8153bfe is good, fee50c208a is bad.

laukik-hase, 2023-05-08:
Hello, @EfanovIvan!

Thank you for sending the issue report. I was investigating this issue but unfortunately could not reproduce it with the `wifi_prov_mgr` example on the ESP-IDF `v4.4.4-235-gfee50c208a` commit.

Could you please attach the following -

- A minimal example that reproduces the issue
- The `sdkconfig` file for your example
- Debug logs (with `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`)

You could also try to find the exact commit your app started failing at using `git bisect` as suggested in the previous comment.

mahavirj, 2023-06-07:
Closing due to lack of update, please drop a comment if you are still facing the issue.

closed 2023-06-07 by mahavirj

EfanovIvan, 2023-06-07:
Thanks to all. I switched to version 5.

AxelLin, 2023-06-07:
@EfanovIvan 
If you still can reproduce with latest v4.4 branch, this still needs fix anyway.

elitno, 2023-10-27:
To reproduce the issue simply enable this flag CONFIG_SW_COEXIST_ENABLE = 1
since the exception is thrown here:

static void coex_schm_status_bit_clear_wrapper(uint32_t type, uint32_t status)
{
#if CONFIG_SW_COEXIST_ENABLE
    coex_schm_status_bit_clear(type, status);
#endif
}

vortex314, 2025-01-04:
Encountered same issue 
```

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace: 0x400817a6:0x3ffc19c0 0x40088735:0x3ffc19e0 0x4008f2c5:0x3ffc1a00 0x40089886:0x3ffc1b20 0x40082ef4:0x3ffc1b60 0x40082f2a:0x3ffc1b80 0x400ddc9d:0x3ffc1ba0 0x400de1fe:0x3ffc1bc0 0x400de24d:0x3ffc1be0 0x400da09a:0x3ffc1c00 0x400d8f7e:0x3ffc1df0 0x400d7aba:0x3ffc1ea0 0x400d7c4b:0x3ffc1f30 0x400d82f9:0x3ffc1fb0 0x400d88e9:0x3ffc2140 0x400d23f1:0x3ffc2200 0x4008ba21:0x3ffc2220

  #0  0x400817a6:0x3ffc19c0 in panic_abort at /home/lieven/.platformio/packages/framework-espidf/components/esp_system/panic.c:408
  #1  0x40088735:0x3ffc19e0 in esp_system_abort at /home/lieven/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
  #2  0x4008f2c5:0x3ffc1a00 in __assert_func at /home/lieven/.platformio/packages/framework-espidf/components/newlib/assert.c:85
  #3  0x40089886:0x3ffc1b20 in xQueueSemaphoreTake at /home/lieven/.platformio/packages/framework-espidf/components/freertos/queue.c:1549 (discriminator 1)
  #4  0x40082ef4:0x3ffc1b60 in pthread_mutex_lock_internal at /home/lieven/.platformio/packages/framework-espidf/components/pthread/pthread.c:620
  #5  0x40082f2a:0x3ffc1b80 in pthread_mutex_lock at /home/lieven/.platformio/packages/framework-espidf/components/pthread/pthread.c:650
  #6  0x400ddc9d:0x3ffc1ba0 in _z_mutex_lock at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/system/espidf/system.c:139
  #7  0x400de1fe:0x3ffc1bc0 in _z_transport_tx_mutex_lock at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/include/zenoh-pico/transport/transport.h:191
      (inlined by) _z_transport_tx_send_n_msg at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/transport/common/tx.c:222
  #8  0x400de24d:0x3ffc1be0 in _z_send_n_msg at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/transport/common/tx.c:404
  #9  0x400da09a:0x3ffc1c00 in _z_write at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/net/primitives.c:240
  #10 0x400d8f7e:0x3ffc1df0 in z_put at .pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/api/api.c:883
  #11 0x400d7aba:0x3ffc1ea0 in ZenohActor::zenoh_publish_binary(char const*, std::vector<unsigned char, std::allocator<unsigned char> > const&) at src/zenoh_actor.cpp:260
  #12 0x400d7c4b:0x3ffc1f30 in ZenohActor::publish_topic_value(char const*, Serializable&) at src/zenoh_actor.cpp:275
  #13 0x400d82f9:0x3ffc1fb0 in ZenohActor::publish_slow() at src/zenoh_actor.cpp:225
  #14 0x400d88e9:0x3ffc2140 in ZenohActor::run() at src/zenoh_actor.cpp:29
  #15 0x400d23f1:0x3ffc2200 in app_main::{lambda(void*)#3}::_FUN(void*) at src/main.cpp:88
      (inlined by) _FUN at src/main.cpp:89
  #16 0x4008ba21:0x3ffc2220 in vPortTaskWrapper at /home/lieven/.platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c:142

```
Using Zenoh pico at high load tests sending data over WIFI
https://github.com/vortex314/zenoh-projects/tree/main/zenoh-lib