espressif/esp-idf #10981: enc28j60 example component has compile error

farbodpm, 2023-03-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I added esp_eth_enc28j60 and I face this error .

```
[1/11] Building C object esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_phy_enc28j60.c.obj
  FAILED: esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_phy_enc28j60.c.obj 
  /root/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/project/target/xtensa-esp32-espidf/release/build/esp-idf-sys-321db7f4ba74d5d5/out/build/config -I/project/components/esp_eth_enc28j60/include -I/root/.espressif/esp-idf/v4.4.2/components/newlib/platform_include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include/esp_additions/freertos -I/root/.espressif/esp-idf/v4.4.2/components/freertos/port/xtensa/include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include/esp_additions -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include/soc -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include/soc/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/port/esp32/. -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/port/esp32/private_include -I/root/.espressif/esp-idf/v4.4.2/components/heap/include -I/root/.espressif/esp-idf/v4.4.2/components/log/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/include/apps -I/root/.espressif/esp-idf/v4.4.2/components/lwip/include/apps/sntp -I/root/.espressif/esp-idf/v4.4.2/components/lwip/lwip/src/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/port/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/port/esp32/include/arch -I/root/.espressif/esp-idf/v4.4.2/components/soc/include -I/root/.espressif/esp-idf/v4.4.2/components/soc/esp32/. -I/root/.espressif/esp-idf/v4.4.2/components/soc/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/platform_port/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/include/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_common/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/port/soc -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/port/public_compat -I/root/.espressif/esp-idf/v4.4.2/components/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/xtensa/include -I/root/.espressif/esp-idf/v4.4.2/components/xtensa/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/driver/include -I/root/.espressif/esp-idf/v4.4.2/components/driver/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_pm/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_ringbuf/include -I/root/.espressif/esp-idf/v4.4.2/components/efuse/include -I/root/.espressif/esp-idf/v4.4.2/components/efuse/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/vfs/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_wifi/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_event/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_netif/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_eth/include -I/root/.espressif/esp-idf/v4.4.2/components/tcpip_adapter/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_phy/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_phy/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_ipc/include -I/root/.espressif/esp-idf/v4.4.2/components/app_trace/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_timer/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=/project/target/xtensa-esp32-espidf/release/build/esp-idf-sys-321db7f4ba74d5d5/out=. -fmacro-prefix-map=/root/.espressif/esp-idf/v4.4.2=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.2-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_phy_enc28j60.c.obj -MF esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_phy_enc28j60.c.obj.d -o esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_phy_enc28j60.c.obj -c /project/components/esp_eth_enc28j60/esp_eth_phy_enc28j60.c
  /project/components/esp_eth_enc28j60/esp_eth_phy_enc28j60.c:11:10: fatal error: eth_phy_802_3_regs.h: No such file or directory
   #include "eth_phy_802_3_regs.h"
            ^~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  [2/11] Building C object esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_mac_enc28j60.c.obj
  FAILED: esp-idf/esp_eth_enc28j60/CMakeFiles/__idf_esp_eth_enc28j60.dir/esp_eth_mac_enc28j60.c.obj 
  /root/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/project/target/xtensa-esp32-espidf/release/build/esp-idf-sys-321db7f4ba74d5d5/out/build/config -I/project/components/esp_eth_enc28j60/include -I/root/.espressif/esp-idf/v4.4.2/components/newlib/platform_include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include/esp_additions/freertos -I/root/.espressif/esp-idf/v4.4.2/components/freertos/port/xtensa/include -I/root/.espressif/esp-idf/v4.4.2/components/freertos/include/esp_additions -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include/soc -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/include/soc/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/port/esp32/. -I/root/.espressif/esp-idf/v4.4.2/components/esp_hw_support/port/esp32/private_include -I/root/.espressif/esp-idf/v4.4.2/components/heap/include -I/root/.espressif/esp-idf/v4.4.2/components/log/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/include/apps -I/root/.espressif/esp-idf/v4.4.2/components/lwip/include/apps/sntp -I/root/.espressif/esp-idf/v4.4.2/components/lwip/lwip/src/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/port/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/lwip/port/esp32/include/arch -I/root/.espressif/esp-idf/v4.4.2/components/soc/include -I/root/.espressif/esp-idf/v4.4.2/components/soc/esp32/. -I/root/.espressif/esp-idf/v4.4.2/components/soc/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/include -I/root/.espressif/esp-idf/v4.4.2/components/hal/platform_port/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/include/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_rom/esp32 -I/root/.espressif/esp-idf/v4.4.2/components/esp_common/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/port/soc -I/root/.espressif/esp-idf/v4.4.2/components/esp_system/port/public_compat -I/root/.espressif/esp-idf/v4.4.2/components/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/xtensa/include -I/root/.espressif/esp-idf/v4.4.2/components/xtensa/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/driver/include -I/root/.espressif/esp-idf/v4.4.2/components/driver/esp32/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_pm/include -I/root/.espressif/esp-idf/v4.4.2/components/esp_ringbuf/include -I/root/.espressif/esp-idf/v4.4.2/components/efuse/include -I/root/.espressif/esp-idf/v4.4.2/compone [... cut ...]
  /project/components/esp_eth_enc28j60/esp_eth_mac_enc28j60.c: In function 'esp_eth_mac_new_enc28j60':
  /project/components/esp_eth_enc28j60/esp_eth_mac_enc28j60.c:1105:20: error: implicit declaration of function 'esp_cpu_get_core_id'; did you mean 'esp_cpu_get_ccount'? [-Werror=implicit-function-declaration]
           core_num = esp_cpu_get_core_id();
                      ^~~~~~~~~~~~~~~~~~~
                      esp_cpu_get_ccount
  cc1: some warnings being treated as errors
```
it seemes that some functions places has changed.

kostaond, 2023-03-17:
Hi @farbodpm, thank you for reporting potential issue. However, I need more information to be able to investigate.

* What ESP-IDF version do you use?
*  What do you mean by "I added esp_eth_enc28j60 "? Added where? Added how?
* Are you able to build ENC28J60 example?

kostaond, 2023-04-03:
@farbodpm could you have a look at my questions above and provide more information so I could help you to resolve the issue? Otherwise, I will have close the issue for lack of feedback. Thank you for understanding.

AxelLin, 2023-05-02:
@farbodpm 
esp-2021r2-patch3-8.4.0 implies you are using esp-idf-v4.3 branch.
However, esp_cpu_get_core_id() is added during v5.0 -dev, it does not exist in esp-idf-v4.x branches.

kostaond, 2023-05-12:
I am closing this issue due to lack of feedback. @farbodpm, it seems you might mixed the newer version of ENC28J60 example with older version of ESP-IDF as suggested by @AxelLin. Feel free to reopen the issue but please provide more information.

closed 2023-05-12 by kostaond