espressif/esp-idf #10983: [v5.0.1][Build] Console Example - dangerous relocation: l32r: literal target out of range (try using text-section-literals): .rtc.entry.literal

chipweinberger, 2023-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-1711-g3088f5c3a6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Console example should build. 

### What is the actual behavior?

```
esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj): in function `esp_wake_stub_entry':
/esp-idf/components/esp_hw_support/sleep_modes.c:230:(.rtc.entry.text+0x3): dangerous relocation: l32r: literal target out of range (try using text-section-literals): .rtc.entry.literal
collect2: fatal error: ld terminated with signal 11 [Segmentation fault: 11]
compilation terminated.
```

### Steps to reproduce.

1. checkout latest release/v5.0 branch
2. copy basic console example
3. run 'idf.py' build

### Build or installation Logs.

```plain
charlesweinberger@Chips-Macbook-Pro-2 basic % idf.py build                                                   
Executing action: all (aliases: build)
Running cmake in directory /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.32.0 (Apple Git-132)") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
^Ccmake failed with exit code -2, output of the command is in the /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/log/idf_py_stderr_output_49941 and /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/log/idf_py_stdout_output_49941
charlesweinberger@Chips-Macbook-Pro-2 basic % idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
[... 111 lines cut ...]
-- Build files have been written to: /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/bootloader
[105/106] Generating binary image from built executableesptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/bootloader/bootloader.bin
[106/106] cd /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/bu...inberger/Desktop/esp-idf-v5/examples/basic/build/bootloader/bootloader.binBootloader binary size 0x4970 bytes. 0x3690 bytes (43%) free.
[896/898] Linking CXX executable console.elfFAILED: console.elf lf_src_esp32s3.c.obj
: && /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  CMakeFiles/console.elf.dir/project_elf_src_esp32s3.c.obj -o console.elf  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/cmd_nvs/libcmd_nvs.a  esp-idf/cmd_system/libcmd_system.a  esp-idf/cmd_wifi/libcmd_wifi.a  -Wl,--cref  -Wl,--defsym=IDF_TARGET_ESP32S3=0  -Wl,--Map="/Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/console.map"  -fno-rtti  -fno-lto  -Wl,--gc-sections  -Wl,--warn-common  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/app_trace/libapp_trace.a  -lgcov  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/console/libconsole.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_ad [... cut ...] )
collect2: fatal error: ld terminated with signal 11 [Segmentation fault: 11]
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/log/idf_py_stderr_output_50432 and /Users/charlesweinberger/Desktop/esp-idf-v5/examples/basic/build/log/idf_py_stdout_output_50432
```

### More Information.

Commenting out code related to 'esp_wake_stub_entry' fixes the issue.

chipweinberger, 2023-03-14:
Checking out v5.0.1 git tag (https://github.com/espressif/esp-idf/commit/a4afa44435ef4488d018399e1de50ad2ee964be8) fixed the issue. It must be caused by a recent commit.

chipweinberger, 2023-03-14:
It might only reproduce on my esp-idf branch for "stack mirroring". https://github.com/chipweinberger/esp-idf/commit/3088f5c3a6a5d557d90d466d10b66be73d744592

But it reproduces even with `CONFIG_COMPILER_STACK_MIRROR` disabled, which means all my changes are `#ifdef` out.

ESP-Marius, 2023-03-15:
Can you double check that you are not mixing toolchains etc between releases? 

E.g. the build you linked says `v5.0-beta1`, while the toolchain printed says `esp-2022r1-11.2.0` which doesnt match the commit.

Try a clean terminal + `install.sh` + `export.sh` + full rebuild and see if that solves the issue.

chipweinberger, 2023-03-15:
Thanks for the suggestion and good eye for spotting that. I'll redoing it tomorrow or so.

chipweinberger, 2023-04-18:
Closing. I have yet to hit this issue again.

I imagine you are correct about mixing toolchain and release.

closed 2023-04-18 by chipweinberger