espressif/esp-idf #10989: Cannot Find Host: esp_https_ota

audreylim, 2023-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDE**: PlatformIO
**Dependencies**: Platform espressif32 @ 6.1.0, framework-espidf @ 3.50001.0
**esp-idf**: v5.0.1
**OS**: Windows
**Board** : ESP32 Wrover

### Issue

Unable to connect to https server on WiFi network

### Steps to Reproduce

I'm running this example: https://github.com/maxgerhardt/pio-espidf-simple-ota-example
The firmware is built and uploaded via PlatformIO, and the python3 binary server built with Command Prompt, Windows.

### Expected Behavior

ESP32 should have downloaded binary from python server, via my home WiFi.

### Actual Behavior

```
E (5907) esp-tls: [sock=54] connect() error: Host is unreachable
```

Network: Home WiFi
Endpoint: simple-ota-example binary server
```
I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://172.20.10.3:8070/hello-world.bin
E (5907) esp-tls: [sock=54] connect() error: Host is unreachable
E (5907) esp-tls: Failed to open new connection
E (5917) transport_base: Failed to open a new connection
E (5917) HTTP_CLIENT: Connection failed, sock < 0
E (5927) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5937) esp_https_ota: Failed to establish HTTP connection
E (5937) simple_ota_example: Firmware upgrade failed
I (6247) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (6247) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
```

Replaced the server IP with https://example.com

Network: Home WiFi
Endpoint: https://example.com
```
I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://example.com
E (5907) esp-tls: couldn't get hostname for :example.com: getaddrinfo() returns 202, addrinfo=0x0
E (5907) esp-tls: Failed to open new connection
E (5917) transport_base: Failed to open a new connection
E (5917) HTTP_CLIENT: Connection failed, sock < 0
E (5927) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5937) esp_https_ota: Failed to establish HTTP connection
E (5937) simple_ota_example: Firmware upgrade failed
```

Replaced the server IP with http://example.com and got this:

Network: Home WiFi
Endpoint: http://example.com
```
I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from http://example.com
I (6727) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (6727) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
I (7177) HTTP_CLIENT: Body received in fetch header state, 0x3ffb8e8c, 156
I (7177) esp_https_ota: Starting OTA...
I (7177) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
E (7187) esp_https_ota: Mismatch chip id, expected 0, found 27757
E (7197) simple_ota_example: Firmware upgrade failed
```

Switched to my iPhone hotspot on the ESP and PC, keeping the server IP https://example.com:

Network: Hotspot
Endpoint: https://example.com
```
I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://example.com
I (6327) esp-x509-crt-bundle: Certificate validated
I (7127) esp_https_ota: Starting OTA...
I (7137) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
E (7137) esp_https_ota: Mismatch chip id, expected 0, found 27757
E (7147) simple_ota_example: Firmware upgrade failed
```

On the hotspot but switching back to the python bin server from this example:

Network: Hotspot
Endpoint: simple-ota-example binary server
```
I (7887) simple_ota_example: Starting OTA example task
I (7897) simple_ota_example: Attempting to download update from https://172.20.10.3:8070/hello-world.bin
E (7957) esp-x509-crt-bundle: Failed to verify certificate
E (7957) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (7967) esp-tls: Failed to open new connection
E (7967) transport_base: Failed to open a new connection
E (7977) HTTP_CLIENT: Connection failed, sock < 0
E (7977) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (7987) esp_https_ota: Failed to establish HTTP connection
E (7987) simple_ota_example: Firmware upgrade failed
```

### Notes

What do I need to do about my WiFi or where do I start looking to debug this issue?

hmalpani, 2023-03-15:
Hello @audreylim 
```
I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://172.20.10.3:8070/hello-world.bin
E (5907) esp-tls: [sock=54] connect() error: Host is unreachable
E (5907) esp-tls: Failed to open new connection
E (5917) transport_base: Failed to open a new connection
E (5917) HTTP_CLIENT: Connection failed, sock < 0
E (5927) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5937) esp_https_ota: Failed to establish HTTP connection
E (5937) simple_ota_example: Firmware upgrade failed
I (6247) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (6247) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
```
In the logs above that you provided, I see that the IP of the device is `192.168.0.16` which is on a different network from which you are trying to download the image i.e `172.20.10.3`. I think the download URL you entered is not correct. Can you please check the download URL has the IP of the host machine on which the server is started?

audreylim, 2023-03-15:
@hmalpani thanks, I ran it against the proper url this time, but got the same error:

```

I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://192.168.0.15:8070/hello-world.bin
E (5907) esp-tls: [sock=54] connect() error: Host is unreachable
E (5907) esp-tls: Failed to open new connection
E (5917) transport_base: Failed to open a new connection
E (5917) HTTP_CLIENT: Connection failed, sock < 0
E (5927) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5937) esp_https_ota: Failed to establish HTTP connection
E (5937) simple_ota_example: Firmware upgrade failed
I (8757) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (8757) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
```

binary server on command prompt:
```
python example_test_no_deps.py
C:\Users\Audrey\Downloads\pio-espidf-simple-ota-example-main\pio-espidf-simple-ota-example-main\example_test_no_deps.py:96: DeprecationWarning: ssl.wrap_socket() is deprecated, use SSLContext.wrap_socket()
  httpd.socket = ssl.wrap_socket(httpd.socket,
Starting HTTPS server at https://192.168.0.15:8070 serving folder C:\Users\Audrey\Downloads\pio-espidf-simple-ota-example-main\pio-espidf-simple-ota-example-main\http_server_root
```

audreylim, 2023-03-15:
@hmalpani , I believe the device is connected to wifi before it pings the server (for comparison, I changed the SSID to one that doesn't exist, and it kept attempting to connect to wifi). I also believe the author of the [codebase ](https://github.com/maxgerhardt/pio-espidf-simple-ota-example)I'm running had no issues completing the OTA. Here are the full logs:

```
Executing task in folder pio-espidf-simple-ota-example-main: C:\Users\Audrey\.platformio\penv\Scripts\platformio.exe device monitor --environment esp32dev 

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
I (28) boot: ESP-IDF 5.0.1 2nd stage bootloader
I (28) boot: compile time 00:10:53
I (28) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (68) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (76) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  3 factory          factory app      00 00 00010000 00100000
I (91) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (98) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (106) boot: End of partition table
I (110) boot: Defaulting to factory image
I (115) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34c8ch (216204) map
I (201) esp_image: segment 1: paddr=00044cb4 vaddr=3ffb0000 size=033c4h ( 13252) load
I (207) esp_image: segment 2: paddr=00048080 vaddr=40080000 size=07f98h ( 32664) load
I (221) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=95cfch (613628) map
I (442) esp_image: segment 4: paddr=000e5d24 vaddr=40087f98 size=0ccfch ( 52476) load
I (475) boot: Loaded app from partition at offset 0x10000
I (475) boot: Disabling RNG early entropy source...
I (486) cpu_start: Pro cpu up.
I (486) cpu_start: Starting app cpu, entry point is 0x400816b4
I (0) cpu_start: App cpu up.
I (503) cpu_start: Pro cpu start user code
I (503) cpu_start: cpu freq: 160000000 Hz
I (503) cpu_start: Application information:
I (508) cpu_start: Project name:     espidf_ota_example
I (514) cpu_start: App version:      1
I (518) cpu_start: Compile time:     Mar 15 2023 18:06:23
I (524) cpu_start: ELF file SHA256:  7b7971d46b85a677...
I (530) cpu_start: ESP-IDF:          5.0.1
I (535) cpu_start: Min chip rev:     v0.0
I (539) cpu_start: Max chip rev:     v3.99 
I (544) cpu_start: Chip rev:         v1.0
I (549) heap_init: Initializing. RAM available for dynamic allocation:
I (556) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (562) heap_init: At 3FFB7770 len 00028890 (162 KiB): DRAM
I (569) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (575) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (581) heap_init: At 40094C94 len 0000B36C (44 KiB): IRAM
I (589) spi_flash: detected chip: gd
I (592) spi_flash: flash io: dio
I (597) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (607) simple_ota_example: OTA example app_main start
I (637) simple_ota_example: SHA-256 for bootloader:  c421feb1b0e123b922d2146ccf4ad41cc89abfa1bc4e8d298e4183598773dfa0
I (867) simple_ota_example: SHA-256 for current firmware:  ab132bc17c818e7159bfbc69d7f157cea33561777aa05b2948910eb6c6c01577
[... 31 lines cut ...]
I (4177) wifi:state: auth -> assoc (0)
I (4187) wifi:state: assoc -> run (10)
I (4197) wifi:<ba-add>idx:0 (ifx:0, fc:4a:e9:31:85:52), tid:0, ssn:0, winSize:64
I (4247) wifi:connected with BCW710J-D137C-G, aid = 3, channel 1, BW20, bssid = fc:4a:e9:31:85:52
I (4247) wifi:security: WPA2-PSK, phy: bgn, rssi: -42
I (4247) wifi:pm start, type: 1

I (4347) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5867) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5867) example_common: Connected to example_netif_sta
I (5877) example_common: - IPv6 address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5887) wifi:Set ps type: 0

I (5887) simple_ota_example: Starting OTA example task
I (5897) simple_ota_example: Attempting to download update from https://192.168.0.15:8070/hello-world.bin
E (5907) esp-tls: [sock=54] connect() error: Host is unreachable
E (5907) esp-tls: Failed to open new connection
E (5917) transport_base: Failed to open a new connection
E (5917) HTTP_CLIENT: Connection failed, sock < 0
E (5927) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (5937) esp_https_ota: Failed to establish HTTP connection
E (5937) simple_ota_example: Firmware upgrade failed
I (8757) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (8757) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
```

hmalpani, 2023-03-15:
Hello @audreylim 
Can you please check if LWIP_IPV4 and EXAMPLE_CONNECT_IPV4 are enabled in menuconfig. The logs seem to suggest that the device is not getting the IPv4. Also can you please try the [simple_ota](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example) example present in IDF with your home wifi setup?

audreylim, 2023-03-15:
They're enabled. 

Running the simple_ota example from here gives me this:

```
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_dr_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
I (28) boot: ESP-IDF 5.0.1 2nd stage bootloader
I (28) boot: compile time 21:46:02
I (28) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34cbch (216252) map
I (174) esp_image: segment 1: paddr=00044ce4 vaddr=3ffb0000 size=033c4h ( 13252) load
I (180) esp_image: segment 2: paddr=000480b0 vaddr=40080000 size=07f68h ( 32616) load
I (193) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=95d54h (613716) map
I (415) esp_image: segment 4: paddr=000e5d7c vaddr=40087f68 size=0cd2ch ( 52524) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400816b4
I (0) cpu_start: App cpu up.
I (476) cpu_start: Pro cpu start user code
I (476) cpu_start: cpu freq: 160000000 Hz
I (476) cpu_start: Application information:
I (481) cpu_start: Project name:     espidf_ota_example
I (486) cpu_start: App version:      8989a97-dirty
I (492) cpu_start: Compile time:     Mar 15 2023 21:51:58
I (498) cpu_start: ELF file SHA256:  4ca53b9dc0692a9b...
I (504) cpu_start: ESP-IDF:          5.0.1
I (509) cpu_start: Min chip rev:     v0.0
I (513) cpu_start: Max chip rev:     v3.99 
I (518) cpu_start: Chip rev:         v1.0
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (530) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (536) heap_init: At 3FFB7770 len 00028890 (162 KiB): DRAM
I (543) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (555) heap_init: At 40094C94 len 0000B36C (44 KiB): IRAM
I (563) spi_flash: detected chip: gd
I (566) spi_flash: flash io: dio
I (571) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (581) simple_ota_example: OTA example app_main start
I (611) simple_ota_example: SHA-256 for bootloader:  4ca7d287397796df73e335bd9e0cac2c83618f8a76d5d383af3794cd10f7b275
I (841) simple_ota_example: SHA-256 for current firmware:  96f16c7e6e1a2883c4f4555eca3c725ce83a47f43dd95f67e8a93656387a70af
I (851) example_connect: Start example_connect.
I (861) wifi:wifi driver task: 3ffbf944, prio:23, stack:6656, core=0
I (861) system_api: Base MAC address is not set
[... 29 lines cut ...]
I (4161) wifi:state: assoc -> run (10)
I (4181) wifi:<ba-add>idx:0 (ifx:0, fc:4a:e9:31:85:52), tid:0, ssn:0, winSize:64
I (4201) wifi:connected with BCW710J-D137C-G, aid = 3, channel 1, BW20, bssid = fc:4a:e9:31:85:52
I (4201) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (4201) wifi:pm start, type: 1

I (4251) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5841) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8211) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (8211) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
I (8221) example_common: Connected to example_netif_sta
I (8221) example_common: - IPv4 address: 192.168.0.16,
I (8231) example_common: - IPv6 address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8241) wifi:Set ps type: 0

I (8241) simple_ota_example: Starting OTA example task
I (8251) simple_ota_example: Attempting to download update from https://192.168.0.15/hello-world.bin
E (13271) esp-tls: [sock=54] select() timeout
E (13271) esp-tls: Failed to open new connection
E (13271) transport_base: Failed to open a new connection
E (13271) HTTP_CLIENT: Connection failed, sock < 0
E (13281) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (13281) esp_https_ota: Failed to establish HTTP connection
E (13291) simple_ota_example: Firmware upgrade failed
```

audreylim, 2023-03-15:
Changing the URL to https://example.com, however, I am now able to connect (previously I wasn't able to). Let me make sure the server I'm running and the certs are correct.

```
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_dr_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
I (28) boot: ESP-IDF 5.0.1 2nd stage bootloader
I (28) boot: compile time 21:46:02
I (28) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34c9ch (216220) map
I (174) esp_image: segment 1: paddr=00044cc4 vaddr=3ffb0000 size=033c4h ( 13252) load
I (180) esp_image: segment 2: paddr=00048090 vaddr=40080000 size=07f88h ( 32648) load
I (193) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=95d54h (613716) map
I (415) esp_image: segment 4: paddr=000e5d7c vaddr=40087f88 size=0cd0ch ( 52492) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400816b4
I (0) cpu_start: App cpu up.
I (476) cpu_start: Pro cpu start user code
I (476) cpu_start: cpu freq: 160000000 Hz
I (476) cpu_start: Application information:
I (480) cpu_start: Project name:     espidf_ota_example
I (486) cpu_start: App version:      8989a97-dirty
I (492) cpu_start: Compile time:     Mar 15 2023 22:03:20
I (498) cpu_start: ELF file SHA256:  af4af2a5b67f4224...
I (504) cpu_start: ESP-IDF:          5.0.1
I (509) cpu_start: Min chip rev:     v0.0
I (513) cpu_start: Max chip rev:     v3.99 
I (518) cpu_start: Chip rev:         v1.0
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (530) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (536) heap_init: At 3FFB7770 len 00028890 (162 KiB): DRAM
I (542) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (555) heap_init: At 40094C94 len 0000B36C (44 KiB): IRAM
I (563) spi_flash: detected chip: gd
I (566) spi_flash: flash io: dio
I (571) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (581) simple_ota_example: OTA example app_main start
I (611) simple_ota_example: SHA-256 for bootloader:  4ca7d287397796df73e335bd9e0cac2c83618f8a76d5d383af3794cd10f7b275
I (841) simple_ota_example: SHA-256 for current firmware:  729675083fce6a13727af06acd3927996a0d042a2766e701d94d22c459721559
I (851) example_connect: Start example_connect.
I (861) wifi:wifi driver task: 3ffbf944, prio:23, stack:6656, core=0
I (861) system_api: Base MAC address is not set
I (861) system_api: read default base MAC address from EFUSE
I (871) wifi:wifi firmware version: 17afb16
[... 24 lines cut ...]
I (3481) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4351) wifi:state: init -> auth (b0)
I (4351) wifi:state: auth -> assoc (0)
I (4361) wifi:state: assoc -> run (10)
I (4381) wifi:<ba-add>idx:0 (ifx:0, fc:4a:e9:31:85:52), tid:0, ssn:0, winSize:64
I (4401) wifi:connected with BCW710J-D137C-G, aid = 3, channel 1, BW20, bssid = fc:4a:e9:31:85:52
I (4401) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (4411) wifi:pm start, type: 1

I (4491) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5841) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8911) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (8911) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
I (8921) example_common: Connected to example_netif_sta
I (8921) example_common: - IPv4 address: 192.168.0.16,
I (8931) example_common: - IPv6 address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8941) wifi:Set ps type: 0

I (8941) simple_ota_example: Starting OTA example task
I (8951) simple_ota_example: Attempting to download update from https://example.com
I (9281) esp-x509-crt-bundle: Certificate validated
I (9961) esp_https_ota: Starting OTA...
E (9961) esp_https_ota: Passive OTA partition not found
E (9961) simple_ota_example: Firmware upgrade failed
```

hmalpani, 2023-03-15:
> They're enabled.
>
> Running the simple_ota example from here gives me this:
>
> ```
> --- Terminal on COM4 | 115200 8-N-1
> --- Available filters and text transformations: colorize, debug, default, direct, [...]

I think the URL is not correct. The URL in the logs is for server started on port 80. Please check if the server is started on port 80 or port 8070(as it was before)

audreylim, 2023-03-15:
Running the HTTPS server [here](https://github.com/espressif/esp-idf/tree/master/examples/system/ota#run-https-server):

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_dr_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
I (28) boot: ESP-IDF 5.0.1 2nd stage bootloader
I (28) boot: compile time 21:46:02
I (28) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34cbch (216252) map
I (174) esp_image: segment 1: paddr=00044ce4 vaddr=3ffb0000 size=033c4h ( 13252) load
I (180) esp_image: segment 2: paddr=000480b0 vaddr=40080000 size=07f68h ( 32616) load
I (193) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=95d54h (613716) map
I (415) esp_image: segment 4: paddr=000e5d7c vaddr=40087f68 size=0cd2ch ( 52524) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400816b4
I (0) cpu_start: App cpu up.
I (476) cpu_start: Pro cpu start user code
I (476) cpu_start: cpu freq: 160000000 Hz
I (476) cpu_start: Application information:
I (480) cpu_start: Project name:     espidf_ota_example
I (486) cpu_start: App version:      8989a97-dirty
I (492) cpu_start: Compile time:     Mar 15 2023 22:17:17
I (498) cpu_start: ELF file SHA256:  e6e463796da04ae5...
I (504) cpu_start: ESP-IDF:          5.0.1
I (509) cpu_start: Min chip rev:     v0.0
I (513) cpu_start: Max chip rev:     v3.99 
I (518) cpu_start: Chip rev:         v1.0
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (530) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (536) heap_init: At 3FFB7770 len 00028890 (162 KiB): DRAM
I (542) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (555) heap_init: At 40094C94 len 0000B36C (44 KiB): IRAM
I (563) spi_flash: detected chip: gd
I (566) spi_flash: flash io: dio
I (571) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (581) simple_ota_example: OTA example app_main start
I (611) simple_ota_example: SHA-256 for bootloader:  4ca7d287397796df73e335bd9e0cac2c83618f8a76d5d383af3794cd10f7b275
I (841) simple_ota_example: SHA-256 for current firmware:  fc8d7eea143fbe101cadde463d5822650431d816d4cafe416166a1ecf7c1de01
I (851) example_connect: Start example_connect.
I (861) wifi:wifi driver task: 3ffbf944, prio:23, stack:6656, core=0
I (861) system_api: Base MAC address is not set
I (861) system_api: read default base MAC address from EFUSE
I (871) wifi:wifi firmware version: 17afb16
I (871) wifi:wifi certification version: v7.0
I (871) wifi:config NVS flash: enabled
I (881) wifi:config nano formating: disabled
I (881) wifi:Init data frame dynamic rx buffer num: 32
I (891) wifi:Init management frame dynamic rx buffer num: 32
I (891) wifi:Init management short buffer num: 32
[... 21 lines cut ...]
I (4161) wifi:state: assoc -> run (10)
I (4181) wifi:<ba-add>idx:0 (ifx:0, fc:4a:e9:31:85:52), tid:0, ssn:0, winSize:64
I (4201) wifi:connected with BCW710J-D137C-G, aid = 3, channel 1, BW20, bssid = fc:4a:e9:31:85:52
I (4201) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (4211) wifi:pm start, type: 1

I (4261) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5841) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8711) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (8711) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
I (8721) example_common: Connected to example_netif_sta
I (8721) example_common: - IPv4 address: 192.168.0.16,
I (8731) example_common: - IPv6 address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (8741) wifi:Set ps type: 0

I (8741) simple_ota_example: Starting OTA example task
I (8751) simple_ota_example: Attempting to download update from https://192.168.0.15:8070/hello-world.bin
E (8901) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (8901) esp-tls: Failed to open new connection
E (8901) transport_base: Failed to open a new connection
E (8911) HTTP_CLIENT: Connection failed, sock < 0
E (8911) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (8921) esp_https_ota: Failed to establish HTTP connection
E (8931) simple_ota_example: Firmware upgrade failed
```

audreylim, 2023-03-15:
Also ran this [script](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/simple_ota_example/pytest_simple_ota.py) and checked these

```
[*] Skip server certificate CN fieldcheck
[*] Support firmware upgrade bind specified interface
  (X) Bind wifi station interface
```

```
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
entry 0x4008064c
I (28) boot: ESP-IDF 5.0.1 2nd stage bootloader
I (28) boot: compile time 21:46:02
I (28) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34d2ch (216364) map
I (174) esp_image: segment 1: paddr=00044d54 vaddr=3ffb0000 size=033c4h ( 13252) load
I (180) esp_image: segment 2: paddr=00048120 vaddr=40080000 size=07ef8h ( 32504) load
I (193) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=95ee8h (614120) map
I (415) esp_image: segment 4: paddr=000e5f10 vaddr=40087ef8 size=0cd9ch ( 52636) load
I (448) boot: Loaded app from partition at offset 0x10000
I (448) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400816b4
I (0) cpu_start: App cpu up.
I (476) cpu_start: Pro cpu start user code
I (476) cpu_start: cpu freq: 160000000 Hz
I (476) cpu_start: Application information:
I (481) cpu_start: Project name:     espidf_ota_example
I (487) cpu_start: App version:      8989a97-dirty
I (492) cpu_start: Compile time:     Mar 15 2023 23:46:10
I (498) cpu_start: ELF file SHA256:  e274b8fe9d81bf70...
I (504) cpu_start: ESP-IDF:          5.0.1
I (509) cpu_start: Min chip rev:     v0.0
I (514) cpu_start: Max chip rev:     v3.99 
I (518) cpu_start: Chip rev:         v1.0
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (531) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (536) heap_init: At 3FFB7770 len 00028890 (162 KiB): DRAM
I (543) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (555) heap_init: At 40094C94 len 0000B36C (44 KiB): IRAM
I (563) spi_flash: detected chip: gd
I (566) spi_flash: flash io: dio
I (571) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (581) simple_ota_example: OTA example app_main start
I (611) simple_ota_example: SHA-256 for bootloader:  4ca7d287397796df73e335bd9e0cac2c83618f8a76d5d383af3794cd10f7b275
I (841) simple_ota_example: SHA-256 for current firmware:  dca74381ab1f1948bd5a68bb50a67b8f73f060190927be9de49c56d2bd87f434
[... 34 lines cut ...]
I (4201) wifi:connected with BCW710J-D137C-G, aid = 3, channel 1, BW20, bssid = fc:4a:e9:31:85:52
I (4201) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (4211) wifi:pm start, type: 1

I (4301) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5851) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (6711) esp_netif_handlers: example_netif_sta ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
I (6711) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.0.16
I (6721) example_common: Connected to example_netif_sta
I (6721) example_common: - IPv4 address: 192.168.0.16,
I (6731) example_common: - IPv6 address: fe80:0000:0000:0000:96b9:7eff:fee2:54ac, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (6741) wifi:Set ps type: 0

I (6741) simple_ota_example: Starting OTA example task
I (6751) simple_ota_example: Bind interface name is st1
I (6751) simple_ota_example: Attempting to download update from https://192.168.0.15:8070/hello-world.bin
E (6911) esp-x509-crt-bundle: Failed to verify certificate
E (6911) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (6911) esp-tls: Failed to open new connection
E (6911) transport_base: Failed to open a new connection
E (6921) HTTP_CLIENT: Connection failed, sock < 0
E (6931) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6931) esp_https_ota: Failed to establish HTTP connection
E (6941) simple_ota_example: Firmware upgrade failed
```

audreylim, 2023-03-16:
Fixed in the other [example ](https://github.com/maxgerhardt/pio-espidf-simple-ota-example/issues/4#issuecomment-1471146901), I'm good for now, thanks!

closed 2023-03-16 by audreylim