espressif/esp-idf #10993: ledc gamma curve fade for ESP32-C3, ESP32-S3

MacWyznawca, 2023-03-15:
### Is your feature request related to a problem?

Will ledc: support gamma curve fade be implemented in other ESP chips (ESP32-C3, ESP-32S3)? 
If so, at what possible date?
If not, please also let me know, as I need to plan my own libraries.

songruo, 2023-03-16:
Hi @MacWyznawca, ESP-IDF supports gamma curve fade through multi-fade hardware feature. And this is a new hardware feature since ESP32C6 chip. 

For earlier chips like C3 and S3, you can still achieve gamma curve fade, but with cpu intervention. You need to feed in the new linear fade parameters and start the new fade when the last linear fade ends. As you can see, the benefit of having multi-fade hardware feature is the linear fades can continue one after another automatically, with only one fade end interrupt being triggered at the end of the multi-fade. 

I think to support gamma curve fade on earlier chips belongs more to application-level. Therefore, we don't plan to support it in the driver.

MacWyznawca, 2023-03-16:
Thanks for the hint @songruo 
I am using interrupts at the end of the linear fade, but for now for flashing. 
Now I see that I can use it to get another value from the gamma array. Only I need to adjust the duration of the linear episodes well, for the whole fade.
Thank you.

closed 2023-03-16 by MacWyznawca