espressif/esp-idf #10994: Cannot read I2S data when I2C & UART are active

dizcza, 2023-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1-320-ge14c47ee44

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8

### Power Supply used.

USB

### What is the expected behavior?

I expect to read I2S data from the channel.

### What is the actual behavior?

The I2S data is not received at all.

### Steps to reproduce.

I've updated your "i2c_record" example:

```c
/* I2S Digital Microphone Recording Example */
#include <stdio.h>
#include <string.h>
#include <math.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_log.h"
#include "esp_err.h"
#include "esp_system.h"
#include "esp_vfs_fat.h"
#include "esp_timer.h"
#include "driver/mcpwm_cap.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "driver/i2s_std.h"
#include "driver/uart.h"
#include "sdcard.h"
#include "sdcard_record.h"
#include "wifi_sta.h"
#include "webserver.h"
#include "ds3231.h"

static const char *TAG = "pdm_rec_example";

#define EXAMPLE_SAMPLE_RATE 16000
#define EXAMPLE_BIT_SAMPLE  32
[... 268 lines cut ...]
Speed: 40 MHz
Size: 29580MB
CSD: ver=2, sector_size=512, capacity=60579840 read_bl_len=9
SSR: bus_width=1
Free: 29517MB
I (437) uart: queue free spaces: 100
I (537) pdm_rec_example: GPS started
I (537) pdm_rec_example: Initializing I2C1 SDA=26 SCL=25 clk=400000
D (537) i2s_common: rx channel is registered on I2S0 successfully
W (537) i2s_common: dma frame num is out of dma buffer size, limited to 1023
D (547) i2s_common: DMA malloc info: dma_desc_num = 4, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 4092
D (557) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 39 [mclk] 4096000 Hz [bdiv] 4 [bclk] 1024000 Hz
D (567) i2s_std: The rx channel on I2S0 has been initialized to STD mode successfully
D (577) i2s_common: i2s rx channel enabled
I (577) pdm_rec_example: Opening file
Read Failed!
Read Failed!
Read Failed!
Read Failed!
```
```

### More Information.

The default sdkconfig is used.

dizcza, 2023-03-15:
Tried putting I2S & UART into IRAM with no luck.

dizcza, 2023-03-15:
I confirm that setting
```
        .intr_alloc_flags = (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_INTRDISABLED),
```
in the legacy I2S driver fixes the issue. Likewise, someone has forgotten to populate `I2S_INTR_ALLOC_FLAGS` from `i2s_common.c` to
https://github.com/espressif/esp-idf/blob/1559b6309fe207add518828246a05f80582ee59d/components/driver/i2s/i2s_std.c#L220

So the fix is to move 
```c
// If ISR handler is allowed to run whilst cache is disabled,
// Make sure all the code and related variables used by the handler are in the SRAM
#if CONFIG_I2S_ISR_IRAM_SAFE
#define I2S_INTR_ALLOC_FLAGS    (ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_LOWMED)
#define I2S_MEM_ALLOC_CAPS      (MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT)
#else
#define I2S_INTR_ALLOC_FLAGS    (ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_LOWMED)
#define I2S_MEM_ALLOC_CAPS      MALLOC_CAP_DEFAULT
#endif //CONFIG_I2S_ISR_IRAM_SAFE
#define I2S_DMA_ALLOC_CAPS      (MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA)
```
from `i2s_common.c` to `i2s_common.h` and then use `I2S_INTR_ALLOC_FLAGS` in the `i2s_channel_init_std_mode` function.

I'll come up with the fix in a PR.

L-KAYA, 2023-03-20:
Thanks for locating the bug! Updated in https://github.com/espressif/esp-idf/pull/10997

dizcza, 2023-04-07:
Fixed

closed 2023-04-07 by dizcza