espressif/esp-idf #10995: Task watchdog does not reset the board

QuentinFarizon, 2023-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

When the task watchdog is triggered, it does not reset the board.

I was exepecting it to be the case, since this is the default behavior of all watchdogs I've encountered on various boards.

I'm not sure what is the expected behaviour from reading the documentation, it should be clearer.

### What is the actual behavior?

It prints a log

### Steps to reproduce.

Add while (true) {} in main loop

### Debug Logs.

```plain
Backtrace: 0x4202FA4E:0x3FC9B8D0 0x40377171:0x3FC9B900 0x42009576:0x3FCF4AE0 0x420B3897:0x3FCF4B30
0x4202fa4e: task_wdt_timeout_handling at /home/quentin/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at /home/quentin/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40377171: _xt_lowint1 at /home/quentin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

0x42009576: app_main at /home/quentin/Code/Boks/boks-link/main/main.c:143 (discriminator 13)

0x420b3897: main_task at /home/quentin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

E (51582) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (51582) task_wdt:  - IDLE (CPU 0)
E (51582) task_wdt: Tasks currently running:
E (51582) task_wdt: CPU 0: main
E (51582) task_wdt: CPU 1: IDLE
E (51582) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4202FA4E:0x3FC9B8D0 0x40377171:0x3FC9B900 0x42009576:0x3FCF4AE0 0x420B3897:0x3FCF4B30
0x4202fa4e: task_wdt_timeout_handling at /home/quentin/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at /home/quentin/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40377171: _xt_lowint1 at /home/quentin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

0x42009576: app_main at /home/quentin/Code/Boks/boks-link/main/main.c:143 (discriminator 13)

0x420b3897: main_task at /home/quentin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

E (56582) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (56582) task_wdt:  - IDLE (CPU 0)
E (56582) task_wdt: Tasks currently running:
E (56582) task_wdt: CPU 0: main
E (56582) task_wdt: CPU 1: IDLE
E (56582) task_wdt: Print CPU 0 (current core) backtrace
```
```

kriegste, 2023-03-15:
Have you set
`CONFIG_ESP_TASK_WDT_PANIC=y`
?

QuentinFarizon, 2023-03-15:
It is not set no.

I guess it should be part of this documentation page : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html

ESP-Marius, 2023-03-16:
@QuentinFarizon you are right, we can add a note stating that the default behavior does not include a board reset.

closed 2023-03-27 by espressif-bot