espressif/esp-idf #10996: OTA update bug

Mowglli, 2023-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.4.4.3-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

The ESP32-S3, is running a webserver, where it is possible to firmware update. The POST function from the websocket is done and functioning. Below a code snippet of the update_get_handler is to be found. It has a hard time finding the right postion.

```
     #include "esp_flash_partitions.h"
     #include "esp_partition.h"
     #include "esp_ota_ops.h"
     #define BUFFSIZE 4024*4
     static char ota_write_data[BUFFSIZE + 1] = { 0 };
     static esp_err_t update_get_handler(httpd_req_t *req) {

	static char *ota_write_dataPtr;
	esp_err_t err;
	/* update handle : set by esp_ota_begin(), must be freed via esp_ota_end() */
	esp_ota_handle_t update_handle = 0;
	const esp_partition_t *update_partition = NULL;

	ESP_LOGE(TAG, "Starting OTA");

	const esp_partition_t *configured = esp_ota_get_boot_partition();
	const esp_partition_t *running = esp_ota_get_running_partition();

	if (configured != running) {
		ESP_LOGW(TAG,
				"Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
				configured->address, running->address);
		ESP_LOGW(TAG,
				"(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
	}
	ESP_LOGI(TAG, "Running partition type %d subtype %d (offset 0x%08x)",
			running->type, running->subtype, running->address);

	update_partition = esp_ota_get_next_update_partition(NULL);
	assert(update_partition != NULL);
	ESP_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
			update_partition->subtype, update_partition->address);

	int binary_file_length = 0;
	/*deal with all receive packet*/
[... 187 lines cut ...]
I (29158) ./main/http_service1.c: Running partition type 0 subtype 16 (offset 0x00090000)
I (29159) ./main/http_service1.c: Writing to partition subtype 17 at offset 0x390000
I (29168) ./main/http_service1.c: Looking for stringt application/octet-stream
I (29175) ./main/http_service1.c: String not found
I (29181) rdm: HexBlock: 64
I (29185) rdm: ADDR 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
I (29191) rdm: 0000 72 20 64 79 6e 61 6d 69 63 20 61 6c 6c 6f 63 61
I (29198) rdm: 0010 74 69 6f 6e 3a 1b 5b 30 6d 0a 00 00 68 65 61 70
I (29206) rdm: 0020 5f 69 64 78 20 3c 3d 20 6e 75 6d 5f 68 65 61 70
I (29213) rdm: 0030 73 00 00 00 6c 6f 63 6b 00 00 00 00 2f 49 44 46
I (29220) ./main/http_service1.c: New firmware version: t-Disposition: form-data; name="myfile"; filename="a1d_firmware.bin"
Content-Tynts/heap/heap_caps.c
I (29234) ./main/http_service1.c: Running firmware version: BasicRDM-DMX-NoUniverse-9-g203e
I (29244) ./main/http_service1.c: esp_ota_begin succeeded
E (29309) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x2d)
E (29311) esp_ota_ops: not found the handle
E (29314) esp_ota_ops: not found the handle
```

### More Information.

Here is a wireshark capture, below There is Data, which is the bin file.
[image]

[capture.zip](https://github.com/espressif/esp-idf/files/10981494/capture.zip)

hmalpani, 2023-03-17:
Hello @Mowglli 
Can you share an example to reproduce this issue? That would be helpful in debugging this. Thanks.

Mowglli, 2023-03-23:
I found the problem, it was a setting in LWIP in the SDKconfig

closed 2023-03-23 by Mowglli

Backdraft007, 2023-04-28:
Hello @Mowglli ,
which setting is it? I have a similar problem.

Mowglli, 2023-04-28:
Under LWIP :  enable copy between Layer2 and Layer3 packets