espressif/esp-idf #10998: Enterprise WiFi: Sending certificates as part of provisioning

dalkvenk, 2023-03-15:
### Describe the solution you'd like.

We would like to send certificates during provisioning. Currently only SSID and passphrase are supported as part of `SetConfig`. Is this already supported? If there's already a PR, can you provide a pointer to it?

laukik-hase, 2023-03-27:
Hello, @dalkvenk!

You can create a custom endpoint for sending your custom certificates for WPA2 Enterprise connections and call the corresponding to establish the connection (Refer [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#additional-endpoints) for more information).

You can also try using the certificate bundle feature (see [here](https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/examples/wifi/wifi_enterprise/main/wifi_enterprise_main.c#L161)) for WPA2-Enterprise connections, which most of the widely used CA certs.

laukik-hase, 2023-04-26:
Hello, @dalkvenk!

Closing this issue due to lack of feedback, feel free to reopen with more updates. Thanks.

closed 2023-04-26 by laukik-hase