espressif/esp-idf #11001: Cannot change BLE tx power when using nimBLE stack

listout, 2023-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-DevKitM-2 and ESP-DevKit-V1

### Power Supply used.

USB

### What is the expected behavior?

BLE tx power level changes after calling api `esp_ble_tx_power_set` either after a connection happen or connection before ble initialization.

### What is the actual behavior?

No effect on tx power after calling the API.

### Steps to reproduce.

1. copy bleprph example from 4.4.3 or 4.4.4
2. Apply the following diff or changes
3. 
4. Step
...

listout, 2023-03-16:
Related #9089

rahult-github, 2023-03-16:
Hi @listout ,

 Please try below change: 
 
 @@ -97,7 +98,7 @@ bleprph_advertise(void)
      * special value BLE_HS_ADV_TX_PWR_LVL_AUTO.
      */
     fields.tx_pwr_lvl_is_present = 1;
-    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;
+    fields.tx_pwr_lvl = esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV);

basically, you need to call esp_ble_tx_power_get API to get the power details, whenever needed.  The above change should show up the dbB information on nrfConnect app , since it reads the information from adv data 

Thanks,
 Rahul

rahult-github, 2023-03-16:
Oh, that is because the API is returning index corresponding to absolute value.

listout, 2023-03-16:
Anyway to get 9db then?

rahult-github, 2023-03-17:
You can refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html?highlight=esp_pwr#_CPPv417esp_power_level_t . 

From the value received, which is the index , you can then use the above mapping to identify the dbm .  Then you can use it to set the adv data field, and that should show up on remote side.

listout, 2023-03-17:
Thanks, working.

closed 2023-03-17 by listout

listout, 2023-03-17:
For anyone coming later on. This what I changed in the bleprph example to make it work

```diff
diff --git a/main/main.c b/main/main.c
index 35770ff..d66dde7 100644
--- a/main/main.c
+++ b/main/main.c
@@ -20,6 +20,7 @@
 #include "esp_log.h"
 #include "nvs_flash.h"
 /* BLE */
+#include "esp_bt.h"
 #include "esp_nimble_hci.h"
 #include "nimble/nimble_port.h"
 #include "nimble/nimble_port_freertos.h"
@@ -97,7 +98,36 @@ bleprph_advertise(void)
      * special value BLE_HS_ADV_TX_PWR_LVL_AUTO.
      */
     fields.tx_pwr_lvl_is_present = 1;
-    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;
+	int power_level_t = esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV);
+	switch(power_level_t) {
+		case ESP_PWR_LVL_N12:
+			fields.tx_pwr_lvl = -12;
+			break;
+		case ESP_PWR_LVL_N9:
+			fields.tx_pwr_lvl = -9;
+			break;
+		case ESP_PWR_LVL_N6:
+			fields.tx_pwr_lvl = -6;
+			break;
+		case ESP_PWR_LVL_N3:
+			fields.tx_pwr_lvl = -3;
+			break;
+		case ESP_PWR_LVL_N0:
+			fields.tx_pwr_lvl = 0;
+			break;
+		case ESP_PWR_LVL_P9:
+			fields.tx_pwr_lvl = 9;
+			break;
+		case ESP_PWR_LVL_P6:
+			fields.tx_pwr_lvl = 6;
+			break;
+		case ESP_PWR_LVL_P3:
+			fields.tx_pwr_lvl = 3;
+			break;
+		default:
+			fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;
+			break;
+	}
 
     name = ble_svc_gap_device_name();
     fields.name = (uint8_t *)name;
@@ -151,7 +181,11 @@ bleprph_gap_event(struct ble_gap_event *event, void *arg)
 
     switch (event->type) {
     case BLE_GAP_EVENT_CONNECT:
+        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);
+        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
+        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,ESP_PWR_LVL_P9);
         /* A new connection was established or a connection attempt failed. */
+        MODLOG_DFLT(INFO, "connected");
         MODLOG_DFLT(INFO, "connection %s; status=%d ",
                     event->connect.status == 0 ? "established" : "failed",
                     event->connect.status);
@@ -159,6 +193,7 @@ bleprph_gap_event(struct ble_gap_event *event, void *arg)
             rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
             assert(rc == 0);
             bleprph_print_conn_desc(&desc);
+        MODLOG_DFLT(INFO, "connected");
         }
         MODLOG_DFLT(INFO, "\n");
 
@@ -374,6 +409,9 @@ app_main(void)
     /* Set the default device name. */
     rc = ble_svc_gap_device_name_set("nimble-bleprph");
     assert(rc == 0);
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
+    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,ESP_PWR_LVL_P9);
 
     /* XXX Need to have template for store */
     ble_store_config_init();
```