espressif/esp-idf #11002: LWIP connect() results in -1, errno 128 in conjuction with some APs

N-Chase, 2023-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Connect it to "test.mosquitto.org" via LWIP through any given AP.

### What is the actual behavior?

Connection attempt fails almost any given time when connecting using some APs I've tested. Works good and rarely fails on others and if it happens, usually a reboot fixes it.

### Steps to reproduce.

For me it consistently fails when commanded to connect via a Hotspot provided by my Redmi Android phone.  For a customer of us, it regularly fails using a common Router.

### Debug Logs.

```plain
This is the output produced with LWIP debug output set to full:

(I) [10:58:37] esp32s3_wifi.cpp::connect():150 | Connecting to AP "Redmi Note 10 Pro"
/ > ~~~~~~~~~~~~etharp_timer
dhcp_coarse_tmr()
dns_tmr: dns_check_entries
netif: setting default interface st
dhcp_start(netif=0x3fcc74c8) st11
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
pbuf_alloc(length=308)
pbuf_alloc(length=308) == 0x3c26f464
transaction id xid(9d811d36)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
pbuf_add_header: old 0x3c26f4ac new 0x3c26f4a4 (8)
pbuf_add_header: old 0x3c26f4a4 new 0x3c26f490 (20)
ip4_output_if: st11
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
+-------------------------------+
[... 1395 lines cut ...]
pbuf_add_header: old 0x3c26f490 new 0x3c26f482 (14)
ethernet_output: sending packet 0x3c26f464
(I) [10:58:58] esp32s3_wifi.cpp::disconnect():242 | pbespu_wifi_dfisconne_ct() ->f 0r
ee(0x3c26f464)
pbuf_free: deallocating 0x3c26f464
dhcp_release: RELEASED, DHCP_STATE_OFF
netif_set_ipaddr: netif address being changed
netif: netmask of interface st set to 0.0.0.0
netif: GW address of interface st set to 0.0.0.0
dhcp_release_and_stop(): dhcp state is OFF
(I) [10:58:58] esp32s3_wifi.cpp::event_handler():63 | Disconnected from "Redmi Note 10 Pro". Reason: 8
netif: setting default interface ''
netif_remove: removed netif
(I) [10:58:58] esp32s3_wifi.cpp::disconnect():249 | esp_wifi_stop() -> 0
(E) [10:58:58] protocol_hub.cpp::process():1136 | Aborting CON (5| attempts)
(I) [10:58:58] protocol_hub.cpp::process():498 | state::error
(I) [10:58:58] protocol_hub.cpp::process():532 | t_retry: 60
(I) [10:58:58] protocol_hub.cpp::log_event_time():388 | Time: 1678964338
(I) [10:58:58] protocol_hub.cpp::log_event_time():393 | Next event UT: 1678964398
(I) [10:58:58] protocol_hub.cpp::log_event_time():403 | Next event in: 60s
```

### More Information.

I can consistently reproduce this issue on V4.4.2 too.

david-cermak, 2023-03-31:
Hi @N-Chase 

From what I see in the log is that the device receives some DHCP packets and then initiates the connection to `91.121.93.94` sending `SYN`, retrying few times and then gives up with `max SYN retries reached`. After those DHCP packets no other packet arrived the whole time, which is the expected behavior to me. 
Could you please check that you physically receive replies to your `SYN` packets?

> For me it consistently fails when commanded to connect via a Hotspot provided by my Redmi Android phone. For a customer of us, it regularly fails using a common Router.

The usual issue when it works on *some routers* and doesn't on others could be firewall settings or iptable *routing.

> Works good and rarely fails on others and if it happens, usually a reboot fixes it.

This again suggests that some kind of "smart" rules applied filtering the incoming traffic. Do you use MQTT over plain TCP? Is it still reproducible when you switch to SSL?

david-cermak, 2023-04-05:
> How should I do this? Is there a way to tap the ESP low level interface?

You can use [wireshark](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html) and check the incoming traffic, or you can place a debug log in the low-level netif input:
```patch
diff --git a/components/esp_netif/lwip/netif/wlanif.c b/components/esp_netif/lwip/netif/wlanif.c
index 92129ea262..dca32f7a46 100644
--- a/components/esp_netif/lwip/netif/wlanif.c
+++ b/components/esp_netif/lwip/netif/wlanif.c
@@ -20,6 +20,7 @@
 #include "netif/etharp.h"
 
 #include "esp_netif.h"
+#include "esp_log.h"
 #include "esp_netif_net_stack.h"
 #include "lwip/esp_netif_net_stack.h"
 #include "esp_compiler.h"
@@ -155,6 +156,8 @@ void wlanif_input(void *h, void *buffer, size_t len, void* l2_buff)
     }
 
 #endif
+    ESP_LOG_BUFFER_HEXDUMP("wlan-input", (uint8_t *)buffer, len, ESP_LOG_VERBOSE);
+
 
     /* full packet send to tcpip_thread to process */
     if (unlikely(netif->input(p, netif) != ERR_OK)) {
```

> Do you use MQTT over plain TCP?
> In this test case yes.
> Is it still reproducible when you switch to SSL?

Since you're connecting to `"test.mosquitto.org"`, could you also try the ws default port = 80 (over the ws transport)? Just to rule out potential firewall filtering due to port number?