espressif/esp-idf #11003: FAT32 sd-card FATFS Library usage doesnot include the sd-card bad-sectors

hardik-dozee, 2023-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The problem is the bad-sectors accumulate in sd-card fat32 filesystem but doesnt count in the usage. How to get the bad sectors to be part of usage to measure the sd-card corruption.

usage:

bool ret_sd_usage = get_fatfs_usage(&bytes_total, &bytes_free);
ESP_LOGI(TAG, "FAT FS returns(%d): %10u kiB total, %10u kiB free",ret_sd_usage, bytes_total, bytes_free);

static bool get_fatfs_usage(size_t* out_total_bytes, size_t* out_free_bytes)
{
	FATFS *fs;
    DWORD fre_clust, fre_sect, tot_sect;
	bool ret = true;
    /* Get volume information and free clusters of drive 0 */
    int res = 	("0:", &fre_clust, &fs);
	printf("\nres == %d FR_OK\n", res);
	if(res == FR_OK)
	{
		/* Get total sectors and free sectors */
		tot_sect = (fs->n_fatent - 2) * fs->csize;
		fre_sect = fre_clust * fs->csize;

		// /* Print the free space (assuming 512 bytes/sector) */
		// printf("%10u KiB total drive space.\n%10u KiB available.\n",
		//        tot_sect / 2, fre_sect / 2);

		if (out_total_bytes != NULL)
		{
			*out_total_bytes = tot_sect / 2;
		}
		else
		{
			ret = false;
		}

		if (out_free_bytes != NULL)
		{
			*out_free_bytes = fre_sect / 2;
		}
		else
		{
			ret = false;
		}
	}
	else
	{
		ret = false;
	}
	return ret;
}

RathiSonika, 2026-08-19:
Thanks for the question. For this use case you may also use the ESP-IDF helper esp_vfs_fat_info(), which provides total and free bytes for the FAT volume.

However, these APIs report logical filesystem usage only. SD card bad-block handling is typically internal to the card controller and is not usually exposed at the FATFS layer, so this information is generally not available as part of filesystem usage.

RathiSonika, 2026-08-19:
A possible direction would be to explore a more general health-check API for storage devices, depending on what information the underlying media can actually expose. We will evaluate whether it makes sense to add a more generic health-check API in ESP-IDF. 

For now, I am closing this issue. Feel free to reopen or create new issue if you encounter similar issue.

closed 2026-08-19 by RathiSonika