espressif/esp-idf #11005: is there a way to just use one ota definition instead of two ota definitions

casitelwithdean, 2023-03-16:
### Is your feature request related to a problem?

  Hi,i am using esp32S3N8R8 ,esp-sr  occupys about more than 3MB,it means only 5MB  FLASH left ,and my bin's size is about 2000k,in this case I can't use ota ,because it needs two ota definitions ,totolly 3+2+2*2=9MB>8MB.(((i know there is N16R8,but N16R8 is not a good resolution for me😀)))
  So i wanna ask if there is a way to just use one ota definition instead of two ota definitions?
  Thanks a lot.

ESP-Marius, 2023-03-17:
@casitelwithdean Unfortunately it is not possible to just have one OTA partition as this would mean you are trying to overwrite the app you are running. 

Even if we were to load the OTA part into memory to allow it to run while writing the new app this would mean that any error during the update process (panic or just a power-loss) would result in an app partition with an app that cannot boot and you would have no way of recovering from this. So allowing this wouldnt be very useful.

 To suggestions:
* Read through the [performance guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html) on minimizing binary size. Maybe you'll be able to trim it down to fit into the available space.
* What is the second 2 MB you mention? 3 MB (esp-sr) 2*2 MB (OTA) and 2 MB what?

chipweinberger, 2023-03-17:
If you don't want to have two full ota apps, I personally use these scheme:

- ota0, 1MB: a minimal app that only has the ability to update ota1
- ota1, 6MB: the full app

ESP-IDF is not really set up for the use case, but it works fine.

For me I always boot into ota1 unless a button is pushed at boot time.

casitelwithdean, 2023-03-17:
> py the contents of the partition table csv file you are

Thanks ,and sorry for replying late . the extra "2MB" is factory app.
I thought unless factory app ,it still needs ota_0 and ota_1,it means totolly 2MB(factory_app)+2*2MB(two ota).
I think I've always been wrong on understanding of OTA for a long time.
If the factory app is unnecessary in OTA,then my problem will be sovled.
So is the factory app unnecessary?

casitelwithdean, 2023-03-17:
> se copy the contents of the partition

Thanks for relpying.the "2MB" is factory app.
I guess i've  misunderstood OTA for the whole time. I used to think OTA totolly needs factory_app,ota0 and ota_so it's 9MB.

ESP-Marius, 2023-03-17:
The factory app is indeed not mandatory if you have the OTA apps.

casitelwithdean, 2023-03-17:
> The factory app is indeed not mandatory if you have the OTA apps.

Thanks a lot!

closed 2023-03-17 by casitelwithdean