espressif/esp-idf #11006: [mesh] v4.4 MTXON task corrupting heap

KonssnoK, 2023-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-116-g00bb43ff24

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

As reported in 
https://github.com/espressif/esp-idf/issues/10992

we are investigating crashes that happen to our devices.

We managed to reduce the triggering conditions to the following:
- have 2 mesh devices connecting to MQTT with dynamic settings
- Disable all watchdogs, cores + interrupts
- make the child device send 1 packet per second with 1024 payload
- reset the root
- the child device will crash
- We enabled comprehensive HEAP tracing 
- We added a piece of code in heap_caps that logs each time a malloc/free is made

- After multiple retries, we see that the address that is corrupted is touched mainly by MTXON
- We isolated the logs to MTXON mallocs.

Here is the result:
```

W (16:07:12.957) mesh_main: <MESH_EVENT_TODS_REACHABLE>reachable:1
I (16:07:12.964) mesh_main: <MESH_EVENT_ROOT_ADDRESS>root address:7c:df:a1:e0:8b:7d
I (16:07:12.973) offline: object saved to PSRAM, free 490247
I (16:07:13.933) esp_netif_handlers: sta ip: 192.168.253.227, mask: 255.255.255.0, gw: 192.168.253.43
W (16:07:13.934) network: GOT IP from sta
I (16:07:13.938) mesh_netif: Interface AP
I (16:07:13.946) network: Network connected
I (16:07:13.948) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (16:07:13.953) mqtt_app: Try [1] connection to mqtts://mqtt-stg.tiko.energy:8883
I (40932) wifi:<ba-add>idx:0 (ifx:0, d6:92:7d:c3:02:1e), tid:0, ssn:3, winSize:64
CORRUPT HEAP: Invalid data at 0x3dea2214. Expected 0xfefefefe got 0x0048fefe
CORRUPT HEAP: Invalid data at 0x3dea2218. Expected 0xfefefefe got 0xfefe0000
(456) heap_history[0] action HEAP_ENTER_MALLOC, task MTXON
(455) heap_history[1] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DE96E08-0x3DE96E38
(454) heap_history[2] action HEAP_ENTER_MALLOC, task MTXON
(453) heap_history[3] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DE96E48-0x3DE96E72
(452) heap_history[4] action HEAP_ENTER_FREE, task MTXON, address 0x3DE11BF4-0x00000000
(451) heap_history[5] action HEAP_EXIT_FREE, task MTXON
[... 439 lines cut ...]
(11) heap_history[445] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA2260-0x3DEA2290
(10) heap_history[446] action HEAP_ENTER_MALLOC, task MTXON
(9) heap_history[447] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA22A0-0x3DEA22CA
(8) heap_history[448] action HEAP_ENTER_MALLOC, task MTXON
(7) heap_history[449] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA0FE4-0x3DEA1014
(6) heap_history[450] action HEAP_ENTER_MALLOC, task MTXON
(5) heap_history[451] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA2260-0x3DEA228A
(4) heap_history[452] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA21F0-0x00000000
(3) heap_history[453] action HEAP_EXIT_FREE, task MTXON
(2) heap_history[454] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA1030-0x00000000
(1) heap_history[455] action HEAP_EXIT_FREE, task MTXON

assert failed: multi_heap_malloc multi_heap_poisoning.c:237 (ret)

Backtrace: 0x403760e2:0x3fcd6fb0 0x40382323:0x3fcd6fd0 0x4038aa91:0x3fcd6ff0 0x40389bde:0x3fcd7110 0x403764ec:0x3fcd7130 0x4037654b:0x3fcd7150 0x403769ed:0x3fcd7170 0x4037dc88:0x3fcd7190 0x420e7535:0x3fcd71b0 0x420db341:0x3fcd71d0 0x420db3bd:0x3fcd71f0 0x420d7573:0x3fcd7210 0x420d61da:0x3fcd7250 0x420d666f:0x3fcd7270 0x420d6a33:0x3fcd72e0 0x420d7c93:0x3fcd7310 0x4216409d:0x3fcd7330 0x42095924:0x3fcd7350 0x40385d8a:0x3fcd7390

```

### What is the actual behavior?

above

### Steps to reproduce.

above

KonssnoK, 2023-03-16:
I confirm it's the TCPIP the other component in this crashes.
```
[heap_corrupt.txt](https://github.com/espressif/esp-idf/files/10993310/heap_corrupt.txt)

	Line    2: CORRUPT HEAP: Invalid data at 0x3de9e130. Expected 0xfefefefe got 0x008cfefe
	Line    3: CORRUPT HEAP: Invalid data at 0x3de9e134. Expected 0xfefefefe got 0xfefe0000
	Line   59: (969) heap_history[1003] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E609
	Line   60: (968) heap_history[1004] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  213: (815) heap_history[133] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E609
	Line  214: (814) heap_history[134] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  367: (661) heap_history[287] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E609
	Line  368: (660) heap_history[288] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  541: (487) heap_history[461] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E5C8
	Line  546: (482) heap_history[466] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  697: (331) heap_history[617] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E5C8
	Line  702: (326) heap_history[622] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  763: (265) heap_history[683] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E5C8
	Line  768: (260) heap_history[688] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  789: (239) heap_history[709] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E5C8
	Line  790: (238) heap_history[710] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  807: (221) heap_history[727] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E5C8
	Line  812: (216) heap_history[732] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  895: (133) heap_history[815] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E6F4
	Line  896: (132) heap_history[816] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
	Line  987: (41) heap_history[907] action HEAP_EXIT_MALLOC, task tiT, address 0x3DE9E10C-0x3DE9E6F4
	Line  996: (32) heap_history[916] action HEAP_ENTER_FREE, task MTXON, address 0x3DE9E10C-0x00000000
```

[heap_corrupt.txt](https://github.com/espressif/esp-idf/files/10993313/heap_corrupt.txt)

chipweinberger, 2023-03-17:
That's an interest approach to debugging heap corruption. 

I usually:
- enable corruption detection
- add lots of calls to heap_caps_check_integrity()
- start commenting out code until I know which code is causing the corruption 

Perhaps you can try this approach to narrow things down further.

KonssnoK, 2023-03-17:
> That's an interest approach to debugging heap corruption.
>
> I usually:
>
>     * enable corruption detection
>
>     * add lots of calls to heap_caps_check_integrity()
>
>     * start commenting out code [...]

we'll refine our investigation on this once we have more resources available.
Problem is, the mesh code is closed sourced and all our done steps lead to those modules.

On top, there are pending changes to the mesh library following our highest priority ticket,
https://github.com/espressif/esp-idf/issues/9955

zhangyanjiaoesp, 2023-03-21:
@KonssnoK Can you provide a demo to reproduce this issue?

KonssnoK, 2023-03-22:
the more the devices, the easier the crash
example:
[device_7.txt](https://github.com/espressif/esp-idf/files/11038748/device_7.txt)
[device_13.txt](https://github.com/espressif/esp-idf/files/11038749/device_13.txt)
[device_5.txt](https://github.com/espressif/esp-idf/files/11038750/device_5.txt)

I don't know why but in these logs the names of the tasks are wrong, in the terminal they are displayed correctly :)

KonssnoK, 2023-03-24:
just to complete the reporting, apparently it's possible to trigger more heap corruption crashes by simply enabling comprehensive heap poisoning in the example

https://github.com/KonssnoK/esp-idf/tree/bug/mesh_wifi/examples/mesh/ip_internal_network

used for 
https://github.com/espressif/esp-idf/issues/9955

One way to make the child device crash is to keep the hotspot off apparently

KonssnoK, 2023-03-27:
@zhangyanjiaoesp can you give us an update?
are you able to reproduce the issue? thanks :)

zhangyanjiaoesp, 2023-03-27:
@KonssnoK Yes, I can reproduce this issue according to your branch. And I'm debugging on it. 
(By the way, the changes in mesh lib has been merged into v4.4, 90d6e45d9ffc6c060d664d50063fe05c712d3042)

KonssnoK, 2023-03-29:
@zRedShift are you using a mesh network or simple wifi?
In our case it's likely an interaction between tcpip task and the MTXON task, which is part of the mesh library.

More generally speaking, it seems that in general this is happening when someone writes in a already released TCP packet.

KonssnoK, 2023-03-29:
@zRedShift we also use mbedtls, but our setup is using the wifi mesh environment. BLE is off for us after provisioning.

When i have some time i'll try to understand if i can do some tests to see if the issue is in an open part of the code.

The possible involved modules i see are:
- lwip tcpip 
- mbedtls
- wifi stack -> not debuggable

zhangyanjiaoesp, 2023-03-31:
@KonssnoK I'm still debugging on this issue, will feedback to you ASAP when I find the root cause.

KonssnoK, 2023-03-31:
@zhangyanjiaoesp did you manage to understand which component is triggering it?
if it's an open sourced one we can focus on it too.

zhangyanjiaoesp, 2023-04-04:
@KonssnoK 
Please use the new wifi lib to test, the corrupt issue has been solved on my side.
[wifi_lib_0404.zip](https://github.com/espressif/esp-idf/files/11144493/wifi_lib_0404.zip)
wifi firmware version: 7fec53d
If your test is passed,  then we will merged the fix into release/v4.4 ASAP.

KonssnoK, 2023-04-04:
thanks @zhangyanjiaoesp , we'll test asap

KonssnoK, 2023-04-04:
just a question, what is it built on top of?
15b1309726f600122b5d4539e73758ae1852168d ?

zhangyanjiaoesp, 2023-04-04:
@KonssnoK The wifi lib is base on your branch https://github.com/KonssnoK/esp-idf/tree/bug/mtxon-heap-corruption/examples/mesh/ip_internal_network, and it contains all the fixes from https://github.com/espressif/esp-idf/issues/9955.
If you want to update your release/v4.4 version, you can base on https://github.com/espressif/esp-idf/commit/90d6e45d9ffc6c060d664d50063fe05c712d3042.

KonssnoK, 2023-04-04:
@zhangyanjiaoesp we saw there were additional fixes to the wifi library, so we are currently aiming at 
4c7d97e2bdbd26b1ad6adc6de8051888e1feec10

First though, let's check if everything is ok with the current branch

KonssnoK, 2023-04-04:
@zhangyanjiaoesp we continue to test but currently you have the green light for merging, we are not able to reproduce the issue with the same examples. Thanks :)

zRedShift, 2023-04-04:
Would this fix affect v5.x versions of `esp-idf`? And is there a possibility that it solves a non-mesh related `esp-wifi` bug I mentioned here: https://github.com/espressif/esp-idf/issues/11006#issuecomment-1487529101?

KonssnoK, 2023-04-04:
> Would this fix affect v5.x versions of `esp-idf`? And is there a possibility that it solves a non-mesh related `esp-wifi` bug I mentioned here: [#11006 (comment)](https://github.com/espressif/esp-idf/issues/11006#issuecomment-1487529101)?

these fixes are usually backported everywhere (still maintained), so it will be pushed also to 5.0 for sure.
The 4.4 push is focused just because our product line works on v4.4 :)

KonssnoK, 2023-04-04:
@zhangyanjiaoesp we were able to reproduce again one crash,
we'll try now to isolate and reproduce

```
W (10:10:57.634) stats: iteration_time_us=1000004 Name=last% IDLE=40 IDLE=63 task_mesh_rx=1 tiT=4 ipc1=24 wifi=5 updater=56 MTXON=1
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40389eb2  PS      : 0x00060e33  A0      : 0x8038a678  A1      : 0x3fcb7370
0x40389eb2: remove_free_block at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:207
 (inlined by) block_locate_free at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:442
 (inlined by) tlsf_malloc at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:849

A2      : 0x3d800014  A3      : 0x0000010c  A4      : 0x3d896e6c  A5      : 0x3d8a8dd4
A6      : 0x3d8112c4  A7      : 0x3d8a8d44  A8      : 0x3d8a8e24  A9      : 0x00000013
A10     : 0x964c06f3  A11     : 0x00000004  A12     : 0xfffffffc  A13     : 0x00007001
A14     : 0x00060e20  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000700d  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x40389eaf:0x3fcb7370 0x4038a675:0x3fcb7390 0x40376709:0x3fcb73b0 0x4037685b:0x3fcb73d0 0x403763f9:0x3fcb7420 0x421156f7:0x3fcb7440 0x42115754:0x3fcb7460 0x42116215:0x3fcb7480 0x4038b149:0x3fcb7640 0x4038b235:0x3fcb7690 0x4003f4bd:0x3fcb76b0 |<-CORRUPTED
0x40389eaf: remove_free_block at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:206
 (inlined by) block_locate_free at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:442
 (inlined by) tlsf_malloc at C:/repos/v3/esp-idf/components/heap/heap_tlsf.c:849

0x4038a675: multi_heap_malloc_impl at C:/repos/v3/esp-idf/components/heap/multi_heap.c:187

0x40376709: heap_caps_malloc_base at C:/repos/v3/esp-idf/components/heap/heap_caps.c:175

0x4037685b: heap_caps_malloc_prefer at C:/repos/v3/esp-idf/components/heap/heap_caps.c:290

0x403763f9: wifi_malloc at C:/repos/v3/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:71

0x421156f7: mesh_malloc at ??:?

0x42115754: esp_mesh_create_context at ??:?

0x42116215: esp_mesh_wifi_recv_cb at ??:?

0x4038b149: sta_input at ??:?

0x4038b235: sta_rx_cb at ??:?
```

KonssnoK, 2023-04-04:
[device_7440_press.zip](https://github.com/espressif/esp-idf/files/11147335/device_7440_press.zip)
@zhangyanjiaoesp @mhdong here are the logs

KonssnoK, 2023-04-04:
Generically speaking the stability of the system has improved thousands of folds.
We still get some crashes, which seem very similar to what was happening before. 
@zhangyanjiaoesp maybe you can check for similar code (to the one you fixed) in the libraries?

To trigger them, instead of doing high traffic, we do parallel firmware updates.
- 3 devices
- trigger a firmware update on all 3 devices at the same time

Here an example :
```
	Line 1609: CORRUPT HEAP: Invalid data at 0x3dea2dbc. Expected 0xfefefefe got 0x3de00018
	Line 2290: (344) heap_history[738] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2CBC-0x3DEA2D4F
	Line 2334: (300) heap_history[782] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA2D3C-0x3DEA2D6C
	Line 2336: (298) heap_history[784] action HEAP_EXIT_MALLOC, task MTXON, address 0x3DEA2D80-0x3DEA2DAA
	Line 2337: (297) heap_history[785] action HEAP_ENTER_FREE, task MTX, address 0x3DEA2D80-0x00000000
	Line 2339: (295) heap_history[787] action HEAP_ENTER_FREE, task MTX, address 0x3DEA2D3C-0x00000000
	Line 2341: (293) heap_history[789] action HEAP_ENTER_FREE, task MRX, address 0x3DEA2D80-0x00000000
	Line 2343: (291) heap_history[791] action HEAP_ENTER_FREE, task MRX, address 0x3DEA2D3C-0x00000000
	Line 2353: (281) heap_history[801] action HEAP_ENTER_FREE, task task_mesh_rx, address 0x3DEA2D3C-0x00000000
	Line 2384: (250) heap_history[832] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2401: (233) heap_history[849] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2414: (220) heap_history[862] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2435: (199) heap_history[883] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2448: (186) heap_history[896] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2449: (185) heap_history[897] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2464: (170) heap_history[912] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2477: (157) heap_history[925] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2500: (134) heap_history[948] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2509: (125) heap_history[957] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2530: (104) heap_history[978] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2543: (91) heap_history[991] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2566: (68) heap_history[1014] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2567: (67) heap_history[1015] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2596: (38) heap_history[20] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
	Line 2609: (25) heap_history[33] action HEAP_ENTER_FREE, task MTXON, address 0x3DEA2D3C-0x00000000
	Line 2632: (2) heap_history[56] action HEAP_EXIT_MALLOC, task tiT, address 0x3DEA2D3C-0x3DEA2D9C
```

zhangyanjiaoesp, 2023-04-06:
@KonssnoK I will check the code

KonssnoK, 2023-04-06:
@zhangyanjiaoesp would it be possible to get the fix on top of 
4c7d97e2bdbd26b1ad6adc6de8051888e1feec10
?
This way we can check also latest fixes on the field, since we have devices isolating themselves (still investigating on why)

zhangyanjiaoesp, 2023-04-06:
@KonssnoK 
the wifi lib based on https://github.com/espressif/esp-idf/commit/4c7d97e2bdbd26b1ad6adc6de8051888e1feec10 is here:
[wifi_lib_0406.zip](https://github.com/espressif/esp-idf/files/11167958/wifi_lib_0406.zip)
wifi firmware version: 85cf6e5

> To trigger them, instead of doing high traffic, we do parallel firmware updates.
>     3 devices
>     trigger a firmware update on all 3 devices at the same time
>     be lucky ( 100 ) ... triggering crashes is now quite difficult. We are investigating in order to replicate these crashes.
This issue is still debugging.

zhangyanjiaoesp, 2023-04-06:
@zRedShift 
Since your issue is not related to wifi mesh, please create a new ticket to report your issue

KonssnoK, 2023-04-06:
> @KonssnoK the wifi lib based on [4c7d97e](https://github.com/espressif/esp-idf/commit/4c7d97e2bdbd26b1ad6adc6de8051888e1feec10) is here: [wifi_lib_0406.zip](https://github.com/espressif/esp-idf/files/11167958/wifi_lib_0406.zip) wifi firmware version: 85cf6e5
> 
> > To trigger them, instead of doing high traffic, we do parallel firmware updates.
> > 3 devices
> > trigger a firmware update on all 3 devices at the same time
> > be lucky ( 100 ) ... triggering crashes is now quite difficult. We are investigating in order to replicate these crashes.
> 
> This issue is still debugging.

meaning you were able to replicate some other crashes?
We are still trying to understand how to trigger them.
The new library much more stable!

zhangyanjiaoesp, 2023-04-06:
@KonssnoK No, I didn't  reproduce any crash issue, just look at your logs for some ideas.  If you find a good way to reproduce the problem, it will speed up the debug process.

zhangyanjiaoesp, 2023-04-06:
@KonssnoK  Since the fix has solved the original crash issue, and the new crash issue is hard to reproduce, we will merge the fix into release/v4.4 firstly. Is it OK?

KonssnoK, 2023-04-06:
@zhangyanjiaoesp that would be great, so we can move inside esp-idf without having to change wifi-lib

KonssnoK, 2023-04-13:
i will close this ticket and open a new one when we manage to replicate other crashes