espressif/esp-idf #11007: LEDC hardware fade fails when automatic light sleep is enabled

therealergo, 2023-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.
v4.4.4

### Operating System used.
Windows

### How did you build your project?
Command line with idf.py

### If you are using Windows, please specify command line type.
Powershell

### Development Kit, Power Supply
 - Custom board with ESP32-WROOM-32D 8MB Flash mounted. Automatic light sleep clock source is a well-tested 32kHz crystal oscillator. Powered by a LiPo battery.
 - Reproduced on SparkFun ESP32 Thing. Automatic light sleep clock source is internal 150kHz RC oscillator. USB Powered.

### General Description
When LEDC fades are performed with automatic light sleep enabled, the `ledc_fade_start(...)` function works on its first call but deadlocks and blocks indefinitely on all subsequent calls. In this case, the LEDC fade end callback is also never called.

Example code that reproduces this issue:
```
#include "esp_pm.h"
#include "esp_log.h"
#include "rom/uart.h"
#include "driver/ledc.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"

static IRAM_ATTR bool cb_ledc_fade_end_event(const ledc_cb_param_t *param, void *user_arg) {
    portBASE_TYPE taskAwoken = pdFALSE;

    if (param->event == LEDC_FADE_END_EVT) {
        SemaphoreHandle_t counting_sem = (SemaphoreHandle_t) user_arg;
        xSemaphoreGiveFromISR(counting_sem, &taskAwoken);
    }

    return (taskAwoken == pdTRUE);
}

extern "C" void app_main() {

    // Enable light sleep mode
    esp_pm_config_esp32_t pmConfig = {
        .max_freq_mhz = 240,
        .min_freq_mhz = 240,
        .light_sleep_enable = false,
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pmConfig));

    // Setup LEDC fade
    ledc_timer_config_t ledc_timer = {
        .speed_mode      = LEDC_LOW_SPEED_MODE, 
        .duty_resolution = LEDC_TIMER_10_BIT, 
        .timer_num       = LEDC_TIMER_0, 
        .freq_hz         = 5000, 
        .clk_cfg         = LEDC_USE_RTC8M_CLK
    };
    ledc_timer_config(&ledc_timer);
    ledc_channel_config_t ledc_channel = {
        .gpio_num   = GPIO_NUM_18,
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .channel    = LEDC_CHANNEL_0,
        .intr_type  = LEDC_INTR_FADE_END, 
        .timer_sel  = LEDC_TIMER_0,
        .duty       = 0,
        .hpoint     = 0,
        .flags = { .output_invert = 0 }
    };
    ledc_channel_config(&ledc_channel);
    ledc_fade_func_install(0);
    ledc_cbs_t callbacks = {
        .fade_cb = cb_ledc_fade_end_event
    };
    SemaphoreHandle_t counting_sem = xSemaphoreCreateCounting(1, 0);
    ledc_cb_register(ledc_channel.speed_mode, ledc_channel.channel, &callbacks, (void*)counting_sem);

    // Fade on-off indefinitely
    while (1) {
        ledc_set_fade_with_time(
            ledc_channel.speed_mode,
            ledc_channel.channel, 
            1024, 
            1000
        );
        ledc_fade_start(
            ledc_channel.speed_mode,
            ledc_channel.channel, 
            LEDC_FADE_NO_WAIT
        );
        ESP_LOGI("TEST", "Mark #1");
        uart_tx_wait_idle(CONFIG_ESP_CONSOLE_UART_NUM);
        xSemaphoreTake(counting_sem, portMAX_DELAY);
        ESP_LOGI("TEST", "Mark #2");
        uart_tx_wait_idle(CONFIG_ESP_CONSOLE_UART_NUM);
        ledc_set_fade_with_time(
            ledc_channel.speed_mode,
            ledc_channel.channel, 
            0, 
            1000
        );
        ledc_fade_start(
            ledc_channel.speed_mode,
            ledc_channel.channel, 
            LEDC_FADE_NO_WAIT
        );
        ESP_LOGI("TEST", "Mark #3");
        uart_tx_wait_idle(CONFIG_ESP_CONSOLE_UART_NUM);
        xSemaphoreTake(counting_sem, portMAX_DELAY);
        ESP_LOGI("TEST", "Mark #4");
        uart_tx_wait_idle(CONFIG_ESP_CONSOLE_UART_NUM);
    }
}
```

### What is the expected behavior?
`Mark #1`, `Mark #2`, `Mark #3`, and `Mark #4` should be printed in sequence, with an LED attached to IO18 flashing at 0.5Hz. This is what occurs when `light_sleep_enable == false`.

Instead, `Mark #1` is printed once and the LED initially fades from 100% to 0%. The task then blocks indefinitely. This is what occurs when `light_sleep_enable == true`. If the `xSemaphoreTake(...)` calls are removed then `Mark #1` and `Mark #2` are each printed, and the task then blocks indefinitely in `ledc_fade_start(...)`.

### What have you tried?
Debugging in the LEDC driver, it seems like the `ledc_fade_isr(...)` ISR isn't being called when it should. This results in `ledc_fade_sem` never being released by the driver after the first fade call.

songruo, 2023-03-23:
Hi @therealergo,

During sleep, APB clock is off, so there is no way for the LEDC peripheral to latch the fade end interrupt status. Therefore, the interrupt is lost.

The possible solution I would recommend is to utilize a `ESP_PM_NO_LIGHT_SLEEP` power management (pm) lock. Before the fade starts, you need to acquire the pm lock, so that auto light sleep could not happen. And inside the fade end callback function, the pm lock should be released. In such way, after the fade ends, auto light sleep can still occur. For more information about PM lock, please refer to https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/system/power_management.html#power-management-locks

therealergo, 2023-03-25:
Hi @songruo 

Thanks for taking a look for me. That makes sense. Based on how the driver is designed, I was hoping that there was special hardware that let this interrupt work in sleep, like there is for GPIO's and UART.

This definitely needs to be noted in the docs for the LEDC driver. The driver documentation claims that the LEDC peripheral works fine in light sleep with the RC_FAST_CLK source, without mentioning that any calls to the fade functions will deadlock the entire chip.
[image]

Locking the chip out of light sleep while using this peripheral does not work for my application. It's unfortunate that the driver is designed this way, since the fade hardware itself works perfectly fine in light sleep. Luckily, the `ledc_set_fade(...)` function bypasses the interrupt-driven part of the driver, so I was able to just copy over the `_ledc_set_fade_with_time(...)` and `_ledc_set_fade_with_step(...)` functions, change them to call into `ledc_set_fade(...)`, and then poll on `ledc_get_duty(...)` to ensure that the fade completes.

I'll attach the code here in case anyone needs to do something similar to work around this driver bug. Be aware that when I copied over those functions I had to manually hardcode in some of my LEDC driver parameters, since the functions that they call to retrieve those parameters are private.

```
static esp_err_t internal_ledc_set_fade_with_step(ledc_mode_t speed_mode, ledc_channel_t channel, uint32_t current_duty, uint32_t target_duty, int scale, int cycle_num) {
    if (current_duty == 1024) {
        current_duty -= 1;
    }
    int step_num = 0;
    ledc_duty_direction_t dir = LEDC_DUTY_DIR_DECREASE;
    if (scale > 0) {
        if (current_duty > target_duty) {
            step_num = (current_duty - target_duty) / scale;
            step_num = step_num > 1023 ? 1023 : step_num;
        } else {
            dir = LEDC_DUTY_DIR_INCREASE;
            step_num = (target_duty - current_duty) / scale;
            step_num = step_num > 1023 ? 1023 : step_num;
        }
    }

    if (scale > 0 && step_num > 0) {
        ledc_set_fade(speed_mode, channel, current_duty, dir, step_num, cycle_num, scale);
        ESP_LOGD(
            tag, 
            "cur duty: %d; target: %d, step: %d, cycle: %d; scale: %d; dir: %d\n",
            current_duty, target_duty, step_num, cycle_num, scale, dir
        );
    } else {
        ledc_set_fade(speed_mode, channel, target_duty, dir, 0, 1, 0);
        ESP_LOGD(
            tag, 
            "Set to target duty: %d", 
            target_duty
        );
    }
    return ESP_OK;
}

static esp_err_t internal_ledc_set_fade_with_time(ledc_mode_t speed_mode, ledc_channel_t channel, uint32_t current_duty, uint32_t target_duty, int max_fade_time_ms) {
    uint32_t freq = ledc_get_freq(speed_mode, LEDC_TIMER_0);
    uint32_t duty_delta = target_duty > current_duty ? target_duty - current_duty : current_duty - target_duty;

    if (duty_delta == 0) {
        return internal_ledc_set_fade_with_step(speed_mode, channel, current_duty, target_duty, 0, 0);
    }
    uint32_t total_cycles = max_fade_time_ms * freq / 1000;
    if (total_cycles == 0) {
        ESP_LOGW(tag, "LEDC Fade too fast");
        return internal_ledc_set_fade_with_step(speed_mode, channel, current_duty, target_duty, 0, 0);
    }
    int scale, cycle_num;
    if (total_cycles > duty_delta) {
        scale = 1;
        cycle_num = total_cycles / duty_delta;
        if (cycle_num > 0x3FF) {
            ESP_LOGW(tag, "LEDC Fade too slow");
            cycle_num = 0x3FF;
        }
    } else {
        cycle_num = 1;
        scale = duty_delta / total_cycles;
        if (scale > 0x3FF) {
            ESP_LOGW(tag, "LEDC Fade too fast");
            scale = 0x3FF;
        }
    }
    return internal_ledc_set_fade_with_step(speed_mode, channel, current_duty, target_duty, scale, cycle_num);
}

// Outside parameters
ledc_channel_t channel = ...;
uint32_t dutyCycleLast = ...;
uint32_t updateTimeMS = ...;
while (1) {
    uint32_t dutyCycle = ...;
    
    // Wait for any previous fade operations to finish...
    uint32_t dutyCycleReal = 0xFFFFFFFF;
    uint32_t dutyCycleRealLast = 0xFFFFFFFF;
    while (1) {
        dutyCycleReal = ledc_get_duty(
            LEDC_LOW_SPEED_MODE, 
            channel
        );
    
        // If the duty cycle has hit its target, the fade is complete
        if (dutyCycleReal == dutyCycleLast) {
            break;
        }
    
        // If the duty cycle is no longer changing without hitting the target, 
        // then a bug has occurred and the fade is complete but missed a step
        if (dutyCycleRealLast == dutyCycleReal) {
            break;
        }
        dutyCycleRealLast = dutyCycleReal;
    
        // Wait around a bit:
        // This lets us check whether the fade is still occurring,
        // while also re-entering light sleep and saving a bit of battery
        vTaskDelay(pdMS_TO_TICKS(10));
    }
    
    // Update our tracked duty cycle to match the real duty cycle,
    // in case any glitches occurred and the target duty cycle was not reached
    dutyCycleLast = dutyCycleReal;
    
    // No need to do anything if the duty cycle hasn't changed...
    if (dutyCycleLast != dutyCycle) {
    
        // Setup a fade to the new duty cycle
        esp_err_t errFade = internal_ledc_set_fade_with_time(
            LEDC_LOW_SPEED_MODE, 
            channel, 
            dutyCycleLast, 
            dutyCycle, 
            updateTimeMS
        );
        assert(errFade == ESP_OK);
    
        // Actually trigger the duty cycle update / fade in the hardware
        esp_err_t errUpdate = ledc_update_duty(LEDC_LOW_SPEED_MODE, channel);
        assert(errUpdate == ESP_OK);
    
        // Update our tracking with the new duty cycle
        dutyCycleLast = dutyCycle;
    }

    // Wait out the fade operation
    vTaskDelay(pdMS_TO_TICKS(updateTimeMS));
}
```

songruo, 2023-03-28:
Hi @therealergo,

Sorry, I want to clarify something I said wrong in the last post. Indeed, our hardware design is able to sync the interrupt status to the peripheral when the chip wakes up. The actual root cause of your issue is the chip goes into auto light sleep at `xSemaphoreTake(...)`, and the duration of the sleep in such case is forever. Therefore, the chip can never wake up, and interrupt can not be synced. (If you disable auto light sleep, but enable timer wakeup with `esp_sleep_enable_timer_wakeup()`, and call `esp_light_sleep_start()` by yourself, you can see the fade end interrupt being triggered after the chip wakes up.)

The solution to the original issue is the same though. We would recommend to use PM lock to force the chip not entering auto light sleep while fading. If you really care the current consumption, your workaround also works. I think you don't have to construct your own `internal_ledc_set_fade_with_step/time()` functions, the driver APIs `ledc_set_fade_with_step/time()` should work fine. The only thing is you can't use is `ledc_fade_start()`, instead, you can call `ledc_update_duty()` directly to skip the semaphore take.

raomin, 2023-04-19:
I did try what you suggested @songruo (or at least I believe so) but I get the same result as @therealergo. The fade with time seems to capture the the timer of and the sleep timer is not happening **if the fade timer is over**.
However, is the fade timer is longer than the sleep timer, then the sleep timer is honored.
This is fine for me, but I would be happy to understand why... :)

So this works only if fadeTime is longer than sleepTime:

```c
#include <stdio.h>
#include <Arduino.h>
#include "driver/ledc.h"
#include "esp_err.h"
#include "esp_pm.h"
#include "esp_sleep.h"
#include "driver/uart.h"
#include "esp32/rom/uart.h"

ledc_channel_config_t ledc_channel = {
        .gpio_num = 4,
        .speed_mode = LEDC_LOW_SPEED_MODE,
        .channel = LEDC_CHANNEL_2,
        .timer_sel = LEDC_TIMER_0,
        .duty = 0,
        .hpoint = 0,

};

void setup()
{
    Serial.begin(115200);

    ledc_timer_config_t ledc_timer = {
        .speed_mode = LEDC_LOW_SPEED_MODE,          // timer mode
        .duty_resolution = LEDC_TIMER_8_BIT, // resolution of PWM duty
        .timer_num = LEDC_TIMER_0,          // timer index
        .freq_hz = 100,                      // frequency of PWM signal
        .clk_cfg = LEDC_USE_RTC8M_CLK,       // Force source clock to RTC8M
    };
    ledc_timer_config(&ledc_timer);
    ledc_channel_config(&ledc_channel);
    printf("Frequency %u Hz\n", ledc_get_freq(LEDC_LOW_SPEED_MODE, LEDC_TIMER_0));
    
    ledc_fade_func_install(0);
}

bool direction = true;
void loop()
{
    uint16_t fadeTime = 7000;//Will only work if fadeTime is longer than sleepTime!
    uint32_t sleepTime = 5000000L;
    ledc_set_fade_with_time(ledc_channel.speed_mode,ledc_channel.channel , direction? 255:0, fadeTime);
    ledc_fade_start(ledc_channel.speed_mode,ledc_channel.channel , LEDC_FADE_NO_WAIT);
    printf("Sleep for %u s\n", sleepTime/1000000);
    esp_sleep_enable_timer_wakeup(sleepTime);
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_AUTO);
    fflush(stdout);
    uart_tx_wait_idle(CONFIG_CONSOLE_UART_NUM);
    esp_light_sleep_start();
    direction = !direction;
}
```

idea--list, 2023-05-07:
@therealergo 
In my case i had an RGB led flickering colors occasionally when automatic light sleep was enabled on ESP-IDF 4.4.3
Recently i ported my code to IDF 5.0.1 and found a new setting in the menuconfig called `RMT_ISR_IRAM_SAFE`. Ever since enabling that, the flickering issue is gone. Not sure if you are willing to update, but it might worth to try if it also solves fading issues.

pat-bert, 2025-03-07:
Hi @songruo,

I have the same use case as @therealergo. When I disable automatic light sleep the fade end interrupt callback is called correctly.
Once I enable automatic light sleep the fade works as expected but the interrupt is never triggered because all other tasks block indefinitely waiting for the end of the fade.
I'm gamma-fading the backlight of an LCD over the course of 15s from 100% to 0% and once it is off I'd like to put the LCD-IC into sleep mode to save some more power.

I was testing a workaround on ESP32C6 with ESP-IDF v5.4 based on:
-  connecting the LEDC_EVT_DUTY_CHNG_END_CH0 event to the event task matrix 
-  trigger an ETM task to set a GPIO level to low
- configure a GPIO interrupt on level low to wake-up the HP-core

This works as expected with automatic light sleep disabled (LED connected to GPIO turns off after fade completes) but not when the sleep mode is enabled.

`
esp_err_t ledc_new_etm_event(ledc_channel_t channel, const ledc_etm_event_config_t *config, esp_etm_event_handle_t *out_event)
{
    esp_etm_event_t *event = NULL;
    esp_err_t ret = ESP_OK;
    const char *TAG = "LEDC-ETM";
    uint32_t event_id{0U};
    ESP_GOTO_ON_FALSE(config && out_event, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
    ESP_GOTO_ON_FALSE(channel < LEDC_CHANNEL_MAX, ESP_ERR_INVALID_ARG, err, TAG, "invalid channel");
    ESP_GOTO_ON_FALSE(config->event_type < LEDC_ETM_EVENT_MAX, ESP_ERR_INVALID_ARG, err, TAG, "invalid event type");
    event = static_cast<esp_etm_event_t *>(heap_caps_calloc(1, sizeof(esp_etm_event_t), MALLOC_CAP_DEFAULT));
    ESP_GOTO_ON_FALSE(event, ESP_ERR_NO_MEM, err, TAG, "no memory for ETM event");

    event_id = LEDC_EVT_DUTY_CHNG_END_CH0 + channel - LEDC_CHANNEL_0;

    ESP_GOTO_ON_FALSE(event_id != 0, ESP_ERR_NOT_SUPPORTED, err, TAG, "not supported event type");

    // fill the ETM event object
    event->event_id = event_id;
    event->trig_periph = ETM_TRIG_PERIPH_GPTIMER;
    event->del = ledc_del_etm_event;
    *out_event = event;
    return ESP_OK;

err:
    if (event)
    {
        ledc_del_etm_event(event);
    }
    return ret;
}

static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    gpio_set_level(GPIO_NUM_15, 1);
}

void setUpBackLight()
{
    esp_etm_channel_config_t etmChannelConfig{};
    etmChannelConfig.flags.allow_pd = false;
    esp_etm_channel_handle_t etmChannelHandle{};
    ESP_ERROR_CHECK(esp_etm_new_channel(&etmChannelConfig, &etmChannelHandle));

    ledc_etm_event_config_t etmLedcEventConfig{};
    etmLedcEventConfig.event_type = LEDC_ETM_EVENT_DUTY_CHANGE_END;
    esp_etm_event_handle_t etmLedcEventHandle{};
    ESP_ERROR_CHECK(ledc_new_etm_event(static_cast<ledc_channel_t>(m_channel), &etmLedcEventConfig, &etmLedcEventHandle));

    gpio_etm_task_config_t etmGpioTaskConfig{};
    etmGpioTaskConfig.action = GPIO_ETM_TASK_ACTION_CLR;
    esp_etm_task_handle_t etmGpioTaskHandle{};
    ESP_ERROR_CHECK(gpio_new_etm_task(&etmGpioTaskConfig, &etmGpioTaskHandle));

    ESP_ERROR_CHECK(gpio_set_direction(GPIO_NUM_15, GPIO_MODE_INPUT_OUTPUT));
    ESP_ERROR_CHECK(gpio_set_level(GPIO_NUM_15, 1));
    ESP_ERROR_CHECK(gpio_sleep_sel_dis(GPIO_NUM_15));
    ESP_ERROR_CHECK(gpio_wakeup_enable(GPIO_NUM_15, GPIO_INTR_LOW_LEVEL));
    ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup());
    
    gpio_install_isr_service(0);
    gpio_isr_handler_add(GPIO_NUM_15, gpio_isr_handler, nullptr);
     
    gpio_intr_enable(GPIO_NUM_15);

    ESP_ERROR_CHECK(gpio_etm_task_add_gpio(etmGpioTaskHandle, GPIO_NUM_15));

    ESP_ERROR_CHECK(esp_etm_channel_connect(etmChannelHandle, etmLedcEventHandle, etmGpioTaskHandle));

    ESP_ERROR_CHECK(esp_etm_channel_enable(etmChannelHandle));

    REG_SET_BIT(LEDC_EVT_TASK_EN0_REG, LEDC_EVT_DUTY_CHNG_END_CH0_EN);

    esp_etm_dump(stdout);
}
`

The ETM Dump also confirms that the correct event and task ID are setup:
===========ETM Dump Start==========
channel 0: event 25 ==> task 9

The event source is enabled as verified by checking without light sleep.
On my oscilloscope I can also see that the duty cycle of the PWM pin drops to zero.
I tried different settings of the ETM channel allow_pd flag and explicitly setting the PD_DOMAIN_TOP to ON without luck.

Do you have any hints why the ETM is no longer working as expected once automatic light sleep is involved?
Perhaps the fade does not finish precisely at zero and therefore never triggers the ETM event?

songruo, 2025-03-12:
@pat-bert I think it is a very good application! However, unfortunately, right now the ETM module is not able to keep working during sleep as its core clock source can only be AHB clock, which is off in sleep. We will consider this usage in our future chips. (BTW, ETM feature is unrelated to interrupts :))

pat-bert, 2025-03-12:
@songruo thanks for the clarification!