espressif/esp-idf #11008: Target-dependent sdkconfig defaults missing when no .chipset file present

gojimmypi, 2023-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5 release and master c77b5752ef

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

As described in the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#target-dependent-sdkconfig-defaults):

> _For example, for `esp32` target, default settings will be taken from `sdkconfig.defaults` first, and then from `sdkconfig.defaults.esp32`_

I would have expected the if no `sdkconfig.defaults.esp32` file was present, that the defaults from `sdkconfig.defaults` would still apply.

### What is the actual behavior?

In my case, I was targeting an `esp32s3`. I had a `sdkconfig.defaults` file, but I did *not* have a `sdkconfig.defaults.esp32s3` file. 

Settings from the `sdkconfig.defaults` did not make it to me `sdkconfig.h` file unless there was also a `sdkconfig.defaults.esp32s3` with the same settings.

I would have expected that if there's no `sdkconfig.defaults.esp32s3` file, that the defaults from `sdkconfig.defaults` would still apply.

### Steps to reproduce.

See my [sample WIP TLS client](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF5/examples/wolfssl_client)

With the `sdkconfig.defaults.esp32s3` present, the code compiles successfully.

Delete the `sdkconfig.defaults.esp32s3` (with the exact same defaults as `sdkconfig.defaults` and the code fails to compile.

```
[1020/1032] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/__/__/main/wifi_connect.c.obj
FAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/__/__/main/wifi_connect.c.obj 
C:\SysGCC\esp32\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1-dev-4033-gc77b5752ef-dirty\" -DLIBWOLFSSL_VERSION_GIT_BRANCH=\"ED25519_SHA2_fix\" -DLIBWOLFSSL_VERSION_GIT_HASH=\"f0aa965dbfa515f717d373857a9795c5a63ecaa5\" -DLIBWOLFSSL_VERSION_GIT_HASH_DATE="\"'Thu Mar 16 17:42:58 2023 -0700'\"" -DLIBWOLFSSL_VERSION_GIT_ORIGIN=\"https://github.com/gojimmypi/wolfssl.git\" -DLIBWOLFSSL_VERSION_GIT_SHORT_HASH=\"f0aa965db\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -Iconfig -I../../../components/wolfssl -I../../../components/wolfssl/include -IC:/SysGCC/esp32/esp-idf/master/components/freertos/FreeRTOS-Kernel/include/freertos -IC:/workspace/wolfssl-gojimmypi -IC:/SysGCC/esp32/esp-idf/v5.0/components/esp_partition/include -IC:/SysGCC/esp32/esp-idf/v5.0/components/nvs_flash/include -IC:/SysGCC/esp32/esp-idf/master/components/newlib/platform_include -IC:/SysGCC/esp32/esp-idf/master/components/freertos/FreeRTOS-Kernel/include -IC:/SysGCC/esp32/esp-idf/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/SysGCC/esp32/esp-idf/master/components/freertos/esp_additions/include/freertos -IC:/SysGCC/esp32/esp-idf/master/components/freertos/esp_additions/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_hw_support/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_hw_support/include/soc -IC:/SysGCC/esp32/esp-idf/master/components/esp_hw_support/include/soc/esp32s3 -IC:/SysGCC/esp32/esp-idf/master/components/esp_hw_support/port/esp32s3/. -IC:/SysGCC/esp32/esp-idf/master/components/esp_hw_support/port/esp32s3/private_include -IC:/SysGCC/esp32/esp-idf/master/components/heap/include -IC:/SysGCC/esp32/esp-idf/master/components/log/include -IC:/SysGCC/esp32/esp-idf/master/components/soc/include -IC:/SysGCC/esp32/esp-idf/master/components/soc/esp32s3 -IC:/SysGCC/esp32/esp-idf/master/components/soc/esp32s3/include -IC:/SysGCC/esp32/esp-idf/master/components/hal/esp32s3/include -IC:/SysGCC/esp32/esp-idf/master/components/hal/include -IC:/SysGCC/esp32/esp-idf/master/components/hal/platform_port/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_rom/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_rom/include/esp32s3 -IC:/SysGCC/esp32/esp-idf/master/components/esp_rom/esp32s3 -IC:/SysGCC/esp32/esp-idf/master/components/esp_common/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_system/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_system/port/soc -IC:/SysGCC/esp32/esp-idf/master/components/esp_system/port/include/private -IC:/SysGCC/esp32/esp-idf/master/components/xtensa/include -IC:/SysGCC/esp32/esp-idf/master/components/xtensa/esp32s3/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/include/apps -IC:/SysGCC/esp32/esp-idf/master/components/lwip/include/apps/sntp -IC:/SysGCC/esp32/esp-idf/master/components/lwip/lwip/src/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/port/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/port/freertos/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/port/esp32xx/include -IC:/SysGCC/esp32/esp-idf/master/components/lwip/port/esp32xx/include/arch -IC:/SysGCC/esp32/esp-idf/master/components/esp_wifi/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_wifi/wifi_apps/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_event/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_phy/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_phy/esp32s3/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_netif/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_timer/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/deprecated -IC:/SysGCC/esp32/esp-idf/master/components/driver/dac/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/gpio/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/gptimer/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/i2c/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/i2s/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/ledc/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/mcpwm/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/parlio/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/pcnt/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/rmt/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/sdio_slave/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/sdmmc/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/sigma_delta/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/spi/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/temperature_sensor/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/touch_sensor/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/twai/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/uart/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/usb_serial_jtag/include -IC:/SysGCC/esp32/esp-idf/master/components/driver/touch_sensor/esp32s3/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_pm/include -IC:/SysGCC/esp32/esp-idf/master/components/esp_ringbuf/include -mlongcalls  -DWOLFSSL_USER_SETTINGS -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fstack-protector -fmacro-prefix-map=C:/workspace/wolfssl-gojimmypi/IDE/Espressif/ESP-IDF5/examples/wolfssl_client=. -fmacro-prefix-map=C:/SysGCC/esp32/esp-idf/master=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/__/__/main/wifi_connect.c.obj -MF esp-idf\wolfssl\CMakeFiles\__idf_wolfssl.dir\__\__\main\wifi_connect.c.obj.d -o esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/__/__/main/wifi_connect.c.obj -c ../../../main/wifi_connect.c
../../../main/wifi_connect.c: In function 'wifi_init_sta':
[... cut ...]
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../main/wifi_connect.c:341:35: note: each undeclared identifier is reported only once for each function it appears in
[1021/1032] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
```

The failure is due to `ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD` not defined, which should have been captured if `CONFIG_ESP_WIFI_AUTH_WPA2_PSK` was defined in `sdkconfig.h`

### More Information.

my specific examples are using VisualGDB

ESP-Marius, 2023-03-17:
Hi @gojimmypi

It should indeed work the way you describe (you dont need a target specific sdkconfig.defaults.X). Are you sure that there is not something else wrong here?

As far as I can see `CONFIG_ESP_WIFI_AUTH_WPA2_PSK` is not an actual kconfig value (it is only defined when building our antenna example) so it will indeed not show up in your `sdkconfig.h` as unknown values will be discarded when running menuconfig.

This doesnt explain why you can suddenly build when `sdkconfig.defaults.esp32s3` is present, but maybe something else got mixed up while testing this?

gojimmypi, 2023-03-17:
Hello @ESP-Marius and thank you for your prompt reply.

I believe your are correct: I've confirmed `CONFIG_ESP_WIFI_AUTH_WPA2_PSK` is not a kconfig value.

I've since been unable to reproduce this. I must have been mistaken.

closed 2023-03-17 by gojimmypi