espressif/esp-idf #11009: Linking CXX executable .elf file failed

mauriceee31, 2023-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My Terminal looks like this: 
`C:\Users\m.lenze\FBO>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\m.lenze\fbo\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D C:\Users\m.lenze\FBO\build\bootloader\esp-idf\esptool_py &&...fset 0x8000 bootloader 0x1000 C:/Users/m.lenze/FBO/build/bootloader/bootloader.bin"Bootloader binary size 0x6de0 bytes. 0x220 bytes (2%) free.
[2/4] Linking CXX executable FBO_Controller.elfFAILED: FBO_Controller.elf 
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\FBO_Controller.elf.rsp -o FBO_Controller.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(periph.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: multiple definition of `filter_status'; esp-idf/main/libmain.a(main.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: first defined here
c:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(serial.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: multiple definition of `filter_status'; esp-idf/main/libmain.a(main.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: first defined here
c:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(filter.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: multiple definition of `filter_status'; esp-idf/main/libmain.a(main.c.obj):C:/Users/m.lenze/FBO/main/filter.h:26: first defined here
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the c:\users\m.lenze\fbo\build\log\idf_py_stderr_output_124 and c:\users\m.lenze\fbo\build\log\idf_py_stdout_output_124`

Can anybody help me with this? I tried doing the stuff form this issue but i made got [https://github.com/espressif/esp-va-sdk/issues/148]

ESP-Marius, 2023-03-17:
@mauriceee31, what does your `filter.h` look like? To me this doesnt seem like an ESP-IDF issue, but rather something related to your code.

mauriceee31, 2023-03-17:
My filter.h looks like this:

mauriceee31, 2023-03-17:
My filter.h looks like this: `
#pragma once
#include <stdint.h>
#include "stdbool.h"

typedef struct {
	bool bWasDPWarning;
	bool bWasDPAlarm;

	bool bIsWaterSlug;
	bool bIsDPWarning;
	bool bIsDPAlarm;
	bool bRelayActivated;
	uint32_t countWaterSlug;

} filter_status_t;

	filter_status_t filter_status;

void filter_changeout();
void filter_init();
void filter_load_nvs();
void filter_store_nvs();

void filter_print_filter_data();
`

ESP-Marius, 2023-03-17:
If you declare a variable in your header file and include the header file in multiple places you will get a [multiple definition error](https://stackoverflow.com/questions/14526153/multiple-definition-of-a-global-variable).

mauriceee31, 2023-03-17:
Tried to erase those definitions but then i get this error:
C:/Users/m.lenze/FBO/main/serial.c:82:33: error: implicit declaration of function 'filter_print_filter_data' [-Werror=implicit-function-declaration]
   82 |                                 filter_print_filter_data();
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/m.lenze/FBO/main/serial.c:85:33: error: implicit declaration of function 'filter_changeout' [-Werror=implicit-function-declaration]
   85 |                                 filter_changeout();
      |                                 ^~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import filter_store_nvs library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the c:\users\m.lenze\fbo\build\log\idf_py_stderr_output_18232 and c:\users\m.lenze\fbo\build\log\idf_py_stdout_output_18232

ESP-Marius, 2023-03-17:
As the error says, you are trying to use a function that hasnt been declared. Did you include the header file?

Github issues are intended for reporting bugs/feature requests related to ESP-IDF, for more generic questions like this I suggest trying our [forum](https://esp32.com/) or [stackoverflow](https://stackoverflow.com/).

If you run into any ESP-IDF specific issues then feel free to open another issue.

closed 2023-03-17 by ESP-Marius