espressif/esp-idf #11010: USB Host: Setting configuration descriptor and then claiming interface

maxslarsson, 2023-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am developing a program that uses the ESP32 as a USB host. However, I ran into an issue when trying to switch to another configuration descriptor. As the API reference [states](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_host.html#devices):

> During enumeration, each device is configured to use configuration 1.

But I need to be able to use configuration 4 for my device and interact with its endpoints. I could not find any function that automatically sets the configuration, so I used `USB_SETUP_PACKET_INIT_SET_CONFIG` to create the control transfer setup packet and sent it to the device. I verified that the device changed its configuration. However, `usb_host_get_active_config_descriptor` doesn't update, it still references configuration 1 (since it was cached on enumeration). Thus when I run `usb_host_interface_claim` it fails with the error `ESP_ERR_NOT_FOUND` because the `usb_host_interface_claim` function claims the interface on the configuration that is stored in the internal cache.

**In summary, the problem I'm facing is that the ESP32 is unable to claim interfaces that are on a configuration other than 1.**

There is a function in the API that does exactly what I need ([`usbh_hub_enum_fill_config_desc`](https://github.com/espressif/esp-idf/blob/c77b5752ef9e4e40c2e819d511b1dd7b83be2996/components/usb/usbh.c#L986)). However, that function is part of the internal usbh.c api and so I am not able to call it from my code. A solution I thought of was to write a function called `usb_host_set_active_config_descriptor` or `usb_host_set_config_descriptor` in the usb_host API that both sends the control transfer to the device to change its configuration and also updates the internal cache. I would be happy to create a pull request if that sounds like a good solution!

Thank you so much for all of your guys's help!

Dazza0, 2023-03-17:
@maxslarsson  Thanks for reporting this issue. Using other configuration isn't currently supported yet by the USB Host stack.

As a quick workaround, does changing the `ENUM_CONFIG_INDEX` in [`hub.c`](https://github.com/espressif/esp-idf/blob/v5.0.1/components/usb/hub.c#L37) to `3` work (it should basically lock the selected configuration to `4` for all devices at compile time).

Supporting run time configuration change won't be trivial as we need ensure that all other clients are not using the same device (i.e., we should not reconfigure the device if there is another class driver using that device). But this feature is on our roadmap.

maxslarsson, 2023-03-17:
Thank you for you response! Unfortunately, my code checks all the config descriptors for the right one and therefore needs to set the configuration descriptor at runtime. However, I only have one class driver and so I don't need to check if another class driver is using that device. So is there a good workaround given that I am only going to have one class driver?

The workaround I am thinking of using is to make the [`usbh_hub_enum_fill_config_desc`](https://github.com/espressif/esp-idf/blob/c77b5752ef9e4e40c2e819d511b1dd7b83be2996/components/usb/usbh.c#L986) function public through the `usb_host.h` file. However, I would love to contribute to the project in any way that I can. Is there any way runtime configuration descriptor could be supported?

Another workaround I came up with is to specify the configuration descriptor when opening the device. So for example modifying the `usb_host_device_open` function to take another argument that is the configuration descriptor to open on enumeration. However, I wasn't able to fully track the device enumeration process (where the entire state machine in `hub.c` runs and the device configuration 1 is opened). It seemed to happen automatically when the device is connected (before the `usb_host_device_open` function is called) which would mean that this would not possible. Is there any chance that I missed it and it occurs when `usb_host_device_open` is called? But does this face the same issue of not ensuring that all client drivers are using the same configuration for the device?

peter-marcisovsky, 2024-02-06:
Hello @maxslarsson

A new feature - setting a device configuration during enumeration was implemented by [this](https://github.com/espressif/esp-idf/commit/24adb6ed3c87664275a5a3992cb27149541cd436) commit in esp-idf, also a chapter in [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_host.html#enumeration-filter) was added explaining the feature. The feature has been added to our USB Host Lib [example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) to present it's usage.

Now, a user is provided with a device descriptor via a callback function and has an ability to set a configuration of a device during the device enumeration. 

In addition, there is an ability to filter through attached USB devices and choose which device to enumerate and which one not to. However, changing a configuration during run time, as already mentioned, still has not been implemented, but it is planned.