espressif/esp-idf #11013: BROWNOUT esp32c6

RuslanSpartak, 2023-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to know how much maximum current esp32c6 use when all is enabled, like periphery, ble and wifi. Because after a while I have brownouts reboots.

esp-lis, 2023-03-17:
You can try to modify the brownout threshold to increase the threshold voltage, the location of `brownout volatage level` option is **(TOP) -> Component config -> ESP System Settings ->Brownout Detector -> Hardware brownout detect & reset -> Brownout voltage level**.

RuslanSpartak, 2023-03-17:
I disable that. And it still brownout.

mythbuster5, 2023-03-22:
So... Can you show your logs after you disabled brownout but there is still a brownout reset? I can't see this on my board @RuslanSpartak

krzychb, 2023-03-22:
> I want to know how much maximum current esp32c6 use when all is enabled, like periphery, ble and wifi.

Hi @RuslanSpartak, [according to ESP32-C6 Dataseet](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf) the power supply to the chip should provide at least 500 mA. If there are other devices connected to the board then respective current requirements should be added on top of that.

[image]

RuslanSpartak, 2023-03-24:
> So... Can you show your logs after you disabled brownout but there is still a brownout reset? I can't see this on my board @RuslanSpartak

Sorry for late responce. So we can disable detecting brownout. But I still get brownout reboots.
[image]

mythbuster5, 2023-03-30:
> Any updates on this issue?

Still in progress, will inform you if there is any updates.

mythbuster5, 2023-04-04:
@RuslanSpartak Hi, this issue is pretty weird. I want to get more information? What modules are you using? Do you both use wifi and bluetooth together?

RuslanSpartak, 2023-04-04:
> @RuslanSpartak Hi, this issue is pretty weird. I want to get more information? What modules are you using? Do you both use wifi and bluetooth together?

I'm using esp32-c6-devkitc-1. Also module from it. I use wifi and ble together (blufi).

RuslanSpartak, 2023-05-05:
So any updates with brownout issue?

RuslanSpartak, 2023-05-05:
[image]
I time to time have brownouts with latest master sdk

AxelLin, 2023-09-19:
> > Any updates on this issue?
> 
> Still in progress, will inform you if there is any updates.

@mythbuster5 @mahavirj  Any new finding?

ginkgm, 2023-09-27:
Hi @RuslanSpartak @AxelLin ,

We suspect this issue is fixed in latest PHY, please upgrade IDF to HEAD of IDF release/v5.1 and have a try again.

RuslanSpartak, 2023-09-27:
> Hi @RuslanSpartak @AxelLin ,
> 
> We suspect this issue is fixed in latest PHY, please upgrade IDF to HEAD of IDF release/v5.1 and have a try again.

Yes. All is fine with latest sdk. Thanks.

closed 2023-09-27 by RuslanSpartak