espressif/esp-idf #11014: component building with compile flag "-fshort-enums"

mejha, 2023-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

az-delivery-devkit-v4

### Power Supply used.

USB

### What is the expected behavior?

Trying to build a component with build flag "-fshort-enums" and running uart echo example on the device.

### What is the actual behavior?

The build process finishes fine, but the uart driver could not be inicialized.

### Steps to reproduce.

1. Step
Create new example project "UART Echo Example", this can be done through VS code. Sources of this example are available at: [link](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo)

2. Step
Add line of code to bottom of the file in main/CMakeLists.txt

      `component_compile_options(-fshort-enums)`

3. Step
Build and Flash.

4. Step
Open monitor window and you can see a crash when calling `uart_param_config ` function.

### Debug Logs.

```plain
ELF file SHA256: 045d9a5ca43d7a50

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6940
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (29) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (29) boot: compile time 21:35:51
I (29) boot: chip revision: v3.0
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a024h ( 40996) map
I (112) esp_image: segment 1: paddr=0001a04c vaddr=3ffb0000 size=01ed4h (  7892) load
I (115) esp_image: segment 2: paddr=0001bf28 vaddr=40080000 size=040f0h ( 16624) load
I (125) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18f0ch (102156) map
I (163) esp_image: segment 4: paddr=00038f34 vaddr=400840f0 size=086e4h ( 34532) load
I (184) boot: Loaded app from partition at offset 0x10000
I (184) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (196) cpu_start: Starting app cpu, entry point is 0x40081140
0x40081140: call_start_cpu1 at C:/Users/Miha/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (181) cpu_start: App cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 160000000 Hz
I (210) cpu_start: Application information:
I (215) cpu_start: Project name:     uart_echo
I (220) cpu_start: App version:      1
I (224) cpu_start: Compile time:     Mar 17 2023 21:35:28
I (230) cpu_start: ELF file SHA256:  045d9a5ca43d7a50...
I (236) cpu_start: ESP-IDF:          v5.0.1
I (241) cpu_start: Min chip rev:     v0.0
I (246) cpu_start: Max chip rev:     v3.99 
I (251) cpu_start: Chip rev:         v3.0
I (256) heap_init: Initializing. RAM available for dynamic allocation:
I (263) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (269) heap_init: At 3FFB27D0 len 0002D830 (182 KiB): DRAM
I (275) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (281) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (288) heap_init: At 4008C7D4 len 0001382C (78 KiB): IRAM
I (296) spi_flash: detected chip: generic
I (299) spi_flash: flash io: dio
W (303) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (317) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (0) uart: uart_param_config(740): data bit error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d5eed
0x400d5eed: echo_task at C:/Users/Miha/Desktop/Newf/aaa/uart_echo/main/uart_echo_example_main.c:61 (discriminator 1)

file: "./main/uart_echo_example_main.c" line 61
func: echo_task
expression: uart_param_config(ECHO_UART_PORT_NUM, &uart_config)

abort() was called at PC 0x400865e7 on core 1
0x400865e7: _esp_error_check_failed at C:/Users/Miha/esp/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x40081bd6:0x3ffb5f50 0x400865f1:0x3ffb5f70 0x4008b39a:0x3ffb5f90 0x400865e7:0x3ffb6000 0x400d5eed:0x3ffb6030 0x40089025:0x3ffb6060
0x40081bd6: panic_abort at C:/Users/Miha/esp/esp-idf/components/esp_system/panic.c:423

0x400865f1: esp_system_abort at C:/Users/Miha/esp/esp-idf/components/esp_system/esp_system.c:153

0x4008b39a: abort at C:/Users/Miha/esp/esp-idf/components/newlib/abort.c:38

0x400865e7: _esp_error_check_failed at C:/Users/Miha/esp/esp-idf/components/esp_system/esp_err.c:47

0x400d5eed: echo_task at C:/Users/Miha/Desktop/Newf/aaa/uart_echo/main/uart_echo_example_main.c:61 (discriminator 1)

ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
Waiting for the device to reconnect0x40089025: vPortTaskWrapper at C:/Users/Miha/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

### More Information.

Are there any workarounds on this?
I would like to build a custom component with above mentioned compile argument that contains a uart inicialization.

igrr, 2023-03-17:
What you observe is probably an ABI mismatch between the two object files:
- the file which calls UART driver init function
- the file where the UART driver is implemented.

This issue is noted in [GCC manual](https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html):

> Warning: the -fshort-enums switch causes GCC to generate code that is not binary compatible with code generated without that switch.

In other words, the numeric values of the same enum member are different between your source file (compiled with -fshort-enums) and the UART driver source file (compiled without this option).

I'm afraid the only option you have to keep the part of the code which initializes the UART driver in a separate source file from the rest of your component code, and don't add -fshort-enums when compiling that source file.

For example:

- In yourcomponent/init_uart.c: define a function init_uart, which calls IDF API to initialize the uart driver. Compile this source file without -fshort-enums flag.
- In yourcomponent/other.c: the rest of your code which is compiled with -fshort-enums. Call init_uart function defined in init_uart.c. Don't use IDF APIs which take enum arguments from this file.

mejha, 2023-03-18:
@igrr thank you for such quick response. Yes, you are right.
I made separate source file as you suggested, inicialization works as expected.

closed 2023-03-18 by mejha

rojer, 2025-06-05:
@igrr would you consider enabling `-fshort-enums` globally? as users, we can't really do it because of the wifi/bt blob ABI