espressif/esp-idf #11017: How to use ESP32C6 LP CPU in ESP-IDF

dreamcmi, 2023-03-18:
### Is your feature request related to a problem?

How to use ESP32C6 LP CPU in ESP-IDF？

### Describe the solution you'd like.

At present, the document does not seem to provide relevant instructions for the LP core of ESP32C6. I would like to ask in what way the LP core will be called by the user in the future update, whether to use `xTaskCreatePinnedToCore` like the ESP32S series, or to load it like ULP?

ESP-Marius, 2023-03-21:
In the initial version of the support the API will be similar to the old ULPs, i.e. a separate binary compiled and loaded in to LP RAM. 

But in the future we'll also look into ways of using it as a more generic coprocessor, e.g. through `xTaskCreatePinnedToCore`, `esp_ipc` or other suitable APIs.

dreamcmi, 2023-04-02:
@ESP-Marius May I ask when will the example about the ulp method be supported?

ESP-Marius, 2023-04-03:
Basic loading and  running of a binary is already supported. Sleep and i2c support are under review. So in a few weeks I have most of the basic features will be supported.

mickeyl, 2023-05-31:
Answering myself. It looks like it. Here's the current build problem w/ ESP-IDF 5.1-beta:

```sh
Executing action: all (aliases: build)
Running ninja in directory /home/mickey/Documents/late/esp-tests/ulp_gpio/build
Executing "ninja all"...
[1/234] Performing build step for 'ulp_main'FAILED: esp-idf/main/ulp_main-prefix/src/ulp_main-stamp/ulp_main-build esp-idf/main/ulp_main/ulp_main.bin esp-idf/main/ulp_main/ulp_main.ld esp-idf/main/ulp_main/ulp_main.h esp-idf/main/ulp_main/ulp_main.map esp-idf/main/ulp_main/ulp_main.sym esp-idf/main/ulp_main/esp32.ulp.ld esp-idf/main/ulp_main/ulp_main 
cd /home/mickey/Documents/late/esp-tests/ulp_gpio/build/esp-idf/main/ulp_main && /usr/bin/cmake --build /home/mickey/Documents/late/esp-tests/ulp_gpio/build/esp-idf/main/ulp_main --target build
[1/5] Building C object CMakeFiles/ulp_main.dir/home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c.objFAILED: CMakeFiles/ulp_main.dir/home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c.obj 
/home/mickey/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc  -I/home/mickey/Documents/late/esp-tests/ulp_gpio/build/config -I/home/mickey/Documents/late/misc/esp-idf/components/newlib/platform_include -I/home/mickey/Documents/late/misc/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/mickey/Documents/late/misc/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/mickey/Documents/late/misc/esp-idf/components/freertos/esp_additions/include/freertos -I/home/mickey/Documents/late/misc/esp-idf/components/freertos/esp_additions/include -I/home/mickey/Documents/late/misc/esp-idf/components/freertos/esp_additions/arch/riscv/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_hw_support/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_hw_support/include/soc -I/home/mickey/Documents/late/misc/esp-idf/components/esp_hw_support/include/soc/esp32c6 -I/home/mickey/Documents/late/misc/esp-idf/components/esp_hw_support/port/esp32c6/. -I/home/mickey/Documents/late/misc/esp-idf/components/esp_hw_support/port/esp32c6/private_include -I/home/mickey/Documents/late/misc/esp-idf/components/heap/include -I/home/mickey/Documents/late/misc/esp-idf/components/log/include -I/home/mickey/Documents/late/misc/esp-idf/components/soc/include -I/home/mickey/Documents/late/misc/esp-idf/components/soc/esp32c6 -I/home/mickey/Documents/late/misc/esp-idf/components/soc/esp32c6/include -I/home/mickey/Documents/late/misc/esp-idf/components/hal/esp32c6/include -I/home/mickey/Documents/late/misc/esp-idf/components/hal/include -I/home/mickey/Documents/late/misc/esp-idf/components/hal/platform_port/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_rom/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_rom/include/esp32c6 -I/home/mickey/Documents/late/misc/esp-idf/components/esp_rom/esp32c6 -I/home/mickey/Documents/late/misc/esp-idf/components/esp_common/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_system/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_system/port/soc -I/home/mickey/Documents/late/misc/esp-idf/components/esp_system/port/include/riscv -I/home/mickey/Documents/late/misc/esp-idf/components/esp_system/port/include/private -I/home/mickey/Documents/late/misc/esp-idf/components/riscv/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/include/apps -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/include/apps/sntp -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/lwip/src/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/port/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/port/freertos/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/port/esp32xx/include -I/home/mickey/Documents/late/misc/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/mickey/Documents/late/misc/esp-idf/components/nvs_flash/include -I/home/mickey/Documents/late/misc/esp-idf/components/spi_flash/include -I/home/mickey/Documents/late/misc/esp-idf/components/esp_partition/include -I/home/mickey/Documents/late/misc/esp-idf/components/ulp/ulp_common/include -I/home/mickey/Documents/late/misc/esp-idf/components/ulp/ulp_common/include/esp32c6 -I/home/mickey/Documents/late/misc/esp-idf/components/ulp/lp_core/include -I/home/mickey/Documents/late/misc/esp-idf/components/ulp/lp_core/shared/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/deprecated -I/home/mickey/Documents/late/misc/esp-idf/components/driver/analog_comparator/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/dac/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/gpio/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/gptimer/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/i2c/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/i2s/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/ledc/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/mcpwm/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/parlio/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/pcnt/include -I/home/mickey/Documents/late/misc/esp-idf/components/driver/rmt/include -I/home [... cut ...] Files/ulp_main.dir/home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c.obj.d -o CMakeFiles/ulp_main.dir/home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c.obj   -c /home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c
/home/mickey/Documents/late/esp-tests/ulp_gpio/main/ulp/main.c:15:10: fatal error: ulp_riscv.h: No such file or directory
   15 | #include "ulp_riscv.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

ESP-Marius, 2023-08-07:
Marking this as closed now that several lp core examples are available in IDF: https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/lp_core

closed 2023-08-07 by ESP-Marius

andylinpersonal, 2024-06-18:
Hi @dreamcmi
LP-Core is indeed able to run its own instance of FreeRTOS :)
But it cannot do too many things with such restricted LP SRAM space.
P4's LP Core should be more capable in this case.
See my port https://github.com/andylinpersonal/esp-idf/tree/dev/lp-core-freertos

ESP-Marius, 2024-06-18:
Cool example @andylinpersonal! 

As you say, it is indeed possible to run it, but the use-cases are probably limited. I think a super-loop approach is still probably the most suitable for a resource constrained system like the LP-Core.

andylinpersonal, 2024-06-18:
Hi @ESP-Marius ,
I'm trying to optimize the tick handler of my LP Core FreeRTOS port.  For the interrupt CSRs, mie and mip, there are "soft interrupt" and "timer interrupt" shown in the TRM. According to RISC-V privileged ISA specification, that timer interrupt is controlled by “machine-level memory mapped timer” and soft interrupt is controlled by "memory-mapped interrupt controller". But there are no such things for LP core. How can I trigger these interrupts? Are they simply hardwired and will never be triggered?

ESP-Marius, 2024-06-18:
@andylinpersonal you are indeed correct that this feature is actually not implemented on the LP CPU. I've notified the doc-team that we should clean up these mentions in the TRM to avoid users getting the wrong impression.

andylinpersonal, 2024-06-18:
Hi @ESP-Marius:
Btw, the "`ecall` from the M mode" exception (mcause = 11) of LP Core is functional but missing from the TRM. It usually can be used to implement context switch.

ESP-Marius, 2024-06-19:
@andylinpersonal, it is mentioned here:

[image]

Or were you referring to something else?

andylinpersonal, 2024-06-19:
Ohh Ok. The register listing doesn't mention this exception type.

bitbank2, 2024-07-28:
> Marking this as closed now that several lp core examples are available in IDF: https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/lp_core

The examples compile + work on the S2 and S3, but fail to build with the same error as above on the C6. I'm using the latest (v5.3) ESP-IDF. Can you re-open this issue until a fix is available?

ESP-Marius, 2024-07-29:
@bitbank2 make sure you are compiling the examples in the path mentioned above, there are separate folders for the new ULP (lp-core) and the older one (ulp-riscv) for S2/S3

bitbank2, 2024-07-29:
> @bitbank2 make sure you are compiling the examples in the path mentioned above, there are separate folders for the new ULP (lp-core) and the older one (ulp-riscv) for S2/S3

Thanks for pointing me to the correct set of examples. I haven't seen any documentation which clarifies which code to use for which target processor. It's even more confusing because they all have risc-v low power coprocessors. Did I miss some memo? :)