espressif/esp-idf #11019: Inconsistent websocket event data length delivery

theodem, 2023-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom ESP32-S3 board

### Power Supply used.

USB

### What is the expected behavior?

I have two ESP32-S3 communicating using websockets, one ESP32 is the WS server while the other is the Ws client.
I am using ESP IDF release/5.0 and esp_websocket_client 1.0.0

I am using 12288 as the WS client buffer value.

While sending a binary frame of 8204 bytes from the ESP websocket server, I should receive only one data event from the client with a length of 8204 as buffer is 12288 bytes.

### What is the actual behavior?

While sending a binary frame of 8204 bytes from the ESP websocket server, I receive inconsistent event data amount, sometimes only one data event of 8204 (which is the expected behaviour) but sometimes I receive that payload splitted in 2 to 5 different data events.

### Steps to reproduce.

**-- SERVER SIDE CODE --**
```
static esp_err_t on_websocket_url(httpd_req_t *req)
{
    if (req->method == HTTP_GET)
    {
        websocket_client_id = httpd_req_to_sockfd(req);
        ESP_LOGI(TAG, "New connection opened, client ID: %d", websocket_client_id);

        uint8_t *buffer = new uint8_t[14 * 586];

        for (size_t frame_index = 0; frame_index < 586; frame_index++)
        {
            buffer[frame_index * 14 + 0] = 0x1F;
            buffer[frame_index * 14 + 1] = 0xFF;
            buffer[frame_index * 14 + 2] = 0xFF;
            buffer[frame_index * 14 + 3] = 0xFF;

            buffer[frame_index * 14 + 4] = 1;
            buffer[frame_index * 14 + 5] = 8;

            buffer[frame_index * 14 + 6] = 0xF1;
            buffer[frame_index * 14 + 7] = 0xF2;
            buffer[frame_index * 14 + 8] = 0xF3;
            buffer[frame_index * 14 + 9] = 0xF4;
[... 289 lines cut ...]
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: xxx==
D (4737) transport_ws: Read header chunk 129, current header size: 129
D (4738) transport_ws: HTTP status code is 101
D (4738) transport_ws: server key=xxx=, send_key=xxx==, expected_server_key=xxx=
D (4751) websocket_client: Transport connected to ws://192.168.1.150:80
D (4758) event: running post WEBSOCKET_EVENTS:1 with handler 0x4200ad94 and context 0x3fce46b0 on loop 0x3fce4378
0x4200ad94: uart_get_baudrate at /Users/xxx/Documents/esp-idf/components/driver/uart.c:321 (discriminator 2)
D (4770) transport_ws: Opcode: 2, mask: 0, len: 126
D (4775) event: running post WEBSOCKET_EVENTS:3 with handler 0x4200ad94 and context 0x3fce46b0 on loop 0x3fce4378
0x4200ad94: uart_get_baudrate at /Users/xxx/Documents/esp-idf/components/driver/uart.c:321 (discriminator 2)
I (4784) WS CLIENT: Received 7296 / 8204 bytes
D (4789) event: running post WEBSOCKET_EVENTS:3 with handler 0x4200ad94 and context 0x3fce46b0 on loop 0x3fce4378
0x4200ad94: uart_get_baudrate at /Users/xxx/Documents/esp-idf/components/driver/uart.c:321 (discriminator 2)
I (4799) WS CLIENT: Received 908 / 8204 bytes
I (5205) WS CLIENT: Websocket connected !
D (5805) transport_base: poll_read: select - Timeout before any socket was ready!
V (5805) websocket_client: Read poll timeout: skipping esp_transport_read()...
```

### More Information.

I was not really able to know if the issue come from the client or the server.

Sometime one **data event length is 1460 which is my MSS** so maybe something to dig there.

theodem, 2023-03-19:
I dont know why, but I usually get more splitted data when I have log level to INFO than with log level VERBOSE.

Here is a log with log INFO:

I (2660) WIFI: Got IP:192.168.1.73
I (2662) WIFI: Connected to: xxx
W (2667) websocket_client: `reconnect_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
W (2680) websocket_client: `network_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
I (2694) WS CLIENT: Connecting to Websocket endpoint..
I (3461) WS CLIENT: Received 2916 / 8204 bytes
I (3535) WS CLIENT: Received 1460 / 8204 bytes
I (3549) WS CLIENT: Received 2920 / 8204 bytes
I (3558) WS CLIENT: Received 908 / 8204 bytes
I (3699) WS CLIENT: Websocket connected !

Data splitted into 4 different events.

gabsuren, 2023-05-17:
@theodem thank you for your notes. Indeed the data can be splited in `tcp_transport` layer and then it send to `websocket` application. But as you can see `esp_websocket_event_data_t` structure contains information regarding `payload_len` and `payload_offset`

https://github.com/espressif/esp-protocols/blob/9637517192ad637d1331cc80332fec24bd090fdb/components/esp_websocket_client/include/esp_websocket_client.h#LL70C1-L80C3

```
typedef struct {
    const char *data_ptr;                   /*!< Data pointer */
    int data_len;                           /*!< Data length */
    bool fin;                               /*!< Fin flag */
    uint8_t op_code;                        /*!< Received opcode */
    esp_websocket_client_handle_t client;   /*!< esp_websocket_client_handle_t context */
    void *user_context;                     /*!< user_data context, from esp_websocket_client_config_t user_data */
    int **payload_len**;                        /*!< Total payload length, payloads exceeding buffer will be posted through multiple events */
    int **payload_offset**;                     /*!< Actual offset for the data associated with this event */
    esp_websocket_error_codes_t error_handle; /*!< esp-websocket error handle including esp-tls errors as well as internal websocket errors */
} esp_websocket_event_data_t; 
```

And the responsibility of collecting that data falls on the application layer not the library. 
Please let us know if you have any other concern.

Regards,
Suren

closed 2023-06-20 by Alvin1Zhang