espressif/esp-idf #11020: [v4.4] DHCP Server registering multiple callbacks on multiple start/stop

KonssnoK, 2023-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-116-g00bb43ff24

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

Coming after https://github.com/espressif/esp-idf/issues/10762 .

The DHCP Server now is able to restart each time with our patch (no further changes provided from espressif to date).

When one client connects to the mesh root, we expect the root to provide one IP address

### What is the actual behavior?

The root provides multiple times the IP address, having a callback triggered for each time the dhcps was started

### Steps to reproduce.

1. Start DHCPS
2. Connect child and give IP
3. Disconnect
4. Stop DHCPS
5. Repeat from 1
...

### Debug Logs.

```plain
W (10:52:40.010) mesh_main: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:2
I (153191982) mesh: <nvs>write assoc:1
I (10:52:40.027) mesh_main: <MESH_EVENT_CHILD_CONNECTED>aid:1, 7c:df:a1:e0:8b:7c
I (10:52:40.261) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2
W (10:52:40.262) mesh_main: IP EVENT 2
I (10:52:40.297) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2
W (10:52:40.299) mesh_main: IP EVENT 2
I (10:52:40.303) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2       
W (10:52:40.309) mesh_main: IP EVENT 2
I (10:52:40.351) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2
W (10:52:40.352) mesh_main: IP EVENT 2
I (10:52:40.357) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2       
W (10:52:40.362) mesh_main: IP EVENT 2
I (10:52:40.367) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2       
W (10:52:40.375) mesh_main: IP EVENT 2
I (10:52:40.404) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2
W (10:52:40.406) mesh_main: IP EVENT 2
I (10:52:40.417) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2       
W (10:52:40.418) mesh_main: IP EVENT 2
I (10:52:40.452) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2
W (10:52:40.453) mesh_main: IP EVENT 2
I (10:52:41.359) mesh_main: Tried to publish layer:1 IP:192.168.253.144
```

xueyunfei998, 2023-03-23:
hi @KonssnoK 

I (10:52:40.452) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 10.0.0.2

Does this log have any impact?

xueyunfei998, 2023-03-23:
hi @KonssnoK 
[dhcp_server.txt](https://github.com/espressif/esp-idf/files/11049422/dhcp_server.txt)
Please add this log test. Thank you

KonssnoK, 2023-03-24:
i can't currently proceed due to crashes on other issues.
I'll continue when those are solved.

https://github.com/espressif/esp-idf/issues/9955
https://github.com/espressif/esp-idf/issues/11006

KonssnoK, 2023-03-30:
@xueyunfei998 as expected the issue is the udp pcbs not being correctly cleaned, as can be seend from the following log

```
I (16:18:09.176) mesh_main: <MESH_EVENT_CHILD_CONNECTED>aid:1, 7c:df:a1:e0:8b:7c
tcpip_thread: PACKET 0x3fcf3094
pcb (100000A, 67) <-- (0, 0)
pcb (0, 68) <-- (0, 67)
pcb (100000A, 67) <-- (0, 0)
pcb (100000A, 67) <-- (0, 0)
pcb (0, 56474) <-- (0, 0)
tcpip_thread: PACKET 0x3fcf7ef0
pcb (100000A, 67) <-- (0, 0)
pcb (0, 68) <-- (0, 67)
pcb (100000A, 67) <-- (0, 0)
pcb (100000A, 67) <-- (0, 0)
pcb (0, 56474) <-- (0, 0)
handle_dhcp,1064
send_ack,706
```

KonssnoK, 2023-03-30:
so, from further analysis it seems the issue is triggered by the example code.

if `esp_netif_dhcps_stop(netif_ap);` is not called when changing netif interface, the udp pcb will continue to contain references , triggering multiple times on a single packet.

i will change the code of the example grouping the destroy code as follows>
[image]

KonssnoK, 2023-04-03:
(the example ip_internal_network should be updated)

closed 2023-04-03 by KonssnoK

xueyunfei998, 2023-04-11:
> (the example ip_internal_network should be updated)

It has been updated to 4.4