espressif/esp-idf #11021: [MESH] v4.4 Multiple nodes connecting as ROOT if started together

KonssnoK, 2023-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
this is more of a question on how the mesh library is supposed to work.

We see that sometimes multiple nodes connect directly to the router, especially when starting devices at the same time, becoming effectively root nodes.

How is this handled?
Will one of the devices detach and connect to the other one?
After how long?

Thanks!

to test this we use (as usual):
https://github.com/KonssnoK/esp-idf/tree/bug/mesh_wifi/examples/mesh/ip_internal_network

KonssnoK, 2024-07-26:
@zhangyanjiaoesp i guess i can close this ticket since you explained
that `ESP_ERROR_CHECK(esp_mesh_allow_root_conflicts(false));`

should handle this.

Waiting for confirmation, thanks!

zhangyanjiaoesp, 2024-07-29:
> @zhangyanjiaoesp i guess i can close this ticket since you explained that `ESP_ERROR_CHECK(esp_mesh_allow_root_conflicts(false));`
> 
> should handle this.
> 
> Waiting for confirmation, thanks!

yes

closed 2024-07-31 by KonssnoK