espressif/esp-idf #11024: ESP32-C3 doesn´t wake up from deep sleep by using GPIO

Kampi, 2023-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a hall sensor (push-pull) attached to GPIO 3 to an ESP32-C3 and I´m using the following code:

```
#include <stdio.h>
#include <stdlib.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_sleep.h"
#include "esp_log.h"
#include "driver/rtc_io.h"
#include "soc/rtc.h"

extern "C" void app_main(void)
{
    const gpio_config_t config = {
        .pin_bit_mask = BIT(GPIO_NUM_3),
        .mode = GPIO_MODE_INPUT,
    };
    ESP_ERROR_CHECK(gpio_config(&config));

/*
    while(1)
    {
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    ESP_LOGI("A", "IO: %u", gpio_get_level(GPIO_NUM_3));
    }
*/
    ESP_ERROR_CHECK(esp_deep_sleep_enable_gpio_wakeup(BIT(GPIO_NUM_3), ESP_GPIO_WAKEUP_GPIO_HIGH));
    printf("Enabling GPIO wakeup on pins GPIO%d\n", GPIO_NUM_3);

    printf("Entering deep sleep\n");

    esp_deep_sleep_start();
}
```

I can validate the logic level switching from the hall sensor by reading the GPIO, but the ESP32-C3 doesn´t wake up. What´s the reason for that?

nopnop2002, 2023-04-03:
GPIO wakeup is only possible in light sleep.

Before entering Light-sleep mode, check if any GPIO pin to be driven is part of the VDD_SDIO power domain. If so, this power domain must be configured to remain ON during sleep.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#wakeup-sources

Kampi, 2023-04-04:
The issue comes from the 100k serial resistors between the sensor and the ESP. The serial resistors influence the pull-down/pull-up resistors.

closed 2023-04-04 by Kampi