espressif/esp-idf #11026: ESP32-S2

nabijaczleweli, 2023-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4051-g4f0769d2ed

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Wemos S2 mini (ESP32-S2FN4R2)

### Power Supply used.

USB

### What is the expected behavior?

Connect to Wi-Fi.

### What is the actual behavior?

```
I (26) boot: ESP-IDF 4f0769d-dirty 2nd stage bootloader

I (26) boot: compile time Mar 20 2023 15:30:13

I (26) boot: chip revision: v0.0

I (26) boot.esp32s2: SPI Speed      : 40MHz

I (27) boot.esp32s2: SPI Mode       : DIO

I (27) boot.esp32s2: SPI Flash Size : 2MB

I (28) boot: Enabling RNG early entropy source...

I (29) boot: Partition Table:

I (29) boot: ## Label            Usage          Type ST Offset   Length

I (30) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (31) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (31) boot:  2 factory          factory app      00 00 00010000 00100000

I (32) boot: End of partition table

I (33) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1fd00h (130304) map

I (69) esp_image: segment 1: paddr=0002fd28 vaddr=3ffc7330 size=002f0h (   752) load

I (69) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=74578h (476536) map

I (198) esp_image: segment 3: paddr=000a45a0 vaddr=3ffc7620 size=02af8h ( 11000) load
[... 276 lines cut ...]
Mon Mar 20 16:35:27 2023 user.err csrMeshGw[6555]: MSGQ :: CsrMeshGetMsgFromTxQueue curTxIndex : 0, tag count : 1, length : 30

Mon Mar 20 16:35:27 2023 user.err csrMeshGw[6555]:

CsrMeshMsgQRead: qId=0 errno=42, No message of desired type!!!!

Mon Mar 20 16:35:37 2023 user.err csrMeshGw[6555]:

CsrMeshMuxDemuxFindModelRes : Event : 0xef01, Response Event : 0xffff

Mon Mar 20 16:35:37 2023 user.err csrMeshGw[6555]: MSGQ :: CsrMeshAddMsgToTxQueue idx: 0x0, Msg Len: 30, Repeat: 2

Mon Mar 20 16:35:37 2023 user.err csrMeshGw[6555]: MSGQ :: CsrMeshGetMsgFromTxQueue curTxIndex : 0, tag count : 2, length : 30

Mon Mar 20 16:35:37 2023 user.err csrMeshGw[6555]: MSGQ :: CsrMeshGetMsgFromTxQueue curTxIndex : 0, tag count : 1, length : 30

Mon Mar 20 16:35:37 2023 user.err csrMeshGw[6555]:

CsrMeshMsgQRead: qId=0 errno=42, No message of desired type!!!!
```
```

### More Information.

Also tried on `v5.1-dev`, also tried the `protocols/https_server/simple` example. Result 200 is `WIFI_REASON_BEACON_TIMEOUT`. It's unclear to me how this helps.

Xiehanxin, 2023-03-21:
hi, could you please try release/v4.4, master or v5.1 is still under development, if you want to develop the ESP32-S2, v4.4 is enough.
and make sure the wifi is 2.4G not 5G

nabijaczleweli, 2023-03-24:
I got Luatos ESP32C3-CORE boards and can confirm the same program (rebuilt, of course) works there.

Xiehanxin, 2023-03-27:
hi @nabijaczleweli, dou you have more ESP32-S2 boards? Could you please try another one

nabijaczleweli, 2023-03-27:
Unfortunately not, I just got two Wemos S2 minis, and am not particularly keen on buying more right now. I'm happy to put this down to "broken boards" – it appears that they function well for everyone else, and the same code versions work with the ESP32-C3. If I ever get more S2 boards I'll post an update, but feel free to close this issue for now.

Indastri, 2024-03-06:
Closing the issue as per request from the user. Feel free to reopen or create a new issue.

Cheers

closed 2024-03-06 by Indastri