espressif/esp-idf #11031: How to perform IOT ES512 ES256 encryption?

loveccv, 2023-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 Hello everyone, I need to link to the IOT cloud and require ES512 and ES256 verification.
I use ESP-IDF 4.4 and ESP32-WROVER-B(8 MB).
The esp-idf/components/mbedtls/mbedtls/programs/pkey/ecdsa.c example is incorrect and cannot be passed https://jwt.io/ verification,  The parameter specified as #define ECPARAMS    MBEDTLS_ECP_DP_SECP256R1.

How can I get the correct results if I use the following information .
JWT header
{"alg":"ES256","typ":"JWT"}
JWT header base64
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9

data
{"iat":51,"exp":3651,"iss":"esp32"}
data base64
eyJpYXQiOjUxLCJleHAiOjM2NTEsImlzcyI6ImVzcDMyIn0

JWT signed 
eyJhbGciOiJFUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjUxLCJleHAiOjM2NTEsImlzcyI6ImVzcDMyIn0.JAWJQ17Vaa5RQ62Lli1WkrKDu95sbwkI1NuL7EE3MphQmZ5BwhbIw2xbiSSjOIfHHP-Oer4SzvrpvHziN1SFxw

-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAErbJbQ9/uCBv+76EkuxzvzSHRNrV+
EM+xpJQXT3eI/d2ilvFM/OvTgWPU3UotDA63Snt+uZaow3YQcihzDhgsRw==
-----END PUBLIC KEY-----

-----BEGIN EC PRIVATE KEY-----
MHcCAQEEIHGydA+QgPlivuLor6T+6oJs+Wf3thxgeegVF/IWqnQsoAoGCCqGSM49 AwEHoUQDQgAErbJbQ9
/uCBv+76EkuxzvzSHRNrV+EM+xpJQXT3eI/d2ilvFM/OvT
gWPU3UotDA63Snt+uZaow3YQcihzDhgsRw==
-----END EC PRIVATE KEY-----

thank you

loveccv, 2023-03-27:
@mahavirj  Thank you. I'm going to test this library

Robtraaws, 2023-03-28:
@mahavirj 
have you got the sketch esp_18w8jwt for arduino on .ino? or do you know to do it in? 'Cause that sketch is .c, thanks

mahavirj, 2023-04-03:
Please refer to Arduino documentation for this, it should be fairly straight-forward to create a small wrapper that is Arduino compatible on top of this component. 

I am closing this issue as its scope is outside of IDF. Please feel free to raise any issues on the repository pointer shared above...

closed 2023-04-03 by mahavirj