espressif/esp-idf #11032: when ota task and mcpwm was running , the wdt task was triggered

luacerbb, 2023-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-V4

### Power Supply used.

External 5V

### What is the expected behavior?

it can ota succee and not abort

### What is the actual behavior?

system abort,when ota and mcpwm was running

### Steps to reproduce.

1、CONFIG_MCPWM_ISR_IN_IRAM=y
2、start system.
3、when mcpwn was  captured singal lev ,the ota task to update fireware，system aborted.

### Debug Logs.

```plain
E (74727) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (74727) task_wdt:  - IDLE (CPU 1)
E (74727) task_wdt: Tasks currently running:
E (74727) task_wdt: CPU 0: ota_task
E (74727) task_wdt: CPU 1: ook_task
E (74727) task_wdt: Aborting.

abort() was called at PC 0x4013bc24 on core 0
0x4013bc24: task_wdt_isr at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/task_wdt.c:176 (discriminator 3)

Backtrace: 0x40081c92:0x3ffbe870 0x40091369:0x3ffbe890 0x400977e2:0x3ffbe8b0 0x4013bc24:0x3ffbe920 0x40082dfd:0x3ffbe940 0x4000bfed:0x3ffeb1d0 0x400943fa:0x3ffeb1e0 0x400823d3:0x3ffeb200 0x4008b6df:0x3ffeb220 0x4008c609:0x3ffeb240 0x4008c647:0x3ffeb260 0x4008ff19:0x3ffeb280 0x4008c13b:0x3ffeb2a0 0x4013862f:0x3ffeb2f0 0x4013777d:0x3ffeb310 0x4014cd2f:0x3ffeb340 0x400e60e8:0x3ffeb370
0x40081c92: panic_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/panic.c:408

0x40091369: esp_system_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/esp_system.c:137

0x400977e2: abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/abort.c:46

0x4013bc24: task_wdt_isr at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/task_wdt.c:176 (discriminator 3)

0x40082dfd: _xt_lowint1 at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x400943fa: vPortClearInterruptMaskFromISR at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/freertos/port/xtensa/port.c:332

0x400823d3: vPortExitCriticalSafe at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:605
 (inlined by) esp_intr_noniram_enable at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_hw_support/intr_alloc.c:809

0x4008b6df: spi_flash_enable_interrupts_caches_and_other_cpu at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/cache_utils.c:211

0x4008c609: cache_enable at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/spi_flash_os_func_app.c:63

0x4008c647: spi1_end at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/spi_flash_os_func_app.c:111

0x4008ff19: spiflash_end_default at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/esp_flash_api.c:143

0x4008c13b: esp_flash_erase_region at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/esp_flash_api.c:631

0x4013862f: esp_partition_erase_range at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/spi_flash/partition.c:540

0x4013777d: esp_ota_begin at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/app_update/esp_ota_ops.c:152

0x4014cd2f: esp_https_ota_perform at /mnt/hgfs/HWG004WRF/components/esp_https_ota/src/esp_https_ota.c:279

0x400e60e8: _bd_ota_task at /mnt/hgfs/HWG004WRF/main/bd_ota.c:258
```

### More Information.

it happened by accident and I'm not sure if it had anything to do with MCPWN

AxelLin, 2023-05-02:
@luacerbb  Could you share your sdkconfig?

luacerbb, 2023-05-24:
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
# CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

[... 1436 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

luacerbb, 2023-05-24:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11551202/sdkconfig.txt)