espressif/esp-idf #11033: The touch_pad_read_filtered function writes the value to other random variables

MNVasilev, 2023-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Build id: 20230207-1119

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC V4

### Power Supply used.

USB

### What is the expected behavior?

The function will work

### What is the actual behavior?

While the function is running, touch_value is written to other variables. 
Probably something with passing a pointer to a variable.

### Steps to reproduce.

1. Coding
2. Compilation
3. Run
4. Fail
[image]
[image]

L-KAYA, 2023-03-23:
Hi @MNVasilev, currently I can't catch any ideas that how `touch_pad_read_filtered ` can affect other variables, I'd like to do a further investigation. Could you provide more information, for example where `Min_Level` and `Max_level` defined and where they are updated?

MNVasilev, 2023-03-23:
> Hi @MNVasilev, currently I can't catch any ideas that how `touch_pad_read_filtered ` can affect other variables, I'd like to do a further investigation. Could you provide more information, for example where `Min_Level` and `Max_level` defined and where they are updated?

#define TOUCH_COUNT	4

static int Min_Level = 0; 
static int Max_Level = 0;

static void TP_Set_Thresholds(void){
    uint16_t touch_value;
    printf("Min_Level init0 = %d\n", Min_Level);
    printf("Max_Level init0 = %d\n", Max_Level);
    
    for (int i = 0; i < TOUCH_COUNT; i++) {
        //read filtered value
        touch_pad_read_filtered(Touch_Num[i], &touch_value);

        printf("Min_Level init1 = %d\n", Min_Level);
        printf("Max_Level init1 = %d\n", Max_Level);

        s_pad_init_val[Touch_Num[i]] = touch_value;

        ESP_LOGI(TAG, "test init: touch pad [%d] val is %d", Touch_Num[i], touch_value);
        //set interrupt threshold.

        ESP_ERROR_CHECK(touch_pad_set_thresh(Touch_Num[i], touch_value * 2 / 3));
    }

}
.....

// Initialize touch pad peripheral, it will start a timer to run a filter
ESP_LOGI(TAG, "Initializing touch pad");
ESP_ERROR_CHECK(touch_pad_init());

// If use interrupt trigger mode, should set touch sensor FSM mode at 'TOUCH_FSM_MODE_TIMER'.
touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);
// Set reference voltage for charging/discharging
// For most usage scenarios, we recommend using the following combination:
// the high reference valtage will be 2.7V - 1V = 1.7V, The low reference voltage will be 0.5V.
touch_pad_set_voltage(TOUCH_HVOLT_2V7, TOUCH_LVOLT_0V5, TOUCH_HVOLT_ATTEN_1V);

// Init touch pad IO
TP_Touch_Pad_Init();

// Initialize and start a software filter to detect slight change of capacitance.
touch_pad_filter_start(TOUCHPAD_FILTER_TOUCH_PERIOD);

// Set thresh hold
TP_Set_Thresholds();

After running TP_Set_Thresholes(), the Min_Level and Max_Level variables change their value.
There is no other code before that.

MNVasilev, 2023-03-26:
There was an error in the code. The array is out of range. The compiler didn't swear, but values were assigned to random variables.

closed 2023-03-26 by MNVasilev