espressif/esp-idf #11035: ESP32-S2FNR2

waruqi, 2023-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4051-g4f0769d2ed

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2FNR2

### Power Supply used.

USB

### What is the expected behavior?

I use `idf.py build` to build `examples/get-started/hello_world` project and flash it 

it will report some errors after downloading.

```
hello_world ruki$ idf.py -p /dev/tty.usbmodem01 flash
Executing action: flash
Running ninja in directory /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build
Executing "ninja flash"...
[1/5] cd /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/esp-idf/esptool_py && /Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/partition_table/partition-table.bin /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/hello_world.bin
hello_world.bin binary size 0x29870 bytes. Smallest app partition is 0x100000 bytes. 0xd6790 bytes (84%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/bootloader/esp-idf/esptool_py && /Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5440 bytes. 0x1bc0 bytes (25%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /Users/ruki/projects/personal/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.25.3/bin/cmake -D IDF_PATH=/Users/ruki/projects/personal/esp-idf -D "SERIAL_TOOL=/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/ruki/projects/personal/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build -P /Users/ruki/projects/personal/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s2 -p /dev/tty.usbmodem01 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port /dev/tty.usbmodem01
Connecting...
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 48:27:e2:55:f5:ea
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x00039fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21568 bytes to 13671...
Writing at 0x00001000... (14 %)
Writing at 0x0000253e... (28 %)
Writing at 0x00003170... (42 %)
Writing at 0x00003cd0... (57 %)
Writing at 0x00004730... (71 %)
Writing at 0x000050f5... (85 %)
Writing at 0x00005c33... (100 %)
Wrote 21568 bytes (13671 compressed) at 0x00001000 in 0.3 seconds (effective 555.4 kbit/s)...
Hash of data verified.
Compressed 170096 bytes to 94553...
Writing at 0x00010000... (2 %)
Writing at 0x0001115c... (4 %)
Writing at 0x0001278e... (6 %)
Writing at 0x0001400c... (8 %)
Writing at 0x000156f7... (10 %)
Writing at 0x00016f6e... (12 %)
Writing at 0x0001896d... (14 %)
Writing at 0x0001a838... (17 %)
Writing at 0x0001b400... (19 %)
Writing at 0x0001bd7a... (21 %)
Writing at 0x0001c95f... (23 %)
Writing at 0x0001d3fb... (25 %)
Writing at 0x0001e02a... (27 %)
Writing at 0x0001ec31... (29 %)
Writing at 0x0001f6d7... (31 %)
Writing at 0x000201c0... (34 %)
Writing at 0x00020e4f... (36 %)
Writing at 0x000218e4... (38 %)
Writing at 0x00022349... (40 %)
Writing at 0x00022d35... (42 %)
Writing at 0x00023a82... (44 %)
Writing at 0x0002451d... (46 %)
Writing at 0x000251d1... (48 %)
Writing at 0x00025d55... (51 %)
Writing at 0x00026917... (53 %)
Writing at 0x00027393... (55 %)
Writing at 0x00027e8f... (57 %)
Writing at 0x000288f6... (59 %)
Writing at 0x000293a8... (61 %)
Writing at 0x00029e91... (63 %)
Writing at 0x0002a9e1... (65 %)
Writing at 0x0002c095... (68 %)
Writing at 0x0002ccad... (70 %)
Writing at 0x0002d6a6... (72 %)
Writing at 0x0002e061... (74 %)
Writing at 0x0002ef1b... (76 %)
Writing at 0x000303ae... (78 %)
Writing at 0x0003247e... (80 %)
Writing at 0x000334a3... (82 %)
Writing at 0x00033fdb... (85 %)
Writing at 0x00034a6c... (87 %)
Writing at 0x000355b0... (89 %)
Writing at 0x000362c2... (91 %)
Writing at 0x00037353... (93 %)
Writing at 0x00037d57... (95 %)
Writing at 0x0003889f... (97 %)
Writing at 0x000396b8... (100 %)
Wrote 170096 bytes (94553 compressed) at 0x00010000 in 1.4 seconds (effective 994.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 500.7 kbit/s)...
Hash of data verified.

Leaving...
WARNING: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
CMake Error at run_serial_tool.cmake:63 (message):
  
  /Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/ruki/projects/personal/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed.

FAILED: CMakeFiles/flash /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash 
cd /Users/ruki/projects/personal/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.25.3/bin/cmake -D IDF_PATH=/Users/ruki/projects/personal/esp-idf -D "SERIAL_TOOL=/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/ruki/projects/personal/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build -P /Users/ruki/projects/personal/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_39498 and /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_39498
```

then I run `idf.py monitor`, I cannot get any serial-io output.

```
hello_world ruki$ idf.py monitor
Executing action: monitor
Serial port /dev/cu.usbmodem01
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world
Executing "/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem01 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/hello_world.elf -m '/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/Users/ruki/projects/personal/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbmodem01 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

### What is the actual behavior?

I can see app output.

### Steps to reproduce.

```
cd examples/get-started/hello_world
idf.py build
idf.py flash
idf.py monitor
```

0xjakob, 2023-03-22:
@waruqi It looks like the chip is still in boot mode. Have you tried manually resetting it? If that does not work, please make sure that GPIO0 is pulled high, then try to reset again.

What you experience looks similar to what happens when you connect your chip without the reset lines or if the reset circuit does not work. This is commonly the case when only connecting RXD, TXD and GND of a USB to UART converter to the chip. In the latter case, a simple manual reset would help.

waruqi, 2023-03-22:
I've tried resetting it several times and it doesn't work.

I just used usb-typec to connect it.

0xjakob, 2023-03-22:
If you run `idf.py monitor`, do you see something like the following when pressing the reset button?:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0/1/SPI))
waiting for download
```

waruqi, 2023-03-22:
startup 

```
$ idf.py monitor
Executing action: monitor
Serial port /dev/cu.usbmodem01
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world
Executing "/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem01 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/hello_world.elf -m '/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/Users/ruki/projects/personal/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbmodem01 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

after pressing reset

```
None
Waiting for the device to reconnect...........................................................................................................................

```

waruqi, 2023-03-22:
> It looks like the chip is still in boot mode

maybe it still in boot mode after flashing.

But I flash other fireware, like micropython, it works fine.

just when I flash idf program, it does not work.

radimkarnis, 2023-03-22:
Hello @waruqi,
your log shows that flashing finished successfully (`Hash of data verified.` means success).

You are flashing the ESP32-S2 over a [software USB console](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-guides/usb-console.html). This has some limitations:
1) You have to reset the chip after flashing to run the app
2) If you reset the chip, the USB port disappears because the console is implemented in software. That's why you see `Waiting for the device to reconnect..` after pressing the reset button.
Other [limitations are explained in the docs](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-guides/usb-console.html#limitations).
3) The app has to be configured to use the right console.

Your problem is that you have not configured your app to use this USB-OTG console.
You have to run `idf.py menuconfig` and then navigate to `Component config > ESP System Settings` and change `Channel for console output` to `USB-CDC`.
Then rebuild the app, flash it, reset the chip manually, and try monitoring.

waruqi, 2023-03-22:
> You have to run idf.py menuconfig and then navigate to Component config > ESP System Settings and change Channel for console output to USB-CDC.
Then rebuild the app, flash it, reset the chip manually, and try monitoring.

I tried it, it still does not work.

[image]

```
$ idf.py menuconfig
Executing action: menuconfig
Running ninja in directory /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/sdkconfig
Compiler supported targets: xtensa-esp32s2-elf

-- App "hello_world" version: v5.1-dev-4051-g4f0769d2ed
-- Adding linker script /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/soc/esp32s2/ld/esp32s2.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/ruki/projects/personal/esp-idf/components/app_trace /Users/ruki/projects/personal/esp-idf/components/app_update /Users/ruki/projects/personal/esp-idf/components/bootloader /Users/ruki/projects/personal/esp-idf/components/bootloader_support /Users/ruki/projects/personal/esp-idf/components/bt /Users/ruki/projects/personal/esp-idf/components/cmock /Users/ruki/projects/personal/esp-idf/components/console /Users/ruki/projects/personal/esp-idf/components/cxx /Users/ruki/projects/personal/esp-idf/components/driver /Users/ruki/projects/personal/esp-idf/components/efuse /Users/ruki/projects/personal/esp-idf/components/esp-tls /Users/ruki/projects/personal/esp-idf/components/esp_adc /Users/ruki/projects/personal/esp-idf/components/esp_app_format /Users/ruki/projects/personal/esp-idf/components/esp_coex /Users/ruki/projects/personal/esp-idf/components/esp_common /Users/ruki/projects/personal/esp-idf/components/esp_eth /Users/ruki/projects/personal/esp-idf/components/esp_event /Users/ruki/projects/personal/esp-idf/components/esp_gdbstub /Users/ruki/projects/personal/esp-idf/components/esp_hid /Users/ruki/projects/personal/esp-idf/components/esp_http_client /Users/ruki/projects/personal/esp-idf/components/esp_http_server /Users/ruki/projects/personal/esp-idf/components/esp_https_ota /Users/ruki/projects/personal/esp-idf/components/esp_https_server /Users/ruki/projects/personal/esp-idf/components/esp_hw_support /Users/ruki/projects/personal/esp-idf/components/esp_lcd /Users/ruki/projects/personal/esp-idf/components/esp_local_ctrl /Users/ruki/projects/personal/esp-idf/components/esp_mm /Users/ruki/projects/personal/esp-idf/components/esp_netif /Users/ruki/projects/personal/esp-idf/components/esp_netif_stack /Users/ruki/projects/personal/esp-idf/components/esp_partition /Users/ruki/projects/personal/esp-idf/components/esp_phy /Users/ruki/projects/personal/esp-idf/components/esp_pm /Users/ruki/projects/personal/esp-idf/components/esp_psram /Users/ruki/projects/personal/esp-idf/components/esp_ringbuf /Users/ruki/projects/personal/esp-idf/components/esp_rom /Users/ruki/projects/personal/esp-idf/components/esp_system /Users/ruki/projects/personal/esp-idf/components/esp_timer /Users/ruki/projects/personal/esp-idf/components/esp_wifi /Users/ruki/projects/personal/esp-idf/components/espcoredump /Users/ruki/projects/personal/esp-idf/components/esptool_py /Users/ruki/projects/personal/esp-idf/components/fatfs /Users/ruki/projects/personal/esp-idf/components/freertos /Users/ruki/projects/personal/esp-idf/components/hal /Users/ruki/projects/personal/esp-idf/components/heap /Users/ruki/projects/personal/esp-idf/components/http_parser /Users/ruki/projects/personal/esp-idf/components/idf_test /Users/ruki/projects/personal/esp-idf/components/ieee802154 /Users/ruki/projects/personal/esp-idf/components/json /Users/ruki/projects/personal/esp-idf/components/log /Users/ruki/projects/personal/esp-idf/components/lwip /Users/ruki/projects/ [... cut ...]
Executing "/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem01 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/hello_world.elf -m '/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/Users/ruki/projects/personal/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbmodem01 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
None
Waiting for the device to reconnect................................................................
```

radimkarnis, 2023-03-22:
You have to first flash the device with idf.py flash, then reset it (either by pressing the button or reconnecting the USB cable) and then start monitoring with idf.py monitor.

Do not reset the chip while monitoring.

waruqi, 2023-03-22:
ok, it works now, thanks! :+1:

```
/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_27407
ruki-2:hello_world ruki$ ls -l /dev/tty.*
crw-rw-rw-  1 root  wheel  0x9000000  3 22 10:23 /dev/tty.BLTH
crw-rw-rw-  1 root  wheel  0x9000002  3 22 23:27 /dev/tty.usbmodem01
ruki-2:hello_world ruki$ ls -l /dev/tty.*
crw-rw-rw-  1 root  wheel  0x9000000  3 22 10:23 /dev/tty.BLTH
crw-rw-rw-  1 root  wheel  0x9000002  3 22 23:28 /dev/tty.usbmodem14401
ruki-2:hello_world ruki$ idf.py monitor
Executing action: monitor
Serial port /dev/cu.usbmodem14401
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world
Executing "/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/ruki/projects/personal/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem14401 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /Users/ruki/projects/personal/esp-idf/examples/get-started/hello_world/build/hello_world.elf -m '/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/Users/ruki/projects/personal/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbmodem14401 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (36) boot: ESP-IDF v5.1-dev-4051-g4f0769d2ed 2nd stage bootloader
I (37) boot: compile time Mar 22 2023 23:08:48
I (37) boot: chip revision: v0.0
I (37) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : DIO
I (38) boot.esp32s2: SPI Flash Size : 2MB
I (38) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (40) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (41) boot:  2 factory          factory app      00 00 00010000 00100000
I (43) boot: End of partition table
I (43) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=083ach ( 33708) map
I (50) esp_image: segment 1: paddr=000183d4 vaddr=3ffbee90 size=018a8h (  6312) load
I (52) esp_image: segment 2: paddr=00019c84 vaddr=40024000 size=06394h ( 25492) load
I (58) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=12c28h ( 76840) map
I (75) esp_image: segment 4: paddr=00032c50 vaddr=4002a394 size=04af4h ( 19188) load
I (86) boot: Loaded app from partition at offset 0x10000
I (87) boot: Disabling RNG early entropy source...
I (107) main_task: Started on CPU0
I (107) main_task: Calling app_main()
Hello world!
This is esp32s2 chip with 1 CPU core(s), WiFi, silicon revision v0.0, 2MB external flash
Minimum free heap size: 250192 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
I (36) boot: ESP-IDF v5.1-dev-4051-g4f0769d2ed 2nd stage bootloader
I (37) boot: compile time Mar 22 2023 23:08:48
I (37) boot: chip revision: v0.0
I (37) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : DIO
I (38) boot.esp32s2: SPI Flash Size : 2MB
I (38) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (40) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (41) boot:  2 factory          factory app      00 00 00010000 00100000
I (43) boot: End of partition table
I (43) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=083ach ( 33708) map
I (50) esp_image: segment 1: paddr=000183d4 vaddr=3ffbee90 size=018a8h (  6312) load
I (52) esp_image: segment 2: paddr=00019c84 vaddr=40024000 size=06394h ( 25492) load
I (58) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=12c28h ( 76840) map
I (74) esp_image: segment 4: paddr=00032c50 vaddr=4002a394 size=04af4h ( 19188) load
I (85) boot: Loaded app from partition at offset 0x10000
I (86) boot: Disabling RNG early entropy source...
I (105) main_task: Started on CPU0
I (105) main_task: Calling app_main()
Hello world!
This is esp32s2 chip with 1 CPU core(s), WiFi, silicon revision v0.0, 2MB external flash
Minimum free heap size: 250192 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
```