espressif/esp-idf #11037: ESP32-S2 RISC-V I2C not generating SCL clock. Getting Stuck forever during master read/write.

ParthaSarathiMishra, 2023-03-21:
I am using **ESP32-S2 MINI 1U** chip for our project.

We need to fetch I2C sensors in active as well as deep sleep mode.

GPIO 2: SCL
GPIO 3: SDA
All pullup enabled.

Presently I am trying to initialize I2C using RISC-V drivers as follows:

```
static void init_i2c(void)
{
    /* Configure RTC I2C */
    printf("Initializing RTC I2C ...\n");
    ulp_riscv_i2c_cfg_t i2c_cfg = ULP_RISCV_I2C_DEFAULT_CONFIG();
    ulp_riscv_i2c_master_init(&i2c_cfg);
}
```

init_i2c() is invoked in app_main() before the ULP is started.

I am starting to read the chip id using following statements:

```
ulp_riscv_i2c_master_set_slave_addr(0x6C);
ulp_riscv_i2c_master_set_slave_reg_addr(0x18);
ulp_riscv_i2c_master_read_from_device(&data_rd, 1);
```

ulp_riscv_i2c_master_read_from_device() is stuck forever. Same happens for ulp_riscv_i2c_master_write_to_device() also.

I checked that the SCL line is always 3.3v. It means the master is not generating clock signal.

What could be the reason?

sudeep-mohanty, 2023-03-22:
Hello @ParthaSarathiMishra,
Thank you for reporting this issue to us.

Could you try out the workaround suggested [here](https://github.com/espressif/esp-idf/issues/8415#issuecomment-1287063676) and let me know if it helps resolve the issue. If so, it could help me in formalising the fix.

It is not very clear from where the sensor is being read from, main core or ULP RISC-V core. If the `ulp_riscv_i2c_master_read_from_device()` call is done from the ULP RISC-V core, then I would suggest to not use the ULP RISC-V core and just do the chip_id read from the main core, i.e, do not initialise the ULP RISC-V all. This can help us understand if indeed the issue lies with the RTC I2C peripheral not generating the SCL clock. At this point, could you dump out the following registers - `RTC_I2C_CTRL_REG`, `SENS_SAR_I2C_CTRL_REG`, `RTC_I2C_STATUS_REG`, `RTC_I2C_INT_RAW_REG`. If possible, could you also share an example application code and the sensor type that you are using so that I can reproduce it at my end? Also, it would help if you could share analyser capture of the SDA and SCL lines if you have a protocol analyser at your disposal.

ParthaSarathiMishra, 2023-03-22:
@sudeep-mohanty 
Thanks for the response.

The ulp_riscv_i2c_master_read_from_device() is called from main core. We need to make sure its working from main CPU, then only we can think of testing from ULP RISC-V core.

ParthaSarathiMishra, 2023-03-22:
@sudeep-mohanty 
I have tried disabling ULP init, but same issue.

SCL line is high 3.3v always.

When our main code didn't work, we decided to use the example code just to communicate with our sensor. Following is the code, we modified from example code for our **MC3414** accelerometer sensor, to read its chip id.

```
extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

/************************************************
 * ULP utility APIs
 ************************************************/
static void init_ulp_program(void);

/************************************************
 * RTC I2C utility APIs
 ************************************************/
static void init_i2c(void);

void app_main(void)
{
    uint8_t data_rd = 0;
    int16_t ut_data = 0;
    int32_t up_data = 0;
    int32_t temperature = 0;
    int32_t pressure = 0;
    oss_mode_t oss_mode;

    gpio_config_t io_conf = {0};

        io_conf.intr_type = GPIO_INTR_DISABLE;
        io_conf.mode = GPIO_MODE_OUTPUT;
        io_conf.pin_bit_mask = (1ULL <<35);
        io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        io_conf.pull_up_en = GPIO_PULLUP_DISABLE;

        ESP_ERROR_CHECK(gpio_config(&io_conf));
        gpio_set_level(35, 0);

    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

    /* Not a wakeup from ULP
     * Initialize RTC I2C
     * Setup BMP180 sensor
     * Store current temperature and pressure values
     * Load the ULP firmware
     * Go to deep sleep
     */
    if (cause != ESP_SLEEP_WAKEUP_ULP) {
        printf("Not a ULP-RISC V wakeup (cause = %d)\n", cause);

        /* Initialize RTC I2C */
        init_i2c();

        /* Configure I2C slave address */
        ulp_riscv_i2c_master_set_slave_addr(0x6C /*BMP180_SENSOR_I2C_ADDR*/);

        /* Reset the BMP180 sensor*/
        //ulp_riscv_i2c_master_set_slave_reg_addr(/*BMP180_SENSOR_REG_ADDR_SOFT_RESET*/);
        //uint8_t data_wr = 0x18;//BMP180_SENSOR_CMD_SOFT_RESET;
        //ulp_riscv_i2c_master_write_to_device(&data_wr, 1);

        /* Confirm that the sensor is alive
         * The BMP180 returns the chip id 0x55 on quering reg addr 0xD0
         */
        ulp_riscv_i2c_master_set_slave_reg_addr(0x18 /*BMP180_SENSOR_REG_ADDR_WHO_AM_I*/);
        data_rd = 0;
        printf("RTC_I2C_CTRL_REG: %04X\n", *(unsigned int *)RTC_I2C_CTRL_REG);
        printf("SENS_SAR_I2C_CTRL_REG: %04X\n", *(unsigned int *)SENS_SAR_I2C_CTRL_REG);
        printf("RTC_I2C_STATUS_REG: %04X\n", *(unsigned int *)RTC_I2C_STATUS_REG);
        printf("RTC_I2C_INT_RAW_REG: %04X\n", *(unsigned int *)RTC_I2C_INT_RAW_REG);
        printf("Reading chip id...\n");
        ulp_riscv_i2c_master_read_from_device(&data_rd, 1);
        //if (data_rd != BMP180_SENSOR_CHIP_ID) {
        if (data_rd == 0) {
            printf("ERROR: Cannot communicate with I2C sensor\n");
            abort();
        }
        else
        {
        	printf("CHIP ID: %d", data_rd);
        }

        /* Load ULP firmware
         *
         * The ULP is responsible of monitoring the temperature and pressure values
         * periodically. It will wakeup the main CPU if the temperature and pressure
         * values are above a certain threshold.
         */
        //init_ulp_program();
    }

    /* Add a delay for everything to the printed before heading in to deep sleep */
    vTaskDelay(100);

    /* Go back to sleep, only the ULP RISC-V will run */
    printf("Entering deep sleep\n\n");

    /* RTC peripheral power domain needs to be kept on to keep RTC I2C related configs during sleep */
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());

    esp_deep_sleep_start();
}

static void init_i2c(void)
{
    /* Configure RTC I2C */
    printf("Initializing RTC I2C ...\n");
    ulp_riscv_i2c_cfg_t i2c_cfg = ULP_RISCV_I2C_DEFAULT_CONFIG();
    ulp_riscv_i2c_master_init(&i2c_cfg);
}
```
Following is the log:
ESP-ROM:ESP-ROM:esp32s2-rc4-20191025
Build:OcESP-ROM:ESP-ROM:ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40024283
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1304
load:0x4004b000,len:0x4
load:0x4004b004,len:0x934
load:0x4004f000,len:0x2f6c
entry 0x4004b154
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
RTC_I2C_CTRL_REG: 20000004
SENS_SAR_I2C_CTRL_REG: C06C
RTC_I2C_STATUS_REG: 0000
RTC_I2C_INT_RAW_REG: 012C
Reading chip id...

As you can see the ulp_riscv_i2c_master_read_from_device() function get stuck.

I have dumped the registers which you asked for.

Let me know what could be wrong ad anything else if I am missing.

sudeep-mohanty, 2023-03-22:
Thanks for sharing the results @ParthaSarathiMishra.

From the register log of `RTC_I2C_INT_RAW_REG`, it seems like the RTC I2C peripheral did send out a start signal and the device address (0x6c) and register address (0x18) _should_ have been sent out as well. (Bit 2, 3 and 8 being set suggest this). This also means that the SCL clock signal is being generated by the hardware. Not sure why it didn't show up on your probe though. But we also see an ACK error (Bit 5 is set). This could mean that either the driver failed to detect the ACK from the device or the device did not generate one. Just to rule out any issues with the sensor itself, could you confirm that it works with the regular I2C peripheral? You may use the [i2c_simple](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_simple) example if you haven't tried this. Did you also try the workaround of removing the [vTaskDelay(1)](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/ulp_riscv_i2c.c#L307) from the `ulp_riscv_i2c_master_read_from_device()` API?

ParthaSarathiMishra, 2023-03-22:
@sudeep-mohanty 
Thanks for the response.

Regular I2C drivers are working with all the sensors we have on the board. Only with RISC-V drivers this problem comes.

I tried removing the vTaskDelay(1) from ulp_riscv_i2c_master_read_from_device() function, it generated task watchdog reset. Hence didn't work.

sudeep-mohanty, 2023-03-24:
@ParthaSarathiMishra 
I tried to recreate the problem that you are facing but wasn't able to on my ESP32-S2-DevKit-C board and with the the BMP180 and BH1750 sensors. Regarding the SCL line being always pulled high, this is somewhat expected until the `I2C start` condition is generated. This should happen as soon as the following [register writes](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/ulp_core/ulp_riscv_i2c.c#L133) are executed in the `ulp_riscv_i2c_master_read_from_device()` call. Here is a reference capture - 

[image]
 
I did not find any reference to MC3414 being an I2C sensor but by any chance did you mean the [MC3419](https://cz.mouser.com/datasheet/2/821/MEMSIC_MC3419_Datasheet-1889799.pdf)? If so, I would try to acquire one for further tests with the RTC I2C on my end.

ParthaSarathiMishra, 2023-03-24:
@sudeep-mohanty 
Thanks.

Actually it's MC3419 not MC3414. Sorry for the typo.

I also tried in DevKit with BMP180, and found no issue. Could there be any issue because of three I2C sensors (MC3419, MCP9600, ENS210) connected? Then how it's working with main I2C driver? Makes no sense. The problem only comes in RISC-V drivers.

sudeep-mohanty, 2023-03-24:
@ParthaSarathiMishra Do the other sensors work as expected on the DevKit as well?

ParthaSarathiMishra, 2023-03-24:
@sudeep-mohanty 
I tested ENS210 from DevKit, it didn't work with RISC-V drivers. But as usual, with main I2C driver its working fine.

What's the difference in main i2c vs risc-v i2c?

ParthaSarathiMishra, 2023-03-27:
@sudeep-mohanty 
The timings were changed in the patch.
.i2c_pin_cfg.scl_io_num = GPIO_NUM_2,   \
         .i2c_pin_cfg.sda_pullup_en = true,      \
         .i2c_pin_cfg.scl_pullup_en = true,      \
-        .i2c_timing_cfg.scl_low_period = 5,     \
-        .i2c_timing_cfg.scl_high_period = 5,    \
-        .i2c_timing_cfg.sda_duty_period = 2,    \
-        .i2c_timing_cfg.scl_start_period = 3,   \
-        .i2c_timing_cfg.scl_stop_period = 6,    \
+        .i2c_timing_cfg.scl_low_period = 1.3,   \
+        .i2c_timing_cfg.scl_high_period = 0.6,  \
+        .i2c_timing_cfg.sda_duty_period = 1,    \
+        .i2c_timing_cfg.scl_start_period = 2,   \
+        .i2c_timing_cfg.scl_stop_period = 3,    \
         .i2c_timing_cfg.i2c_trans_timeout = 20, \

Now even the initialization failing, giving following error:
[image]

sudeep-mohanty, 2023-03-27:
@ParthaSarathiMishra,
Thanks for the update. The I2C bus config timings were updated to make sure that the SCL clock runs at a higher frequency. I suppose these did not work with the sensor you are working with. My apologies.
Could you please ignore the timing updates and retry the patch?

ParthaSarathiMishra, 2023-03-29:
@sudeep-mohanty 
The patch only fixed the stuck issue, but didn't made it operational for ENS210 and MCP6900 sensors.

The BMP180 working fine. But don't know what's wrong with ENS210 and MCP6900 sensors for risc-v drivers. With main I2C drivers, bothENS210 and MCP6900 sensors working fine. With risc-v i2c drivers it's nor working. Even the register writing is failing.
[image]

Am I using it wrong?

For example, following is the way the ENS210 is communicated with main I2C driver:

```
i2cBuffer[0] = ENS210_SYS_CTRL;
i2cBuffer[1] = 0;
esp_err_t ret = i2c_master_write_to_device(I2C_MASTER_NUM, ENS210_ADDR, i2cBuffer, 2, I2C_MASTER_TIMEOUT_MS);
if(ret != ESP_OK)
{
	ESP_LOGI(TAG, "Failed to send active mode write request to ENS210.");
	return ret;
}

vTaskDelay(1000 / portTICK_PERIOD_MS);

i2cBuffer[0] = ENS210_SYS_STAT;
ret = i2c_master_write_to_device(I2C_MASTER_NUM, ENS210_ADDR, i2cBuffer, 1, I2C_MASTER_TIMEOUT_MS);
if(ret != ESP_OK)
{
	ESP_LOGI(TAG, "Failed to send sys stat read request to ENS210.");
	return ret;
}

ret = i2c_master_read_from_device(I2C_MASTER_NUM, ENS210_ADDR, i2cBuffer, 1, I2C_MASTER_TIMEOUT_MS);
if(ESP_OK != ret)
{
	ESP_LOGI(TAG, "Failed to receive sys stat response from ENS210.");
}
else
{
	ESP_LOGI(TAG, "ENS210 sys stat: %d", i2cBuffer[0]);
}

```
ENS210_ADDR is the sensor slave address.

Following is how the same thing I tried to implement the same using RISC-V I2C drivers:
printf("Setting to active mode\n");
ulp_riscv_i2c_master_set_slave_addr(ENS210_ADDR);
ulp_riscv_i2c_master_set_slave_reg_addr(ENS210_SYS_CTRL);
i2cBuffer[0] = 0;
ulp_riscv_i2c_master_write_to_device(i2cBuffer, 1);
vTaskDelay(10 / portTICK_PERIOD_MS);

printf("Reading system stat\n");
ulp_riscv_i2c_master_set_slave_addr(ENS210_ADDR);
ulp_riscv_i2c_master_set_slave_reg_addr(ENS210_SYS_STAT);
ulp_riscv_i2c_master_read_from_device(data, 1);

The write and read both failing.
[image]

What else Am I missing?

This makes no sense.

Any suggestion?

ParthaSarathiMishra, 2023-04-04:
@sudeep-mohanty 
Sorry for delayed response.

We tried probing the SDA/SCL lines for both main and RTC drivers.

For RTC drivers, the ENS is not responding with right response.
[image]

The ENS210 address is 0x43. So it shall be left shifted to 1, and for write, the LSB shall be 0. So it shall become 0x86. But instead it's generation 0xC3. Seems address is not being left shifted and the MSB is set to 1.

We are running out of options.

Any suggestions?

sudeep-mohanty, 2023-04-05:
From the capture you have shared, I can see that the device address written is indeed 0x86 and not 0xC3 during a write operation. And I also see an ACK being generated by the ENS210. However, I do see the analyzer reporting a 0xC3 during a read setup. (Only seen in the decoded results on the side, not in the probe capture). This seems incorrect as the address should have been 0x87 for a read operation. Could you share a probe capture when the read operation is happening? Maybe you could just do only a read from the ENS210_SYS_STAT register and capture the SDA/SCL lines?

Another experiment which I would like you to try is to turn off the ACK check by the RTC I2C controller when a write operation is performed and see if the write goes through. This can be done by updating the [ulp_riscv_i2c.c](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/ulp_riscv_i2c.c#L410) file as below -

From:
```
ulp_riscv_i2c_format_cmd(cmd_idx++, ULP_I2C_CMD_WRITE, 0, 0, 1, 2 + size);
```

To:
```
ulp_riscv_i2c_format_cmd(cmd_idx++, ULP_I2C_CMD_WRITE, 0, 0, 0, 2 + size);
```

ParthaSarathiMishra, 2023-04-06:
@sudeep-mohanty 
The RTC I2C working fine with ENS210 after I disabled ACK checking in ulp_riscv_i2c_format_cmd() function. Also commeted out the printf logs in the driver.

Also transaction complying with multiple bytes (reading or writing) are not working. To make it work, I have to call the ulp_riscv_i2c_master_read_from_device() or ulp_riscv_i2c_master_write_to_device() multiple type with the register offset and buffer offset as parameter.

Can you let me know what could be the reason for this? Why ACK EN is working for BMP sensor but not for ENS? And why multiple byte data is not working even though the APIs support it?

sudeep-mohanty, 2023-04-06:
@ParthaSarathiMishra 
> The RTC I2C working fine with ENS210 after I disabled ACK checking in ulp_riscv_i2c_format_cmd() function. Also commented out the printf logs in the driver.

Glad to know this works for you!

> Also transaction complying with multiple bytes (reading or writing) are not working. To make it work, I have to call the ulp_riscv_i2c_master_read_from_device() or ulp_riscv_i2c_master_write_to_device() multiple times with the register offset and buffer offset as parameter.

The `ulp_riscv_i2c_master_write_to_device()` and `ulp_riscv_i2c_master_read_from_device()` should be able to do multiple byte writes and reads. What kind of error do you notice when attempting multi-byte read/writes? Could you share a sample code for this?

> Can you let me know what could be the reason for this? Why ACK EN is working for BMP sensor but not for ENS? And why multiple byte data is not working even though the APIs support it?

What I noticed from your previous analyzer capture is that the ENS210 does send out an ACK signal but the RTC I2C peripheral reports an ACK error. After which, it seems that the peripheral's state machine is in a bad state and is not able to process further read/write commands. Unfortunately, I am not sure why this is the case. This is something I'll have to consult the digital designers of this peripheral about. I have taken a note and will raise an internal ticket for this.

ParthaSarathiMishra, 2023-04-08:
@sudeep-mohanty 
The ulp_riscv_i2c_master_write_to_device() and ulp_riscv_i2c_master_read_from_device() are not doing multibyte. Following is the error I am getting for multibyte read.
[image]

[image]
[image]

I was trying to read 2 bytes from register 0x00 like following:
```
ulp_riscv_i2c_master_set_slave_addr(0x43);
ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
ulp_riscv_i2c_master_read_from_device(data, 1);
```

Instead of to make it work, I was trying like this:
```
data[0] = 0;
data[1] = 0;
ulp_riscv_i2c_master_set_slave_addr(0x43);
ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
ulp_riscv_i2c_master_read_from_device(data, 1);

ulp_riscv_i2c_master_set_slave_addr(0x43);
ulp_riscv_i2c_master_set_slave_reg_addr(0x01);
ulp_riscv_i2c_master_read_from_device(data + 1, 1);
```

Any idea why it's happening?

sudeep-mohanty, 2023-04-17:
Hello @ParthaSarathiMishra,
Apologies for the delayed reply.

The `ulp_riscv_i2c_master_read_from_device()` can do a multi-byte read from _one_ device sub-register address. It cannot do multi-byte read from _multiple_ device sub-register address. For instance, you should be able to do a 2 byte read from the address `0x00` but cannot do a multi-byte read from `0x00` and `0x01` with one invocation of the API.
Hence, the way you are performing the reads from `0x00` and `0x01` is the only way to do such a read.

Another tip I have here is that you would not need to program the device address (`0x43`) for every transaction. You should be able to program it once and continue with reads/writes.

ParthaSarathiMishra, 2023-04-17:
@sudeep-mohanty 
_For instance, you should be able to do a 2 byte read from the address 0x00 but cannot do a multi-byte read from 0x00 and 0x01 with one invocation of the API._
I didn't get this.

I was trying to read 2 bytes from 0x00 only, and it didn't work. Hence read 1 byte from 0x00 and 1 byte from 0x01.

ParthaSarathiMishra, 2023-04-17:
@sudeep-mohanty 
Thanks for the info.

Presently the ENS is responding with following changes:
1. In initialization, the timeout is increased to 2sec:
i2c_cfg.i2c_timing_cfg.i2c_trans_timeout = 2000;

2. Disabled ack checking.

But why there is no return statement for the ulp APIs? It would have been helpful for retry.

ParthaSarathiMishra, 2023-04-17:
@sudeep-mohanty 
Please see the update in the previous comment.

sudeep-mohanty, 2023-04-17:
@ParthaSarathiMishra 
Yes I agree that the application would be better served had the APIs had a return value. Unfortunately, we can't the change the API signature between major version updates of the SDK as it would be a breaking change and this is an upgrade which we will plan in the future.

ParthaSarathiMishra, 2023-04-17:
@sudeep-mohanty 
Since we are using multiple sensors, we are facing another write issue for OPT3001 light sensor.
The problem is their registers are just 1 byte apart in addressing but are of 16-bit.
image

You can see 01, 02, 03 are all 16-bit registers but 1 byte apart.

To do that we tried reading the first byte as:

ulp_riscv_i2c_master_set_slave_addr(0x01);
ulp_riscv_i2c_master_read_from_device(data, 1);
and second byte as:
ulp_riscv_i2c_master_read_from_device(data + 1, 2);

Reading working fine.

But writing failed with following approach:

ulp_riscv_i2c_master_write_to_device(data, 2)
It writes only the first byte and giving error for second byte.

Here 01 and 02 are different register.

Any suggestions?

sudeep-mohanty, 2023-04-18:
May I ask if the intention is the write 2 bytes to the same address or multiple address. Just like the read API, the write cannot do an auto-increment of the device sub-register address so if the intention is to write data on to register addresses incrementally then you would have to set the address for each write.

ParthaSarathiMishra, 2023-04-18:
@sudeep-mohanty 
Then how will interface with OPT3001 light sensor from RTC I2C?

The registers are 16-bit but addresses are only 1 byte apart.
[image]

So how should registers 01, 02 etc shall be read/write to?

sudeep-mohanty, 2023-04-18:
I'm not sure exactly how to OPT3001 expects data to be read or written but if you are trying to write 2 bytes to each of these addresses then your code would look something like this -

```
// Set OPT3001 device address
ulp_riscv_i2c_master_set_slave_addr(OPT3001_ADDR);

// Read 2 bytes from sub-reg 0x00
ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
ulp_riscv_i2c_master_read_from_device(data, 2);

// Read 2 bytes from sub-reg 0x01
ulp_riscv_i2c_master_set_slave_reg_addr(0x01);
ulp_riscv_i2c_master_read_from_device(data, 2);

// Write 2 bytes to sub-reg 0x00
ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
ulp_riscv_i2c_master_write_to_device(data, 2);

// Write 2 bytes to sub-reg 0x01
ulp_riscv_i2c_master_set_slave_reg_addr(0x01);
ulp_riscv_i2c_master_write_to_device(data, 2);

ParthaSarathiMishra, 2023-04-18:
@sudeep-mohanty 
Thanks.

But the problem is its only writing the first byte and the second byte is throwing error. Only the upper byte writing works fine.

```
E (13647) ulp_riscv_i2c: Write Failed!
E (13647) ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x10c
E (13647) ulp_riscv_i2c: RTC I2C Status Reg 0x666a0001
```

But with main driver its working fine.

What could be the reason?

o-marshmallow, 2023-04-18:
Hi @ParthaSarathiMishra ,

From what I see in, the registers of the OPT3001 are indeed 16-bit, even if indexed by an 8-bit value, so the code given by @sudeep-mohanty in the message https://github.com/espressif/esp-idf/issues/11037#issuecomment-1512552228 should be correct.

Can you please try writing the register 2 ("Low Limit") with some data, like 0x0102. So something like:
```
uint8_t data[] = { 0x1, 0x2 };
ulp_riscv_i2c_master_set_slave_addr(OPT3001_ADDR);
ulp_riscv_i2c_master_set_slave_reg_addr(0x02);
ulp_riscv_i2c_master_write_to_device(data, 2);
```

And can you please show us what the logic analyzer gives, as well as the error messages generated by the ESP chip?

Also, which frequency are you using and what pull-up resistors are you using? Internal pull-ups? External pull-ups?

sudeep-mohanty, 2023-06-20:
Closing issue as the original problem reported regarding SCL clock not being generated is resolved. Please feel free to open a new issue regarding any further problems.

closed 2023-06-20 by sudeep-mohanty

sidwarkd, 2024-06-13:
@sudeep-mohanty Having similar issues. I see the resolution was to modify ulp_riscv_i2c.c in the IDF but that change was never an official fix in the IDF. Is there an official fix expected for this or is the expectation that we have to alter the IDF manually in this case?

sudeep-mohanty, 2024-06-13:
@sidwarkd There have been a few fixes made to the RTC I2C driver but it may happen that the fix suggested here, i.e, disabling the ACK check wasn't a normal operation mode for the RTC I2C peripheral. May I ask if you observe similar errors? If so, we can investigate it further and formalize the fix. Would you mind opening a new ticket for us with your observations so that we could have a fresh look at it?

sidwarkd, 2024-06-13:
@sudeep-mohanty Thanks for the super fast reply. We are in the early stages of debugging the issue and came across this issue so we will continue to test and, if unable to resolve the issues, will open a new ticket with details and reference this one. Thanks.