espressif/esp-idf #11038: ESP32-H2 support status

igrr, 2023-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# ESP32-H2 support status in ESP-IDF

ESP-IDF v5.1, currently in development, will include initial support for the mass production version of the ESP32-H2 chip.

In the list below, supported features are marked with a checked checkbox, unsupported features have an unchecked checkbox. An internal issue reference (such as "IDF-1234") is listed at the end of the feature description to help us keep this list up to date:

- [ ] Unsupported feature (IDF-1234)
- [X] Supported feature

This list will be updated as the support for ESP32-H2 is being developed.

## Reporting issues about ESP32-H2

If you have an issue to report about any of the ESP32-H2 features, please create a separate issue. This issue is only intended for announcements of ESP32-H2 support status.

## Support for ESP32-H2 engineering sample revisions

In the past, Espressif has distributed a limited number of engineering samples of ESP32-H2 chip: ESP32-H2-beta1 and ESP32-H2-beta2. Preview support for these chip revisions is available in release/v5.0 branch.

These engineering sample revisions of the chip will *not* be supported in IDF v5.1. Likewise, support for the mass production version of ESP32-H2 will only be available in IDF v5.1 and later. It will not be backported to IDF v5.0.x.

---

### 802.15.4
- [X] 802.15.4 Mac
- [X] 802.15.4 CSL
### Thread
- [X] Thread FTD/MTD
- [X] Thread RCP
### Zigbee
- [ ] Zigbee 3.0 Home Automation (TZ-60)
- [ ] Zigbee RCP (TZ-60)
### Coexistence
- [X] Coexistence of Bluetooth LE and 802.15.4
- [ ] Coexist API IRAM timing optimization (TZ-82)
- [ ] 802.15.4 & BT Coexistence Scheme (TZ-81)
- [ ] 802.15.4 External Coexistence (TZ-61)
### Bluetooth LE
- [X] Bluetooth 5 Controller (Bluetooth 5.3 Certified)
- [X] Bluedroid Host
- [X] NimBLE Host
- [X] ESP-BLE-MESH
- [ ] BLE Modem Sleep and Light Sleep (IDF-6924）
- [ ] Bluetooth Power Control (IDF-6925）
- [X] HCI
### Peripherals
- ADC
  - [ ] ADC Calibration (IDF-6216)
  - [X] ADC continuous mode driver
  - [X] ADC oneshot mode driver
- [X] Dedicated GPIO driver
- Ethernet driver
  - [ ] SPI Ethernet driver (IDF-6232)
- [X] Event Task Matrix
- [X] GDMA
- [X] GPIO driver
  - [X] Analog Comparator
  - [X] Glitch Filter
  - [X] Hysteresis
- [X] GPIO sigma delta driver
- GPSPI
  - [X] SPI Master driver
  - [X] SPI Slave HD driver
  - [X] SPI Slave driver
- [X] GPTimer driver
- I2C
  - [X] I2C master driver
  - [X] I2C slave driver
- [X] I2S driver
  - I2S PDM rx mode
  - I2S PDM tx mode
  - I2S STD mode
  - I2S TDM mode
- [X] LCD driver
  - I2C LCD driver
  - SPI LCD driver
- [X] LEDC driver
  - [X] Gamma Curve Generation
- [X] MCPWM driver
- [X] PCNT driver
- Parallel IO
  - [ ] Parallel IO RX Driver (IDF-7038)
  - [X] Parallel IO TX Driver
- [X] RMT driver
- SD/SDIO/MMC driver
  - [X] SDSPI Host driver
- [X] SPI Flash driver
- Systimer driver
  - [X] Esptimer implementation over systimer
  - [X] OS tick port over systimer
- [X] TWAI driver
- [X] Temperature Sensor driver
- UART
  - [X] UART FIFO mode driver
  - [ ] UHCI (UART DMA mode) driver (IDF-6248)
- [X] USB Serial JTAG
- [X] memory copy by DMA
### Power management
- [ ] Auto Light Sleep (IDF-6269)
- [ ] Deep Sleep (IDF-6268)
- [ ] Dynamic Frequency Switch (IDF-6266)
- [ ] Light Sleep (IDF-6267)
### Security Features
- [X] AES accelerator
- [X] Digital Signature driver
- [X] ECC accelerator
- [ ] ECDSA driver (IDF-6287)
- [X] Flash encryption
- [X] HMAC driver
- [X] Permission Control
- [ ] RNG (IDF-6274)
- [X] RSA hardware accelerator driver
- [X] SHA accelerator
- [X] Secure boot
- Trusted Execution Environnment
  - [X] Access Permission Management
  - [X] TEE Controller
### System Features
- [X] Brownout Detector
- Cache
  - [X] Cache Driver
- [X] Efuse controller driver
- [X] Heap
- [X] MMU
- MSPI
  - [ ] MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz) (IDF-6256)
  - [ ] SPI Flash auto suspend (IDF-6257)
- Panic Handling Features
  - [ ] Bus Access Timeout Protection (IDF-6250)
  - [ ] Core Dump (IDF-6865)
  - [ ] GDB Stub (IDF-6863)
  - [X] Panic Handler
- [X] Watch Dog Timers

*Note: Nested features can be considered as supported, if the parent feature is shown as "Supported", and it's not explicitly mentioned as "Not supported".*

igrr, 2023-03-28:
Zigbee features are now supported on ESP32-H2.

hipal31, 2023-03-31:
`A fatal error occurred: This chip is ESP32-H2(beta2) not ESP32-H2. Wrong --chip argument?`
@igrr 
Which branch should I use for ESP32-H2(beta2)?

igrr, 2023-04-18:
GDB Stub, Core Dump, SPI Ethernet support has been added.

shinasys, 2023-04-19:
Does this provide APIs for not only the Zigbee 3.0 stack but also the 802.15.4 MAC? Or is there a plan to provide such APIs?
The reason is that I wanted it to be compatible with the 802.15.4 MAC device I developed in the past.

lygstate, 2023-05-18:
Does ESP32-H2 support for PSRAM DDR?

andylinpersonal, 2023-05-19:
Or at least any native support for memory-mapped read/write to external SPI PSRAM.

jack0c, 2023-08-17:
updated:
- [X] Auto Light Sleep
- [X] Deep Sleep
- [X] Dynamic Frequency Switch
- [X] Light Sleep

tom-borcin, 2024-01-17:
Update 2024/01/17:
- 802.15.4 & BT Coexistence Scheme is Done
- 802.15.4 External Coexistence is Done
- BLE Modem Sleep and Light Sleep is Done
- Bluetooth Power Control is Done
- ADC Calibration is Done
- SPI Flash auto suspend is Done

Only remaining items are:
- Coexist API IRAM timing optimization
- Parallel IO RX Driver
- UHCI (UART DMA mode) 
- MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz)
- Bus Access Timeout Protection

These remaining items will be supported based on customer requests. We are closing the issue and unpinning it. Please open new issue if you have a feature request, question or any other problem.

closed 2024-01-17 by tom-borcin