espressif/esp-idf #11039: Installation error: universal online installer of ESP-IDF V5.0.1 with C6 target coverage

loraahd, 2023-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Hello, Unfortunately I am not able to use my ESP32-C6 DEVKITC-1. That is because I am not able to install ESP-IDF V5.0.1 which is the only version that supports C6 targets. Please find attached the error message and the log file. I have no firewall or proxy server issues. 

I am new to ESP32 ecosystem, all what I need for the moment is to flash my C6 and send commands to scan and associate the C6 WiFi to an access point.

Anyone could give support please? BR

[image]
[ADH.Setup.Log.2023-03-21.001.txt](https://github.com/espressif/esp-idf/files/11037272/ADH.Setup.Log.2023-03-21.001.txt)

### What is the actual behavior?

ESP-IDF V5.0.1 installation error

### Steps to reproduce.

try to install the universal online installer of ESP-IDF V5.0.1 with C6 target coverage, using window installer from the page: https://dl.espressif.com/dl/esp-idf/?idf=4.4

dobairoland, 2023-03-22:
Hi @loraahd. Please see https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility. ESP32-C6 is going to be supported by the upcoming v5.1 release (currently the master branch). It is not supported in v5.0.1.

Here is the work-in-progress support status issue: https://github.com/espressif/esp-idf/issues/10423

closed 2023-03-22 by dobairoland

loraahd, 2023-03-22:
> Hi @loraahd. Please see https://github.com/espressif/esp-idf#esp-idf-release-and-soc-compatibility. ESP32-C6 is going to be supported by the upcoming v5.1 release (currently the master branch). It is not supported in v5.0.1.
> 
> Here is the work-in-progress support status issue: #10423

thanks @dobairoland

timfrankentwf, 2023-03-22:
I will share some steps that I did to setup my tools on Windows, including built-in JTAG debugging on VScode.
To support the C6, you need to use the Master branch.
_Please note that this is no official guide_

## 1. Install ESP-IDF and TOOLS
1. Create the following installation folders:
    - **ESP-IDF** folder: `C:/dev/idf/`
    - **ESP-IDF TOOLS** folder: `C:/dev/tools/`
2. Download the [Universal Online installer](https://dl.espressif.com/dl/esp-idf/)
3. Apply any fix it may present during the pre-installation system check.
4. Select the option to download the software: `Download ESP-IDF`
5. Select the latest version (pre-release): `master (development branch)`
6. Choose the **ESP-IDF** folder from step 1 using the `Browse...` button
7. Choose the **ESP-IDF TOOLS** folder from step 1 using the `Browse...` button
    - This may also create a subfolder by itself
8. Select `Full installation`
    - Check **ESP32-C6** under *Chip Targets*
    - Check all the boxes under *Drivers*
9. Continue all steps...

## 2. Update OpenOCD package in tools.json
1. Edit the *tools.json* file
    - `C:/dev/idf/tools/tools.json`
2. *Win32 and Win64* - Change to the followig version or the lastest release to support debugging the ESP32-C6:
    - Line 771: name to `"v0.12.0-esp32-20230221"`
    - Line 774 and 779: sha256 to `"85d1f72f442f8e362655040e22b9ff2666432ac3d9022c8238725e3349b3fad1"`
    - Line 765 and 780: size to `"2491691"`
    - Line 776 and 781: url to `"https://github.com/espressif/openocd-esp32/releases/download/v0.12.0-esp32-20230221/openocd-esp32-win32-0.12.0-esp32-20230221.zip"`
3. Save changes

## 3. Setup paths
1. Open Command prompt with Admin rights
2. Change directory to the *IDF* folder
    - `cd C:\dev\idf`
3. Run the install script
    - `install.bat`
    - This should install all the tools, python virtual env and paths.
4. Run the export script
    - `export.bat`
5. Varify that all the paths and tools are installed correctly
    - Open Winodows Environment variables
    - Check if **IDF_PATH** says `C:\dev\idf`
    - Check if **IDF_TOOLS_PATH** says `C:\dev\tools\Espressif`
    - Check if **Path** includes `%IDF_PATH%\tools`

## 4. Manually check if the tools work
1. Follow the steps on the [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/get-started/windows-setup.html#get-started-windows-first-steps).

## 5. Setup VSCode with ESP-IDF plugin
1. Open VSCode
2. Install the [Espressif IDF](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) plugin
3. Choose the advanced setup option
4. Select the following information
    - Download server: `Github`
    - ESP-IDF version: `Find ESP-IDF in your system`
    - ESP-IDF directory: `C:\dev\idf`
    - ESP-IDF Tools directory: `C:\dev\tools\Espressif`
    - Install, should skip most steps since everything is already installed.

## 6. Use example project: hello world
1. Press `F1` and select `ESP-IDF: Show Examples Projects`
2. Choose `hello_world`
3. Select a folder where the project subfolder will be saved
    - e.g. `C:\dev\projects\`
4. Select the correct Serial port
5. Select the correct Target
6. Run Menuconfig
    - Set the correct Flash size
    - Maybe set lower optimisation
7. Build project
8. Select `JTAG` as flash method
9. Flash program

## 7. Debug program
1. Add the following section to the `launch.json`:
    ``` json
    {
      "name": "GDB",
      "type": "cppdbg",
      "request": "launch",
      "MIMode": "gdb",
      "miDebuggerPath": "C:\\dev\\tools\\Espressif\\tools\\riscv32-esp-elf-gdb\\12.1_20221002\\riscv32-esp-elf-gdb\\bin\\riscv32-esp-elf-gdb.exe",
      "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
      "windows": {
        "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
      },
      "cwd": "${workspaceFolder}",
      "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
      "setupCommands": [
        { "text": "target remote :3333" },
        { "text": "mon reset halt" },
        { "text": "thb app_main" },
        { "text": "flushregs" }
      ],
      "externalConsole": false,
      "logging": {
        "engineLogging": true
      }
    }
    ```
2. Start OpenOCD server
3. *First upload program by JTAG after every hardreset before debugging, else debug session fails to connect.*
4. Start debug session by using `GDB` under *run and debug* tab
    - Program should automaticly break at app_main()
5. [Optional] Add SVD file to the main directory of the project with the name `esp32.svd`

timfrankentwf, 2023-03-23:
The ESP32-C6 support is currently being developed. Therefore, if you want to use the ESP32-C6 you need to use the master branch (which support most of the features already).
I cannot say anything about the ETA of the next official release.

When you use the setup, please remove all folders related to the IDF setup and start over using this option:
[image]

loraahd, 2023-03-23:
Hello, 
just noticed that I have the warnings below. knowing that I am not using a proxy or a firewall that can block the access to the servers!
[image]

timfrankentwf, 2023-03-23:
Looks like you are unable to connect to the servers, maybe the program is being blocked?
Good luck!

loraahd, 2023-03-24:
Hello,
I've solved server access related issues. I've installed the master branch of the online IDF with C6 support. The, under CMD, I set target as follows: C:\dev\idf\examples\get-started\hello_world>id.fpy --preview set-target esp32c6

but it exits with the following error: 
CMake Error at C:/dev/idf/tools/cmake/component.cmake:479 (add_library):
  Cannot find source file:

    C:/dev/idf/components/unity/unity/src

Indeed, src folder is missing.

why am I having that error? how to fix it?
thanks 
BR